FreeCAD/homebrew-freecad #187: 404 error on coin@4.0.0_1 and nglib@6.2.2101 bottles

dirkolbrich, 2021-08-02:
Upgrade for the coin & nglib bottles not possible, gives 404 error on download. This can be verified too, if URL of bootle is opened in browser window.

New installs of the FreeCAD dependency packages with `brew install --only-dependencies freecad --with-packaging-utils` fails as well.

```bash
➜  ~ brew upgrade
==> Upgrading 2 outdated packages:
freecad/freecad/nglib v6.2.2007 -> 6.2.2101
freecad/freecad/coin@4.0.0 4.0.0 -> 4.0.0_1
==> Upgrading freecad/freecad/coin@4.0.0
  4.0.0 -> 4.0.0_1

==> Downloading https://justyour.parts:8080/freecad/coin%404.0.0-4.0.0_1.big_sur.bottle.tar.gz
-#O#- #   #
curl: (22) The requested URL returned error: 404
Error: Failed to download resource "coin@4.0.0"
Download failed: https://justyour.parts:8080/freecad/coin%404.0.0-4.0.0_1.big_sur.bottle.tar.gz
==> Upgrading freecad/freecad/nglib
  v6.2.2007 -> 6.2.2101

==> Downloading https://justyour.parts:8080/freecad/nglib-6.2.2101.big_sur.bottle.tar.gz
-#O#- #   #
curl: (22) The requested URL returned error: 404
Error: Failed to download resource "nglib"
Download failed: https://justyour.parts:8080/freecad/nglib-6.2.2101.big_sur.bottle.tar.gz
```

ipatch, 2021-08-02:
thanks for reporting this, this is something i noticed while grabbing all the bottles for _big sur_ and _catalina_ the other day.

i believe the formulae files were updated (the version numbers and URLs to download the `.tar.gz` of the source) the other day but bottles were never created for these new versions.

one possible work around is to download a snapshot (the specific version of the formulae) defined within the formula file, and build from source for your specific version of macos. that is, if the source URLs in the formulae file are still valid.

...let me check real quick. ⏱

(well that wasn't quick 😖)

```
brew install freecad/freecad/coin@4.0.0 -s
brew install freecad/freecad/nglib -s
```

basically you can get around to the bottles not being present just by building the formulae from source using the `-s` flag.

i hope this helps.

unfortunately @vejmarie is the only person i know who can accept PRs and make admin changes to the repo. hopefully when he gets back from his vacations he'll add me as an admin to this repo so i can make these fixes.

sgrogan, 2021-08-02:
@ipatch 
I can merge pull requests.

ipatch, 2021-08-02:
@sgrogan

yeah give me a little bit, and i'll get one going. gotta hand AFK things for a minute two.

as for admin access to the repo, i don't suppose you can grant me admin access to this repo.

i'd like to publish a release to house the existing bottles that i gathered and the ones i built myself, so i can update the formulae files to have bottles for 3 major releases of macos.

also i could a little administrative tasks for the repo.

sundtek, 2021-08-04:
brew upgrade nglib -s
==> Upgrading 1 outdated package:
freecad/freecad/nglib v6.2.2007 -> 6.2.2101
==> Upgrading freecad/freecad/nglib
  v6.2.2007 -> 6.2.2101 

==> Cloning https://github.com/ngsolve/netgen.git
Updating /Users/markus/Library/Caches/Homebrew/nglib--git
==> Checking out tag v6.2.2101
HEAD is now at 5e489319 Merge branch 'fix_csg2d' into 'master'
HEAD is now at 5e489319 Merge branch 'fix_csg2d' into 'master'
Entering 'external_dependencies/pybind11'
/Users/markus/Library/Caches/Homebrew/nglib--git/external_dependencies/pybind11
==> cmake -DUSE_PYTHON=OFF -DUSE_GUI=OFF -DUSE_OCC=ON -DCMAKE_PREFIX_PATH="/usr/
==> make -j4 install
Last 15 lines from /Users/markus/Library/Logs/Homebrew/nglib/02.make:
[ 52%] Building CXX object libsrc/meshing/CMakeFiles/mesh.dir/__/general/mystring.cpp.o
[ 52%] Building CXX object libsrc/meshing/CMakeFiles/mesh.dir/__/general/optmem.cpp.o
[ 53%] Building CXX object libsrc/meshing/CMakeFiles/mesh.dir/__/general/parthreads.cpp.o
[ 54%] Building CXX object libsrc/meshing/CMakeFiles/mesh.dir/__/general/seti.cpp.o
[ 54%] Building CXX object libsrc/meshing/CMakeFiles/mesh.dir/__/general/sort.cpp.o
[ 55%] Building CXX object libsrc/meshing/CMakeFiles/mesh.dir/__/general/spbita2d.cpp.o
[ 55%] Building CXX object libsrc/meshing/CMakeFiles/mesh.dir/__/general/table.cpp.o
[ 56%] Building CXX object libsrc/meshing/CMakeFiles/mesh.dir/__/general/mpi_interface.cpp.o
[ 57%] Building CXX object libsrc/meshing/CMakeFiles/mesh.dir/__/general/gzstream.cpp.o
[ 57%] Linking CXX shared library libmesh.so
[ 57%] Built target mesh
make[3]: *** [all] Error 2
make[2]: *** [dependencies/src/netgen-stamp/netgen-build] Error 2
make[1]: *** [CMakeFiles/netgen.dir/all] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/freecad/homebrew-freecad/issues

These open issues may also help:
404 error on coin@4.0.0_1 and nglib@6.2.2101 bottles https://github.com/FreeCAD/homebrew-freecad/issues/187

compiling netgen manually does not throw any error though... so I'm still checking.

ipatch, 2021-08-04:
thanks for the updates. i'll spend some time later this afternoon looking into this. i'll report back and hopefully i can updated the affected formulae files. 🤞

ipatch, 2021-08-04:
i have a bigsur vm setup, so i'll get some bottles generated for big sur, but it'll take a few to get a catalina vm setup to get bottles going for it. so stay tuned please.

sundtek, 2021-08-04:
FreeCAD 0.20 works again here (Catalina)

sundtek, 2021-08-05:
well most of FreeCAD works but... I saw following in the freecad log:
10:09:34  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.9/site-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/Users/markus/devel/freecad/build/Mod/MeshPart/MeshFlatteningCommand.py", line 5, in <module>
    import MeshPartGui
  File "/usr/local/lib/python3.9/site-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ImportError'>: dlopen(/Users/markus/devel/freecad/build/Mod/MeshPart/MeshPartGui.so, 2): Library not loaded: /usr/local/opt/gcc/lib/gcc/10/libgfortran.5.dylib
  Referenced from: /usr/local/opt/med-file/lib/libmedfwrap.11.0.0.dylib
  Reason: image not found

----
/usr/local/opt/gcc/lib/gcc/11/libgfortran.5.dylib

----

==> Cloning https://code.qt.io/cgit/pyside/pyside-setup.git/tag/?h=v5.15.2
Cloning into '/Users/markus/Library/Caches/Homebrew/shiboken2--git'...
fatal: repository 'https://code.qt.io/cgit/pyside/pyside-setup.git/tag/?h=v5.15.2/' not found
Error: Failed to download resource "shiboken2"
Failure while executing; `git clone --branch 5.15.2 -c advice.detachedHead=false https://code.qt.io/cgit/pyside/pyside-setup.git/tag/\?h=v5.15.2 /Users/markus/Library/Caches/Homebrew/shiboken2--git` exited with 128. Here's the output:
Cloning into '/Users/markus/Library/Caches/Homebrew/shiboken2--git'...
fatal: repository 'https://code.qt.io/cgit/pyside/pyside-setup.git/tag/?h=v5.15.2/' not found

sundtek, 2021-08-05:
@ipatch can you fix the shiboken2 link?

dirkolbrich, 2021-08-05:
See https://github.com/FreeCAD/homebrew-freecad/issues/189#issuecomment-893358081

dirkolbrich, 2021-08-05:
URL issue on download links within brew setup vs hosted file -> https://github.com/FreeCAD/homebrew-freecad/issues/189#issuecomment-893381430

ipatch, 2021-08-05:
> @ipatch can you fix the shiboken2 link (see the backlog)?

@sundtek yeah i'll get to hear sometime today 🤞

first, i'm making sure all bottles are accessible via the release i published yesterday.

closed 2021-08-06 by dirkolbrich