FreeCAD/homebrew-freecad #189: migrate tap bottles to a github release

ipatch, 2021-08-04:
- [ ] migrate all bottles from existing TLD to a published github release.
- [ ] take existing bottles from my local fork and publish bottles on upstream repo
- [ ] fix/build bottles for _coin_ & _nglib_ on big sur
  - [ ] specific to me 🙋‍♂️, setup a catalina vm, so i can build a bottles of _coin_ & _nglib_ for catalina.
- [ ] after, bottles have been published in a github release, update formulae files to point to the github release.

ipatch, 2021-08-04:
pr #190 updates a majority of the formulae files within this repo to use the github release i published earlier today, and also the release houses the links for the bottles

- mojave
- catalina
- big sur _x86_64_ (presently do not have access to a m1 mac)

dirkolbrich, 2021-08-05:
As of 2021-08-05:12:44 CEST download of coin Big Sur bottle still returns 404 error:

```bash
➜  freecad-build brew install --only-dependencies freecad --with-packaging-utils
==> Installing freecad from freecad/freecad
==> Downloading https://ghcr.io/v2/homebrew/core/zstd/manifests/1.5.0
######################################################################## 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/zstd/blobs/sha256:eae17621cfc664d6e527a6d6aa6a000343eced0f60c81b4e2dd9a9aed7b79c3f
==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:eae17621cfc664d6e527a6d6aa6a000343eced0f60c81b4e2dd9a9aed7b79c3f?se=2021-08
######################################################################## 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/ccache/manifests/4.3_1
######################################################################## 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/ccache/blobs/sha256:083ebd7ddc08f386a9005df99d8c83f6f6b271068ae5b95fc548e4505ab5051b
==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:083ebd7ddc08f386a9005df99d8c83f6f6b271068ae5b95fc548e4505ab5051b?se=2021-08
######################################################################## 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/cmake/manifests/3.21.1-1
######################################################################## 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/cmake/blobs/sha256:4ffc49c18cfc6e55a1f61fd62657de2aed1c7debec7e676e5c0160849f391287
==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:4ffc49c18cfc6e55a1f61fd62657de2aed1c7debec7e676e5c0160849f391287?se=2021-08
######################################################################## 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/pcre/manifests/8.45
######################################################################## 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/pcre/blobs/sha256:fb2fefbe1232706a603a6b385fc37253e5aafaf3536cb68b828ad1940b95e601
==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:fb2fefbe1232706a603a6b385fc37253e5aafaf3536cb68b828ad1940b95e601?se=2021-08
######################################################################## 100.0%
==> Downloading https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/swig%404.0.2-4.0.2.big_sur.bottle.tar.gz
==> Downloading from https://github-releases.githubusercontent.com/18778796/296af0d4-d03f-4736-bb26-7d7879cf16c8?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credenti
######################################################################## 100.0%
==> Downloading https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/icu4c%4067.1-67.1.big_sur.bottle.tar.gz
==> Downloading from https://github-releases.githubusercontent.com/18778796/3d8b56c0-6198-4df2-b637-1a28ed4fa6b2?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credenti
######################################################################## 100.0%
==> Downloading https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/boost%401.75.0-1.75.0.big_sur.bottle.tar.gz
==> Downloading from https://github-releases.githubusercontent.com/18778796/c786d90b-04f9-4eb3-941e-2e3a7683f560?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credenti
######################################################################## 100.0%
==> Downloading https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/python3.9-3.9.0_5.big_sur.bottle.tar.gz
==> Downloading from https://github-releases.githubusercontent.com/18778796/78fec6d7-0e2c-4ac6-8073-0393d9645ebf?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credenti
######################################################################## 100.0%
==> Downloading https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/boost-python3%401.75.0-1.75.0.big_sur.bottle.tar.gz
==> Downloading from https://github-releases.githubusercontent.com/18778796/b70407f6-fd61-4ab8-80b2-896836f7bf2e?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credenti
######################################################################## 100.0%
==> Downloading https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/coin%404.0.0-4.0.0_1.big_sur.bottle.tar.gz

curl: (22) The requested URL returned error: 404
Error: Failed to download resource "coin@4.0.0"
Download failed: https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/coin%404.0.0-4.0.0_1.big_sur.bottle.tar.gz
➜  freecad-build date
Do  5 Aug 2021 12:44:04 CEST
```

dirkolbrich, 2021-08-05:
same 404 error holds for tbb bottle

```bash
==> Downloading https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/tbb%402020_u3-2020_U3_1.big_sur.bottle.1.tar.gz

curl: (22) The requested URL returned error: 404
Error: Failed to download resource "tbb@2020_u3"
Download failed: https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/tbb%402020_u3-2020_U3_1.big_sur.bottle.1.tar.gz
➜  freecad-build date
Do  5 Aug 2021 13:08:14 CEST
```

dirkolbrich, 2021-08-05:
There seems to be a URL issue:

The linked bottles under the Release page at https://github.com/FreeCAD/homebrew-freecad/releases/tag/07.28.2021 are downloadable. Some of these bottles have an additional `.1` before the file ending `.tar.gz`.

Link listed in brew
https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/coin%404.0.0-4.0.0_1.big_sur.bottle.tar.gz

Link listed on Release page:
https://github.com/FreeCAD/homebrew-freecad/releases/download/07.28.2021/coin@4.0.0-4.0.0_1.big_sur.bottle.1.tar.gz

This holds for several files. I think you should rename ALL linked bottles to remove the additional `.1` and unify the file name endings.

ipatch, 2021-08-05:
> There seems to be an URL issue:
> 
> The linked bottles under the Release page at [`07.28.2021` (release)](https://github.com/FreeCAD/homebrew-freecad/releases/tag/07.28.2021) are downloadable. Some of these bottles have an additional `.1` before the file ending `.tar.gz`, some not.
> 
> Link for coin@4.0.0 bottle listed in brew:
> [freecad/homebrew-freecad@`07.28.2021` coin%404.0.0-4.0.0_1.big_sur.bottle.tar.gz (download)](https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/coin%404.0.0-4.0.0_1.big_sur.bottle.tar.gz)
> 
> Link for coin@4.0.0 bottle listed on Release page:
> [`07.28.2021` coin@4.0.0-4.0.0_1.big_sur.bottle.1.tar.gz (download)](https://github.com/FreeCAD/homebrew-freecad/releases/download/07.28.2021/coin@4.0.0-4.0.0_1.big_sur.bottle.1.tar.gz)
> 
> Link for tbb@2020_u3 bottle listed in brew:
> [freecad/homebrew-freecad@`07.28.2021` tbb%402020_u3-2020_U3_1.big_sur.bottle.1.tar.gz (download)](https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/tbb%402020_u3-2020_U3_1.big_sur.bottle.1.tar.gz)
> 
> Link for tbb@2020_u3 bottle listed on Release page:
> [`07.28.2021` tbb@2020_u3-2020_U3_1.big_sur.bottle.tar.gz (download)](https://github.com/FreeCAD/homebrew-freecad/releases/download/07.28.2021/tbb@2020_u3-2020_U3_1.big_sur.bottle.tar.gz)
> 
> This holds for several files. I think you should rename ALL linked bottles to remove the additional `.1`, unify the file name endings and upload again.

yup, i started working on this last evening, and will try and have all the bottle names updated accordingly. i posted a question about bottle naming [here](https://github.com/Homebrew/discussions/discussions/1945), but haven't gotten a reply yet.

so i need to change `--` to `-` and remove the `.1` in the names.

stay tuned, i'm working on a `sed` expression to do a batch rename of the files.

ipatch, 2021-08-05:
just removed `brew remove ...` all formulae from tap on macos mojave. then reinstalled all formulae from tap, and it appears all bottles download and install on macos mojave. so a little progress on that front.

built a bottle for coin@4.0.0 for big sur. uploaded to the release i published, so doing

```
brew install freecad/freecad/coin@4.0.0 -v
```

should download and install coin@4.0.0 from the bottle if one so chooses, and building from source should still work if that's something you desire.

i'm building the bottle for `nglib` on my big sur vm right, and i'll get that uploaded to the release here in a few.

---

still need to setup a catalina vm to build nglib, coin bottles for catalina.

and i'll test, and make sure all formulae from the tap can be install via bottles if one so chooses.

ipatch, 2021-08-05:
most if not all **big sur** bottle issues should be resolved now.

onto catalina 🛶

dirkolbrich, 2021-08-06:
Downloading of bottles on Big Sur is fixed now. But at the end of the installation process, there is an `Error: SHA256 mismatch` for the `freecad/freecad/vtk@8.2.0` bottle:

```bash
➜  ~ date
Fr  6 Aug 2021 12:23:42 CEST
➜  ~ brew install --only-dependencies freecad --with-packaging-utils
==> Installing freecad from freecad/freecad
.
. # lots of downloads and installations
.
==> Pouring xerces-c--3.2.3.big_sur.bottle.2.tar.gz
🍺  /usr/local/Cellar/xerces-c/3.2.3: 1,684 files, 37MB
==> No outdated dependents to upgrade!
==> Checking for dependents of upgraded formulae...
==> Reinstalling 2 broken dependents from source:
freecad/freecad/med-file, freecad/freecad/vtk@8.2.0
==> Downloading http://files.salome-platform.org/Salome/other/med-4.0.0.tar.gz
==> Downloading from https://files.salome-platform.org/Salome/other/med-4.0.0.tar.gz
######################################################################## 100.0%
==> Reinstalling freecad/freecad/med-file
==> cmake . -DMEDFILE_BUILD_PYTHON=ON -DMEDFILE_BUILD_TESTS=OFF -DMEDFILE_INSTALL_DOC=OFF -DPYTHON_INCLUDE_DIR=/usr/local/opt/python3.9/Frameworks/Python.framew
==> make install
🍺  /usr/local/Cellar/med-file/4.0.0: 132 files, 11MB, built in 2 minutes 9 seconds
Warning: freecad/freecad/vtk@8.2.0 has been deprecated because it is a versioned formula!
==> Downloading https://gitlab.kitware.com/vtk/vtk/commit/ca3b5a50d945b6e65f0e764b3138cad17bd7eb8d.diff
==> Downloading from https://gitlab.kitware.com/vtk/vtk/-/commit/ca3b5a50d945b6e65f0e764b3138cad17bd7eb8d.diff
##O=#  #
Error: SHA256 mismatch
Expected: 49574bb914e2564b21ab0fb23cadcccd7dd323ae7f0f26f64fd6346c3db14cd7
  Actual: b9f7a3ebf3c29f3cad4327eb15844ac0ee849755b148b60fef006314de8e822e
    File: /Users/dirkolbrich/Library/Caches/Homebrew/downloads/551eba2e15e57a4491b3819395b9556196df6b9f2550e2b6bc4f19a101ce9d2c--ca3b5a50d945b6e65f0e764b3138cad17bd7eb8d.diff
To retry an incomplete download, remove the file above.
➜  ~
```

ipatch, 2021-08-06:
@dirkolbrich 

i ran into that sha256 mismatch issue myself. i uninstalled vtk and then reinstalled without any issues. not sure it was a diff that was left around from a prior installation. and i double check right now to make sure that it downloads and installs as it should. but basically i did the below

```
brew remove freecad/freecad/vtk@8.2.0 --verbose --ingore-dependencies
```

followed by

```
brew install freecad/freecad/vtk@8.2.0 --verbose
```

---

...

okay so those patches are specific to building vkt@8.2.0 from source. ie.

```
brew install freecad/freecad/vtk@8.2.0 -s
```

if installing vtk@8.2.0 from bottle then the sha256sum mismatches would output/(present themselves) during the install process, because the bottle URLs have been updated.

either way, #195 should fix the sha256sum mismatches for vtk.

almost done with migrating all these bottles, just need to upload the nglib and coin bottles for catalina, then i'll probably close this issue, and move on to the other open issues. 🥵

ipatch, 2021-08-07:
the below commit should have all the fixes for getting bottles working on

- big sur
- catalina
- mojave

if there any issues with the current batch of bottles please open a new issue.

thanks ipatch.

closed 2021-08-07 by ipatch

dirkolbrich, 2021-08-08:
Thank you for your effort. It seems to work all now.

A note: For the next iteration of updates maybe rename the download URL according to ISO-8601 date format YYYY-MM-DD.

So instead of:
```
https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021
```

better to use:
```
https://github.com/freecad/homebrew-freecad/releases/download/2021-07-28
```