FreeCAD/homebrew-freecad #196: Formula Elmer

ageeye, 2021-08-06:
Which opportunities afforded Elmer. How is it available in FreeCAD?

ipatch, 2021-08-06:
pretty sure it provides additional solvers for working with the fem workbench. however did a quick search through the cmake build flags and didn't see a mention of it. all i noticed were,

```
BUILD_FEM:BOOL=0
BUILD_FEM_NETGEN:BOOL=0
```

ageeye, 2021-08-07:
Thank you for your efforts.

ageeye, 2021-08-07:
`==> Cloning https://github.com/ElmerCSC/elmerfem.git
Updating /Users/bal/Library/Caches/Homebrew/elmer--git
==> Checking out revision c40a03f2d0c77fa66afa7a476e8981fa8b7d74ac
HEAD is now at c40a03f2 Better naming for cyclic restart
HEAD is now at c40a03f2 Better naming for cyclic restart
Entering 'contrib/Zoltan_v3.83'
/Users/bal/Library/Caches/Homebrew/elmer--git/contrib/Zoltan_v3.83
Error: Operation already in progress for python3.9.formula
Another active Homebrew process is already using python3.9.formula.
Please wait for it to finish or terminate it to continue.`

Hmm.  Sup?

ipatch, 2021-08-07:
yes the elmer formula file was checked in to the repo at some point, but was never constructed properly to actually build elmer. there's some notes on the official elmer github repo page for building elmer non gui and gui. there are also some youtube videos where the demonstrates how to build elmer.

if you want to update the formula file so elmer can build and install properly send PR through and i'll merge it in.

vejmarie, 2021-08-08:
hi @ipatch what are the error you see with the elmer formula, it works on my mac ;) and for some elmer people in singapor

ipatch, 2021-08-08:
@vejmarie 

```
╰─λ brew install freecad/freecad/elmer                                                                                                       
==> Installing elmer from freecad/freecad
==> Downloading https://downloads.sourceforge.net/project/qwt/qwt/6.1.6/qwt-6.1.6.tar.bz2
Already downloaded: /Users/capin/Library/Caches/Homebrew/downloads/96bf99f5d841cb803fef60ae94dfca5246db8e5cdb1044a3f3e7b942b6606799--qwt-6.1.6.tar.bz2
==> Cloning https://github.com/ElmerCSC/elmerfem.git
Updating /Users/capin/Library/Caches/Homebrew/elmer--git
==> Checking out revision c40a03f2d0c77fa66afa7a476e8981fa8b7d74ac
Note: switching to 'c40a03f2d0c77fa66afa7a476e8981fa8b7d74ac'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at c40a03f2 Better naming for cyclic restart
HEAD is now at c40a03f2 Better naming for cyclic restart
Entering 'contrib/Zoltan_v3.83'
Submodule path 'contrib/Zoltan_v3.83': checked out '0d6fc95372719b560bb2e5b250134a16837582ac'
/Users/capin/Library/Caches/Homebrew/elmer--git/contrib/Zoltan_v3.83
Error: Operation already in progress for python3.9.formula
Another active Homebrew process is already using python3.9.formula.
Please wait for it to finish or terminate it to continue.
```

ipatch, 2021-08-08:
i was messing around with the formula the other day, when i was working through all the style fixes, and this is what i came up with that would install. it's non gui version of the installation.

https://github.com/ipatch/homebrew-freecad/blob/aeb402769163372510e1780eddfac2901fbd10ff/Formula/elmer.rb

ageeye, 2021-08-08:
`cd ~/Library/Caches/Homebrew/elmer--git 
git submodule update --init
objc[1305]: Class AMSupportURLConnectionDelegate is implemented in both /usr/lib/libauthinstall.dylib (0x20a0e3ad8) and /System/Library/PrivateFrameworks/MobileDevice.framework/Versions/A/MobileDevice (0x113e242b8). One of the two will be used. Which one is undefined.
objc[1305]: Class AMSupportURLSession is implemented in both /usr/lib/libauthinstall.dylib (0x20a0e3b28) and /System/Library/PrivateFrameworks/MobileDevice.framework/Versions/A/MobileDevice (0x113e24308). One of the two will be used. Which one is undefined.`

Oh.

ipatch, 2021-12-10:
pr #274 should fix issues related to building elmer, qwtelmer, and opencamlib.

if there are any new issues or existing issues i have missed feel free to comment in this issue or even better open a new issue so we can begin a new discussion about elmer build / run related issues.

closed 2021-12-10 by ipatch