FreeCAD/homebrew-freecad #237: Problem with BOOST version installing `freecad --HEAD --with-macos-app` on BigSur

cowboyd, 2021-09-07:
Trying to build from source yields this error:

```
Error: Operation already in progress for boost@1.75.0.formula
Another active Homebrew process is already using boost@1.75.0.formula.
Please wait for it to finish or terminate it to continue.
/usr/local/Homebrew/Library/Homebrew/lock_file.rb:23:in `lock'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1308:in `lock'
/usr/local/Homebrew/Library/Homebrew/formula_installer.rb:1224:in `each'
/usr/local/Homebrew/Library/Homebrew/formula_installer.rb:1224:in `lock'
/usr/local/Homebrew/Library/Homebrew/formula_installer.rb:340:in `install'
/usr/local/Homebrew/Library/Homebrew/install.rb:303:in `install_formula'
/usr/local/Homebrew/Library/Homebrew/cmd/install.rb:207:in `block in install'
/usr/local/Homebrew/Library/Homebrew/cmd/install.rb:205:in `each'
/usr/local/Homebrew/Library/Homebrew/cmd/install.rb:205:in `install'
/usr/local/Homebrew/Library/Homebrew/brew.rb:129:in `<main>'
```

```
➜ brew install freecad --HEAD --with-macos-app
Warning: Treating freecad as a formula. For the cask, use homebrew/cask/freecad
==> Installing freecad from freecad/freecad
==> Downloading https://ghcr.io/v2/homebrew/core/pcre/manifests/8.45
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/4a3bba0e8d1899c13ac3442d2c49df6c1999948bdc0943f2f179693257d82545--pcre-8.45.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/pcre/blobs/sha256:fb2fefbe1232706a603a6b385fc37253e5aafaf3536cb68b828ad1940b95e601
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/36d8eda3485a3534637f69cd7daf30332ff4a5622746550092948587ccee36f3--pcre--8.45.big_sur.bottle.tar.gz
==> Downloading https://github.com/FreeCAD/homebrew-freecad/releases/download/swig@4.0.2-4.0.2/swig%404.0.2-4.0.2.big_sur.bottle.1.tar.gz
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/f0cddb30ae636bed933627b2cb6928e362a3c2813022d029beab7bdbb7852ccd--swig
==> Downloading https://ghcr.io/v2/homebrew/core/hiredis/manifests/1.0.0-1
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/04662c8441f94373705cb0c8ba67e1bc31a66bba0c6afc55f29775d80b158aed--hiredis-1.0.0-1.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/hiredis/blobs/sha256:b6938bbdfbc95f2fb3affb4bde281a7369b0b36cae2372f5a875edf2b67bc7f4
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/9c1fbd89cda5c533688ffef32b97caa18b66ca08715fc1fa0bfee3cef316c633--hiredis--1.0.0.big_sur.bottle.1.tar.gz
==> Downloading https://ghcr.io/v2/homebrew/core/zstd/manifests/1.5.0
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/3ef60ce0b7c777555d3eb0a208d8d3ffe1af2a217b203ca30a494924e13a24cf--zstd-1.5.0.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/zstd/blobs/sha256:eae17621cfc664d6e527a6d6aa6a000343eced0f60c81b4e2dd9a9aed7b79c3f
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/9b2357a4e28094ff3cc655920bf22feb86ec12816deb948b38714f5d95ede62a--zstd--1.5.0.big_sur.bottle.tar.gz
==> Downloading https://ghcr.io/v2/homebrew/core/ccache/manifests/4.4
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/235ac564c4dc46137fb6f49b7ce8982a2db6a0dbabfd39d863e191f555f322a9--ccache-4.4.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/ccache/blobs/sha256:01a6eb140be4b65cd4309e5248dea5cb715ffb790bada58a0b1bbca569660543
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/daae7776f623f10af67d96e94c51718bf420ac40d268c82a65dc9a82399d7633--ccache--4.4.big_sur.bottle.tar.gz
==> Downloading https://github.com/FreeCAD/homebrew-freecad/releases/download/icu4c@67.1-67.1_1/icu4c%4067.1-67.1_1.big_sur.bottle.tar.gz
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/064a432fe93c4887248c8c4d6597b7c3be4cf68096bbea3baad28d2a07e43a41--icu4c
==> Downloading https://github.com/FreeCAD/homebrew-freecad/releases/download/boost@1.75.0-1.75.0_1/boost%401.75.0-1.75.0_1.big_sur.bottl
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/4d06877280cb2cf7192da70b435aa789d8f341bf5cd4a8fde98b5cdad112195a--boost
==> Downloading https://ghcr.io/v2/homebrew/core/gdbm/manifests/1.20
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/c27d8af232565f7f16e912a20505b0d5ff83f6782835b9492840213cbf5fa5d2--gdbm-1.20.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/gdbm/blobs/sha256:ea88ce09e934407b1c7dfcc1b74e2d4f1b409f8264b4475b816369a129c6cd25
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/039ddb2682e5b93b3cd0ad30e130b5eab5bdb193e9e8d0239c9556958ccd992b--gdbm--1.20.big_sur.bottle.tar.gz
==> Downloading https://ghcr.io/v2/homebrew/core/openssl/1.1/manifests/1.1.1l
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/7cb540b15662ae88caf307947df0254e5c64c04933630f7e9543cb018107ecbe--openssl@1.1-1.1.1l.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/openssl/1.1/blobs/sha256:ff8b2a965c680b4d9baccd60e799d0989e7dc562d2ba81696a9996bab256a6a
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/b6c57e7f9acfe19e13efdfa925e9d77bbc52f6965e0f48326f64f73d18ef9c81--openssl@1.1--1.1.1l.big_sur.bottle.tar.gz
==> Downloading https://ghcr.io/v2/homebrew/core/sqlite/manifests/3.36.0
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/f624e7fc68307d01f25a03c15c0a84ff3b65d4d984d85616e4181a685efd0548--sqlite-3.36.0.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/sqlite/blobs/sha256:2c9c5f05c16c1fa972fc49c4b2705d7877b99640b6bb0b3908333e17f63dd71b
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/c27cd6f405f31a0171c05374d0dc9c4281f577e1822b52f5914cb2f40b647a7b--sqlite--3.36.0.big_sur.bottle.tar.gz
==> Downloading https://www.bytereef.org/contrib/decimal.diff
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/f60b5004541eb3c87cce87ef3bf94933a2684ab267346afdc45ae1622ffa923a--decimal.diff
==> Downloading https://files.pythonhosted.org/packages/88/fc/d17731c0cc67a5a8e385e4f47c3b0b186720e198b70f076ccb4676804a8f/setuptools-57.
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/812e9281c6415480b6e58ee84ec32b756f0996a312720020f8fff48c9c1ad384--setuptools-57.0.0.tar.gz
==> Downloading https://files.pythonhosted.org/packages/4d/0c/3b63fe024414a8a48661cf04f0993d4b2b8ef92daed45636474c018cd5b7/pip-21.1.3.tar
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/04d9aa9c912086512d114b9736d78433e9402068de3a794b6630905d46361407--pip-21.1.3.tar.gz
==> Downloading https://files.pythonhosted.org/packages/ed/46/e298a50dde405e1c202e316fa6a3015ff9288423661d7ea5e8f22f589071/wheel-0.36.2.t
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/2628149c388cda23d833947d9d838f27eba9442101b2958948fbe661e96b689b--wheel-0.36.2.tar.gz
==> Downloading https://www.python.org/ftp/python/3.9.6/Python-3.9.6.tar.xz
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/a36caa0287ce822bf44c0fded664315c44b391afaa4814ceb2bc57e2223b0029--Python-3.9.6.tar.xz
==> Downloading https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/boost-python3%401.75.0-1.75.0.big_sur.bottle.tar
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/6897977ad980ddcb68b20bef1cff555affca5ac8c26e72455b6e7c826464e006--boost-python3
==> Downloading https://github.com/FreeCAD/homebrew-freecad/releases/download/boost@1.75.0-1.75.0_1/boost%401.75.0-1.75.0_1.big_sur.bottl
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/4d06877280cb2cf7192da70b435aa789d8f341bf5cd4a8fde98b5cdad112195a--boost
[... 160 lines cut ...]
Already downloaded: /Users/cowboyd/Library/Caches/Homebrew/downloads/ad57114a9c4b64aa431ef4eea65db30044df215882c54f068afde89b3b4be863--xerces-c--3.2.3.big_sur.bottle.2.tar.gz
==> Cloning https://github.com/freecad/FreeCAD.git
Updating /Users/cowboyd/Library/Caches/Homebrew/freecad--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 1c8b803f9e Draft: Introduced Hatch command + object
Error: Operation already in progress for boost@1.75.0.formula
Another active Homebrew process is already using boost@1.75.0.formula.
Please wait for it to finish or terminate it to continue.
```

Here are the contents of the brew lockfiles. If I run `brew cleanup` in between to clear out the old lockfiles, it still happens.

```
[... cut ...]
ccache.formula.lock               icu4c@67.1.formula.lock           pcre.formula.lock                 xz.formula.lock
cmake.formula.lock                isl.formula.lock                  pkg-config.formula.lock           zstd.formula.lock
cython@0.29.21.formula.lock       libmpc.formula.lock               python@3.9.6.formula.lock
freecad.formula.lock              mpdecimal.formula.lock            python@3.9.formula.lock
gcc.formula.lock                  mpfr.formula.lock                 readline.formula.lock
```

bblacey, 2021-09-07:
Two observations:

1. In order to build from source, you have to pass the `--build-from-source` option to the `brew install` command
2. It seems that you may have had another `brew` session active in a separate terminal window or in the background.  You need to stop/kill that process first.  You can use `ps waux | grep brew` to display all brew processes in terminal

cowboyd, 2021-09-08:
Hi @bblacey, and thanks for the quick response!

It did occur to me that perhaps there was another brew process running that was getting in the way, and so I checked that there were no `brew` and no `ruby` processes running. I even went so far as to reboot my computer in case there was a process hanging out that I couldn't see. After taking all those measures (and running `brew cleanup`) in between, I still see the exact same error.

Here is the output with the the `--build-from-source` option:

```
brew install freecad --HEAD --with-macos-app --build-from-source --verbose --debug
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/freecad.rb
Warning: Treating freecad as a formula. For the cask, use homebrew/cask/freecad
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/./swig@4.0.2.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/pcre.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/ccache.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/asciidoctor.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/cmake.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/sphinx-doc.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/python@3.9.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/pkg-config.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/gdbm.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/mpdecimal.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/openssl@1.1.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/readline.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/sqlite.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/xz.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/hiredis.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/zstd.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/./boost-python3@1.75.0.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::TapLoader): loading /usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/numpy@1.19.4.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::TapLoader): loading /usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/cython@0.29.21.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::TapLoader): loading /usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/python@3.9.6.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/gcc.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/gmp.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/isl.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/libmpc.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/mpfr.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/openblas.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::TapLoader): loading /usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/boost@1.75.0.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::TapLoader): loading /usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/icu4c@67.1.rb
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/./boost@1.75.0.rb
[... cut ...]
/usr/local/Homebrew/Library/Homebrew/formula_installer.rb:340:in `install'
/usr/local/Homebrew/Library/Homebrew/install.rb:323:in `install_formula'
/usr/local/Homebrew/Library/Homebrew/install.rb:284:in `block in install_formulae'
/usr/local/Homebrew/Library/Homebrew/install.rb:283:in `each'
/usr/local/Homebrew/Library/Homebrew/install.rb:283:in `install_formulae'
/usr/local/Homebrew/Library/Homebrew/cmd/install.rb:205:in `install'
/usr/local/Homebrew/Library/Homebrew/brew.rb:129:in `<main>'
```

ES-Alexander, 2021-09-26:
Any luck on solving this @cowboyd?

I'm running into the same issue trying to follow the [Compile on MacOS](https://wiki.freecadweb.org/Compile_on_MacOS) instructions (also on Big Sur). I've also tried 
- `brew cleanup`
- restarting my computer
- checked for any running processes, and
- the options in [this stackoverflow post](https://stackoverflow.com/questions/39797078/another-active-homebrew-process-is-already-in-progress)
none of which solved the issue.

Given others have seemed to solve similar issues by removing homebrew's locks, I'm wondering if the installation formula is trying to install boost twice or something, and foiling itself by downloading it, locking it, then reaching that lock the second time around and deciding it can't continue because something else is already using it. Tried to test that hypothesis by repeatedly deleting `boost*` locks while it was doing the install, but no luck unfortunately.

```
user@device ~ %  brew install --only-dependencies freecad --with-packaging-utils                                                                                                   18:10:53
Warning: Treating freecad as a formula. For the cask, use homebrew/cask/freecad
==> Downloading https://github.com/FreeCAD/homebrew-freecad/releases/download/swig@4.0.2-4.0.2/swig%404.0.2-4.0.2.big_sur.bottle.1.tar.gz
Already downloaded: /Users/eliots/Library/Caches/Homebrew/downloads/f0cddb30ae636bed933627b2cb6928e362a3c2813022d029beab7bdbb7852ccd--swig
==> Downloading https://ghcr.io/v2/homebrew/core/hiredis/manifests/1.0.0-1
Already downloaded: /Users/eliots/Library/Caches/Homebrew/downloads/04662c8441f94373705cb0c8ba67e1bc31a66bba0c6afc55f29775d80b158aed--hiredis-1.0.0-1.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/hiredis/blobs/sha256:b6938bbdfbc95f2fb3affb4bde281a7369b0b36cae2372f5a875edf2b67bc7f4
Already downloaded: /Users/eliots/Library/Caches/Homebrew/downloads/9c1fbd89cda5c533688ffef32b97caa18b66ca08715fc1fa0bfee3cef316c633--hiredis--1.0.0.big_sur.bottle.1.tar.gz
==> Downloading https://ghcr.io/v2/homebrew/core/zstd/manifests/1.5.0
Already downloaded: /Users/eliots/Library/Caches/Homebrew/downloads/3ef60ce0b7c777555d3eb0a208d8d3ffe1af2a217b203ca30a494924e13a24cf--zstd-1.5.0.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/zstd/blobs/sha256:eae17621cfc664d6e527a6d6aa6a000343eced0f60c81b4e2dd9a9aed7b79c3f
Already downloaded: /Users/eliots/Library/Caches/Homebrew/downloads/9b2357a4e28094ff3cc655920bf22feb86ec12816deb948b38714f5d95ede62a--zstd--1.5.0.big_sur.bottle.tar.gz
==> Downloading https://ghcr.io/v2/homebrew/core/ccache/manifests/4.4.1
Already downloaded: /Users/eliots/Library/Caches/Homebrew/downloads/7a11a5b4690ad58b911ac38a5d3edd8557fcdbbfb340728b3f57882d7c239c10--ccache-4.4.1.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/ccache/blobs/sha256:56c36dccabd0d36973c1ade740b4d75abe7fe8755f96c2a03add7667548bbfde
Already downloaded: /Users/eliots/Library/Caches/Homebrew/downloads/d6663ebc0dc86c55d9f20d4f3a8ca0a422ad4c63ed9eb695ed79a1c7368bd83c--ccache--4.4.1.big_sur.bottle.tar.gz
==> Downloading https://ghcr.io/v2/homebrew/core/cmake/manifests/3.21.3
Already downloaded: /Users/eliots/Library/Caches/Homebrew/downloads/5359300c2a8427101649d279de6b300e5ef0b30153006a97ae7b4798dfa547bd--cmake-3.21.3.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/cmake/blobs/sha256:871469119eb1fc5b005971866925259b27960e2105935ad70e934002e429eaa9
Already downloaded: /Users/eliots/Library/Caches/Homebrew/downloads/335c59233e9139547fc3c6f78fa6dbf9e0be8846dbd11ed4bcb2b658e78f4f21--cmake--3.21.3.big_sur.bottle.tar.gz
==> Downloading https://github.com/FreeCAD/homebrew-freecad/releases/download/icu4c@67.1-67.1_1/icu4c%4067.1-67.1_1.big_sur.bottle.tar.gz
Already downloaded: /Users/eliots/Library/Caches/Homebrew/downloads/064a432fe93c4887248c8c4d6597b7c3be4cf68096bbea3baad28d2a07e43a41--icu4c
==> Downloading https://github.com/FreeCAD/homebrew-freecad/releases/download/boost@1.75.0-1.75.0_1/boost%401.75.0-1.75.0_1.big_sur.bottle.tar.gz
Already downloaded: /Users/eliots/Library/Caches/Homebrew/downloads/4d06877280cb2cf7192da70b435aa789d8f341bf5cd4a8fde98b5cdad112195a--boost
==> Downloading https://ghcr.io/v2/homebrew/core/gdbm/manifests/1.21
Already downloaded: /Users/eliots/Library/Caches/Homebrew/downloads/0727ed9109e25bee50343875ea6f8d618e4e03806bbe25064904b0528119c6be--gdbm-1.21.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/gdbm/blobs/sha256:7abb94a713a1e25f5fc4c5218067abf6ad09bbd68967c21f01d5d69f88308dd4
Already downloaded: /Users/eliots/Library/Caches/Homebrew/downloads/85e0fbdd144f59a1589db939a856fd8301b01eb6d8e23d953fe336dfd513ed6b--gdbm--1.21.big_sur.bottle.tar.gz
==> Downloading https://ghcr.io/v2/homebrew/core/openssl/1.1/manifests/1.1.1l
Already downloaded: /Users/eliots/Library/Caches/Homebrew/downloads/7cb540b15662ae88caf307947df0254e5c64c04933630f7e9543cb018107ecbe--openssl@1.1-1.1.1l.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/openssl/1.1/blobs/sha256:ff8b2a965c680b4d9baccd60e799d0989e7dc562d2ba81696a9996bab256a6ad
Already downloaded: /Users/eliots/Library/Caches/Homebrew/downloads/b6c57e7f9acfe19e13efdfa925e9d77bbc52f6965e0f48326f64f73d18ef9c81--openssl@1.1--1.1.1l.big_sur.bottle.tar.gz
==> Downloading https://www.bytereef.org/contrib/decimal.diff
[... cut ...]
==> Downloading https://ghcr.io/v2/homebrew/core/xerces-c/blobs/sha256:6a561a0f4175e7da6790b2beaedf185516a118116402a674a6b936d5c3236575
Already downloaded: /Users/eliots/Library/Caches/Homebrew/downloads/ad57114a9c4b64aa431ef4eea65db30044df215882c54f068afde89b3b4be863--xerces-c--3.2.3.big_sur.bottle.2.tar.gz
==> Installing freecad from freecad/freecad
Error: Operation already in progress for boost@1.75.0.formula
Another active Homebrew process is already using boost@1.75.0.formula.
Please wait for it to finish or terminate it to continue.
```

cowboyd, 2021-09-27:
@ES-Alexander Nope, no luck. I had to give up :'-(

ES-Alexander, 2021-09-27:
Damn. Thanks for getting back at least :-)
I was hoping to take a crack at fixing the Mac touchpad support, but can't exactly do that if I can't compile.

Hopefully someone else is able to make some recommendations to fix the dependency issues. Seems a shame to be thwarted by a build error before even getting to try making changes.

ES-Alexander, 2021-09-27:
@ipatch excuse the intrusion, you've been super active with homebrew-freecad PRs lately - any ideas on how to get around this build issue? :-)

ipatch, 2021-09-27:
@ES-Alexander 

yeah i don't mind. i'll see what i can come up with. if i find / fix anything i'll post back with another comment.

ipatch, 2021-09-27:
so i had  sometime today to mess around with issue. i setup a completely new user on macos big sur vm.

`$USER` = `microbrewer` and setup homebrew within the `$USER` dir, ie. `/Users/microbrewer/homebrew`

and ran into the below issue with the following command,

```
brew install freecad/freecad/freecad --verbose --ignore-dependencies
```

```

100  7322  100  7322    0     0  18919      0 --:--:-- --:--:-- --:--:-- 18919
==> Downloading https://ghcr.io/v2/homebrew/core/xerces-c/blobs/sha256:6a561a0f4175e7da6790b2beaedf185516a118116402a674a6b936d5c3236575
==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:6a561a0f4175e7da6790b2beaedf185516a118116402a674a6b936d5c3236575?se=2021-09-27T17%3A30%3A00Z&sig=ZLFqHEY7zEdHLhxyMAe4VtHK%2Buft%2BWjiMjZB4HsOMgg%3D&sp=r&spr=https&sr=b&sv=2019-12-12
/usr/bin/curl --disable --cookie-jar /dev/null --globoff --show-error --user-agent Homebrew/3.2.13-66-gfef4ba0\ \(Macintosh\;\ Intel\ Mac\ OS\ X\ 11.5.2\)\ curl/7.64.1 --header Accept-Language:\ en --fail --retry 3 --location --remote-time --output /Users/microbrewer/Library/Caches/Homebrew/downloads/ad57114a9c4b64aa431ef4eea65db30044df215882c54f068afde89b3b4be863--xerces-c--3.2.3.big_sur.bottle.2.tar.gz.incomplete https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:6a561a0f4175e7da6790b2beaedf185516a118116402a674a6b936d5c3236575\?se=2021-09-27T17\%3A30\%3A00Z\&sig=ZLFqHEY7zEdHLhxyMAe4VtHK\%2Buft\%2BWjiMjZB4HsOMgg\%3D\&sp=r\&spr=https\&sr=b\&sv=2019-12-12
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 6354k  100 6354k    0     0  9303k      0 --:--:-- --:--:-- --:--:-- 9303k
==> Verifying checksum for 'ad57114a9c4b64aa431ef4eea65db30044df215882c54f068afde89b3b4be863--xerces-c--3.2.3.big_sur.bottle.2.tar.gz'
==> Installing freecad from freecad/freecad
Error: Operation already in progress for boost@1.75.0.formula
Another active Homebrew process is already using boost@1.75.0.formula.
Please wait for it to finish or terminate it to continue.
```

i'm able to get around this error by manually specifying boost and boost-python, ie. running the below commands,

```
brew install freecad/freecad/boost-python3@1.75.0;
brew install freecad/freecad/boost@1.75.0;
```

both of those installed on my macos big sur vm.

i'm the process of installing the additional dependencies.

IMHO, i'd manually install freecad from a `git clone` instead using `brew install` ie. you'll have more granular control of adding/editing/checking/testing the freecad source as opposed to install freecad from this repo at the time being.

however, this repo is a good source for installing all the dependencies for freecad and using `cmake, make, make install` to actually build and install freecad from a `git clone`.

ES-Alexander, 2021-09-28:
>i'm able to get around this error by manually specifying boost and boost-python, ie. running the below commands,
>```
>brew install freecad/freecad/boost-python3@1.75.0;
>brew install freecad/freecad/boost@1.75.0;
> ```
>both of those installed on my macos big sur vm.

@ipatch Unfortunately didn't solve the issue for me, although it did seem to install `boost-python3@1.75.0`. I'm intrigued that you're using `--ignore-dependencies` (which I get a warning about as "an unsupported Homebrew developer flag"), whereas the command from the macos compile docs uses `--only-dependencies`, which I believe is intended to install just the dependencies so that freecad can then be built from source (as you also suggested).

It may be relevant that running your `brew install freecad/freecad/freecad ...` command completes successfully for me (although I did only try it after running the `boost[-python3]@1.75.0` installs), but the command I'm trying to run to install the dependencies isn't working. That's also kind of odd though, because presumably installing only the dependencies shouldn't be trying to install freecad itself, but it does seem to be (maybe there's something wrong with the dependencies installation). I expect your workaround likely fixes @cowboyd's original issue though, since they were actually trying to build/install freecad itself.

cowboyd, 2021-09-28:
Thank you for posting the workaround, but sadly, though both packages installed successfully, it did not work.

ipatch, 2021-09-28:
@ES-Alexander 

> Unfortunately didn't solve the issue for me, although it did seem to install boost-python3@1.75.0. I'm intrigued that you're using --ignore-dependencies (which I get a warning about as "an [...]

yeah i had my wires crossed when i typed that out. thanks for catching that.

stepheneb, 2021-10-20:
Having the same problem:

```
$ brew cleanup
$ ps waux | grep brew
stephen          54298   0.0  0.0  4268424    728 s000  S+   11:57PM   0:00.00 grep brew

$ brew install --only-dependencies freecad --with-packaging-utils
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/cask).
==> Updated Casks
Updated 6 casks.

Warning: Treating freecad as a formula. For the cask, use homebrew/cask/freecad
==> Downloading https://github.com/FreeCAD/homebrew-freecad/releases/download/swig@4.0.2-4.0.2/swig%404.0.2-4.0.2.big_sur.bottle.2.tar.g
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/101d36d7aedea3e825b628eea6bfe754aa501209f4b05bf9db901b08589f6f9f--swig
==> Downloading https://github.com/FreeCAD/homebrew-freecad/releases/download/icu4c@67.1-67.1_1/icu4c%4067.1-67.1_1.big_sur.bottle.tar.g
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/064a432fe93c4887248c8c4d6597b7c3be4cf68096bbea3baad28d2a07e43a41--icu4c
==> Downloading https://github.com/FreeCAD/homebrew-freecad/releases/download/boost@1.75.0-1.75.0_1/boost%401.75.0-1.75.0_1.big_sur.bott
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/4d06877280cb2cf7192da70b435aa789d8f341bf5cd4a8fde98b5cdad112195a--boost
==> Downloading https://www.bytereef.org/contrib/decimal.diff
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/f60b5004541eb3c87cce87ef3bf94933a2684ab267346afdc45ae1622ffa923a--decimal.diff
==> Downloading https://files.pythonhosted.org/packages/88/fc/d17731c0cc67a5a8e385e4f47c3b0b186720e198b70f076ccb4676804a8f/setuptools-57
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/812e9281c6415480b6e58ee84ec32b756f0996a312720020f8fff48c9c1ad384--setuptools-57.0.0.tar.gz
==> Downloading https://files.pythonhosted.org/packages/4d/0c/3b63fe024414a8a48661cf04f0993d4b2b8ef92daed45636474c018cd5b7/pip-21.1.3.ta
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/04d9aa9c912086512d114b9736d78433e9402068de3a794b6630905d46361407--pip-21.1.3.tar.gz
==> Downloading https://files.pythonhosted.org/packages/ed/46/e298a50dde405e1c202e316fa6a3015ff9288423661d7ea5e8f22f589071/wheel-0.36.2.
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/2628149c388cda23d833947d9d838f27eba9442101b2958948fbe661e96b689b--wheel-0.36.2.tar.gz
==> Downloading https://www.python.org/ftp/python/3.9.6/Python-3.9.6.tar.xz
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/a36caa0287ce822bf44c0fded664315c44b391afaa4814ceb2bc57e2223b0029--Python-3.9.6.tar.xz
==> Downloading https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/boost-python3%401.75.0-1.75.0.big_sur.bottle.ta
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/6897977ad980ddcb68b20bef1cff555affca5ac8c26e72455b6e7c826464e006--boost-python3
==> Downloading https://github.com/FreeCAD/homebrew-freecad/releases/download/boost@1.75.0-1.75.0_1/boost%401.75.0-1.75.0_1.big_sur.bott
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/4d06877280cb2cf7192da70b435aa789d8f341bf5cd4a8fde98b5cdad112195a--boost
==> Downloading https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/coin%404.0.0-4.0.0_1.big_sur.bottle.tar.gz
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/3a8973d101eb977b3035547d48b9f3fe36d2e9474a13083fdec14eeca7074a79--coin
==> Downloading https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/numpy%401.19.4-1.19.4.big_sur.bottle.tar.gz
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/dc966050fb35a02c1da8d7d73398f3093e097e12a03acb23bbd6d0cdf549a928--numpy
==> Downloading https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/matplotlib-2.1.1.big_sur.bottle.tar.gz
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/ff9fcd7717286b824932cbc903dca3819eb2c661e30317561d0e506dbd0a41bb--matplotlib-2.1.1.big_sur.bottle.tar.gz
==> Downloading https://ghcr.io/v2/homebrew/core/hdf5/1.10/manifests/1.10.7_1
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/1d2e90ecc8aae0d267ac3032dc0e4f90e8f1266618a7febb3ce31181f61bf99c--hdf5@1.10-1.10.7_1.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/hdf5/1.10/blobs/sha256:aaf4050cbc4a4dea9bf6fae1c54ce4f0a93537eaf3b45f5ed9041f6771e04583
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/c7d8df103afb3f3c8d4f5dc1d48e4609c9608a5d2280d9b5378dfeba206d1260--hdf5@1.10--1.10.7_1.big_sur.bottle.tar.gz
==> Downloading https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/med-file-4.0.0.big_sur.bottle.tar.gz
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/6f3ec26327bfacdc71b54c250fd6bfa1abead97c5c58c4fc32870a0874674454--med-file-4.0.0.big_sur.bottle.tar.gz
==> Downloading https://github.com/freecad/homebrew-freecad/releases/download/07.28.2021/tbb%402020_u3-2020_U3_1.big_sur.bottle.tar.gz
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/48f6634230fc0ea3e271ef3a17bacfd565782557aa0b8f0872fcbb09d1293d0f--tbb
==> Downloading https://ghcr.io/v2/homebrew/core/freeimage/manifests/3.18.0-3
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/8899777a9da0a17f96d97c1968a0f1a2324e8f83f3917230905395c36bda9491--freeimage-3.18.0-3.bottle_manifest.json
[... cut ...]
==> Downloading https://ghcr.io/v2/homebrew/core/orocos-kdl/blobs/sha256:1c7f7a1fc384bd107acd0d3b8d302beed12c89198617f19b2c74d89e437e3ee
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/d1f60f6bd6001829d0d090005dd2b17f981d4adbaf4fcfeba264da9952cff37e--orocos-kdl--1.5.1.big_sur.bottle.tar.gz
==> Installing freecad from freecad/freecad
Error: Operation already in progress for boost@1.75.0.formula
Another active Homebrew process is already using boost@1.75.0.formula.
Please wait for it to finish or terminate it to continue.
```

stepheneb, 2021-11-02:
I'm not getting the `Operation already in progress` error anymore.

```
$ brew install freecad --HEAD --with-macos-app --build-from-source --verbose --debug
```

There error appears to be in this linker step:

```
[ 56%] Linking CXX shared library ../../../../Mod/Part/Part.so
```
Which seems to perhaps be related to: `fatal error: 'GL/gl.h' file not found`:

I only see one instance of `GL/gl.h` on my system.

```
$ locate GL/gl.h
/opt/X11/include/GL/gl.h
```
file: freecad/Build/CMakeFiles/CMakeError.log
```
$ cat ~/Library/Logs/Homebrew/freecad/Build/CMakeFiles/CMakeError.log 
Determining if the include file GL/gl.h exists failed with the following output:
Change Dir: /tmp/freecad-20211102-5104-1d8whyz/Build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/local/Homebrew/Library/Homebrew/shims/mac/super/gmake -f Makefile cmTC_03b51/fast && /Applications/Xcode.app/Contents/Developer/usr/bin/make  -f CMakeFiles/cmTC_03b51.dir/build.make CMakeFiles/cmTC_03b51.dir/build
Building C object CMakeFiles/cmTC_03b51.dir/CheckIncludeFile.c.o
/usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang   -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.sdk -MD -MT CMakeFiles/cmTC_03b51.dir/CheckIncludeFile.c.o -MF CMakeFiles/cmTC_03b51.dir/CheckIncludeFile.c.o.d -o CMakeFiles/cmTC_03b51.dir/CheckIncludeFile.c.o -c /tmp/freecad-20211102-5104-1d8whyz/Build/CMakeFiles/CMakeTmp/CheckIncludeFile.c
/tmp/freecad-20211102-5104-1d8whyz/Build/CMakeFiles/CMakeTmp/CheckIncludeFile.c:1:10: fatal error: 'GL/gl.h' file not found
#include <GL/gl.h>
         ^~~~~~~~~
1 error generated.
make[1]: *** [CMakeFiles/cmTC_03b51.dir/CheckIncludeFile.c.o] Error 1
make: *** [cmTC_03b51/fast] Error 2
```

More detail from the original log:
```
[ 56%] Linking CXX shared library ../../../../Mod/Part/Part.so
cd /tmp/freecad-20211102-5104-1d8whyz/Build/src/Mod/Part/App && /usr/local/Cellar/cmake/3.21.4/bin/cmake -E cmake_link_script CMakeFiles/Part.dir/link.txt --verbose=1
/usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -Wall -Wextra -Wpedantic -Wno-write-strings  -Wno-undefined-var-template -O3 -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.sdk -dynamiclib -Wl,-headerpad_max_install_names  -Wl,-undefined,dynamic_lookup -o ../../../../Mod/Part/Part.so -install_name @rpath/Part.so CMakeFiles/Part.dir/Part_autogen/mocs_compilation.cpp.o CMakeFiles/Part.dir/FeaturePartBoolean.cpp.o CMakeFiles/Part.dir/FeaturePartBox.cpp.o CMakeFiles/Part.dir/FeaturePartCircle.cpp.o CMakeFiles/Part.dir/FeaturePartCommon.cpp.o CMakeFiles/Part.dir/FeaturePartCurveNet.cpp.o CMakeFiles/Part.dir/FeaturePartCut.cpp.o CMakeFiles/Part.dir/FeaturePartFuse.cpp.o CMakeFiles/Part.dir/FeaturePartImportBrep.cpp.o CMakeFiles/Part.dir/FeaturePartImportIges.cpp.o CMakeFiles/Part.dir/FeaturePartImportStep.cpp.o CMakeFiles/Part.dir/FeaturePartPolygon.cpp.o CMakeFiles/Part.dir/FeaturePartSection.cpp.o CMakeFiles/Part.dir/FeaturePartSpline.cpp.o CMakeFiles/Part.dir/FeatureChamfer.cpp.o CMakeFiles/Part.dir/FeatureCompound.cpp.o CMakeFiles/Part.dir/FeatureExtrusion.cpp.o CMakeFiles/Part.dir/FeatureFace.cpp.o CMakeFiles/Part.dir/FeatureFillet.cpp.o CMakeFiles/Part.dir/FeatureMirroring.cpp.o CMakeFiles/Part.dir/FeatureRevolution.cpp.o CMakeFiles/Part.dir/FeatureOffset.cpp.o CMakeFiles/Part.dir/PartFeatures.cpp.o CMakeFiles/Part.dir/PartFeature.cpp.o CMakeFiles/Part.dir/PartFeatureReference.cpp.o CMakeFiles/Part.dir/Part2DObject.cpp.o CMakeFiles/Part.dir/PrimitiveFeature.cpp.o CMakeFiles/Part.dir/FeatureGeometrySet.cpp.o CMakeFiles/Part.dir/CustomFeature.cpp.o CMakeFiles/Part.dir/BodyBase.cpp.o CMakeFiles/Part.dir/DatumFeature.cpp.o CMakeFiles/Part.dir/AttachExtension.cpp.o CMakeFiles/Part.dir/PrismExtension.cpp.o CMakeFiles/Part.dir/PropertyTopoShape.cpp.o CMakeFiles/Part.dir/PropertyGeometryList.cpp.o CMakeFiles/Part.dir/ArcPyImp.cpp.o CMakeFiles/Part.dir/ArcOfConicPyImp.cpp.o CMakeFiles/Part.dir/ArcOfCirclePyImp.cpp.o CMakeFiles/Part.dir/ArcOfParabolaPyImp.cpp.o CMakeFiles/Part.dir/BodyBasePyImp.cpp.o CMakeFiles/Part.dir/ConicPyImp.cpp.o CMakeFiles/Part.dir/CirclePyImp.cpp.o CMakeFiles/Part.dir/ArcOfEllipsePyImp.cpp.o CMakeFiles/Part.dir/EllipsePyImp.cpp.o CMakeFiles/Part.dir/HyperbolaPyImp.cpp.o CMakeFiles/Part.dir/ArcOfHyperbolaPyImp.cpp.o CMakeFiles/Part.dir/ParabolaPyImp.cpp.o CMakeFiles/Part.dir/OffsetCurvePyImp.cpp.o CMakeFiles/Part.dir/GeometryPyImp.cpp.o CMakeFiles/Part.dir/GeometryExtensionPyImp.cpp.o CMakeFiles/Part.dir/GeometryIntExtensionPyImp.cpp.o CMakeFiles/Part.dir/GeometryStringExtensionPyImp.cpp.o CMakeFiles/Part.dir/GeometryBoolExtensionPyImp.cpp.o CMakeFiles/Part.dir/GeometryDoubleExtensionPyImp.cpp.o CMakeFiles/Part.dir/GeometryCurvePyImp.cpp.o CMakeFiles/Part.dir/BoundedCurvePyImp.cpp.o CMakeFiles/Part.dir/TrimmedCurvePyImp.cpp.o CMakeFiles/Part.dir/GeometrySurfacePyImp.cpp.o CMakeFiles/Part.dir/LinePyImp.cpp.o CMakeFiles/Part.dir/LineSegmentPyImp.cpp.o CMakeFiles/Part.dir/PointPyImp.cpp.o CMakeFiles/Part.dir/BezierCurvePyImp.cpp.o CMakeFiles/Part.dir/BSplineCurvePyImp.cpp.o CMakeFiles/Part.dir/PlanePyImp.cpp.o CMakeFiles/Part.dir/ConePyImp.cpp.o CMakeFiles/Part.dir/CylinderPyImp.cpp.o CMakeFiles/Part.dir/SpherePyImp.cpp.o CMakeFiles/Part.dir/ToroidPyImp.cpp.o CMakeFiles/Part.dir/BezierSurfacePyImp.cpp.o CMakeFiles/Part.dir/BSplineSurfacePyImp.cpp.o CMakeFiles/Part.dir/OffsetSurfacePyImp.cpp.o CMakeFiles/Part.dir/PlateSurfacePyImp.cpp.o CMakeFiles/Part.dir/RectangularTrimmedSurfacePyImp.cpp.o CMakeFiles/Part.dir/SurfaceO [... cut ...]
-- SPNAV:               [/usr/local/lib/libspnav.dylib] [/usr/local/include]
-- Matplotlib:          3.3.3
-- Rift:                not enabled (BUILD_VR)
-- Doxygen:             1.9.2
--  Language:           English
--  Coin3D_DOC:         not found
=================================================
Now run 'cmake --build /tmp/freecad-20211102-5104-1d8whyz/Build' to build FreeCAD
=================================================

-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/freecad-20211102-5104-1d8whyz/Build
```

jbaehr, 2021-11-09:
> I'm not getting the `Operation already in progress` error anymore.

@stepheneb How did you get around this? I'm still facing this error when trying to setup a freecad build environment (nb: on macos 10.14; I know, unsupported by homebrew, but the issue seems to appear on more recent systems too).

@ipatch commands for installing boost-python manually worked flawless, but then (even after `brew cleanup`), I have the same symptom as @cowboyd and @ES-Alexander:
```
brew install --only-dependencies freecad --with-packaging-utils
...
Error: Operation already in progress for boost@1.75.0.formula
Another active Homebrew process is already using boost@1.75.0.formula.
Please wait for it to finish or terminate it to continue.
```
-------
@stepheneb regarding the GL error: There has been a discussion in the forum regarding Apple not listing  recent Macs in the list of OpenGL-supported devices. I don't have the thread at hand, you may want to try a search there. IIRC the libs are still there, but not the headers. That could explain your issue.

ipatch, 2021-11-09:
@jbaehr

presently i'm building freecad using a git clone of the freecad source. and am using homebrew to install the deps required for freecad to compile, but i do not install freecad using a 

```
brew install freecad
```

i'd like to get the freecad formula fixed up so that it can be installed with above such command and create the drag and drop app bundle.

i've worked on some fixes for the formula locally but haven't fixed everything yet unfortunately so thus i haven't submitted a PR to make `brew install freecad` work flawlessly yet. (at least on my system i havent).

neurodigitaltechnologies, 2021-11-10:
Hi @vidmik and @ipatch

I modified all .rb Formulas to switch from local versions of the formulas AKA (./$Formula) to home-core or "vanilla" versions. We must do this change recursively across all dependencies.

**For example:**
In Freecad.rb, changing ".boost@1.7.5" with "boost" will use the untouched and latest version from the core. 
This eventually works till we face nglib. It's not compiling due to a miss OCC Lib included in the CmakeList.txt I cannot solve. Anyone else is reaching this point?

Kindest!
Luis.

vidmik, 2021-11-10:
Disclaimer: I'm still learning all the homebrew concepts etc. so excuse me if I butcher the terminology or sound confused (I am).

I started doing pretty much exactly that - use the homebrew-core versions - and ended up on a tangent trying to get gcc to work. The reason, I think, is that I decided to perform the experiment in a separate, non-standard homebrew prefix/directory (*not* `/opt/homebrew`), which means the bottles cannot be used and the dependencies need to be rebuilt, including gcc (for numpy, I think). 

Getting gcc to work turns out to be challenging (while promising, it's clearly still work in progress), so for now I'll consider that a dead end and my next experiment would likely be to go back to using the standard homebrew prefix to see if that means skipping some builds, leveraging the bottles instead.

vidmik, 2021-11-10:
In the spirit of not really knowing what I'm doing - try this:

```
diff --git a/Formula/nglib.rb b/Formula/nglib.rb
index dd58ed7..67edc79 100644
--- a/Formula/nglib.rb
+++ b/Formula/nglib.rb
@@ -15,16 +15,16 @@ class Nglib < Formula
   end

   depends_on "cmake" => :build
-  depends_on "./opencascade@7.5.0"
+  depends_on "opencascade"

   def install
-    cmake_prefix_path = Formula["#{@tap}/opencascade@7.5.0"].opt_prefix + "/lib/cmake;"
-
+    cmake_prefix_path = Formula["opencascade"].opt_prefix + "/lib/cmake;"
     args = std_cmake_args + %W[
       -DUSE_PYTHON=OFF
       -DUSE_GUI=OFF
       -DUSE_OCC=ON
       -DCMAKE_PREFIX_PATH=#{cmake_prefix_path}
+      -DOCC_INCLUDE_DIR=#{Formula["opencascade"].opt_prefix}/include/opencascade
     ]

     mkdir "Build" do
```

vidmik, 2021-11-10:
With that patch applied to `nglib.rb`, and all the other changes to the Formulae I then ran into https://tracker.dev.opencascade.org/view.php?id=32328 which I worked around by simply including the additional header file:

```
--- ShapeUpgrade_UnifySameDomain.hxx.old	2021-11-10 15:36:27.000000000 -0800
+++ ShapeUpgrade_UnifySameDomain.hxx.new	2021-11-10 15:36:34.000000000 -0800
@@ -21,6 +21,7 @@
 #include <Standard.hxx>
 #include <Standard_Type.hxx>

+#include <TopoDS_Edge.hxx>
 #include <TopoDS_Shape.hxx>
 #include <Standard_Boolean.hxx>
 #include <Standard_Transient.hxx>
```

With that fix applied the freecad build completed successfully and when I run `/opt/homebrew/bin/FreeCAD` the GUI starts up, but there are some `No module named "PySide2"` errors, which could well be caused by my changes to the various Formulae.

vidmik, 2021-11-10:
Running `FreeCad -P /opt/homebrew/Cellar/pyside@2/5.15.2_1/lib/python3.9/site-packages` starts up the UI successfully and everything seems to be working. That said, I've never used freecad before so I'm not sure I would know if it works as expected or not :)

neurodigitaltechnologies, 2021-11-11:
Hi @vidmik!
Wow, many thanks for the investigation.
How do you apply the patches? I mean, you out all of this in the terminal? You create a .diff file? On that case, how do you apply it?
Regarding pyside2, check you hadn’t pysideat2 or puside@2 in formulas.
Kindest!

vidmik, 2021-11-11:
For the `ShapeUpgrade_UnifySameDomain.hxx` change I just opened the file - `/opt/homebrew//Cellar/opencascade/7.5.3/include/opencascade/ShapeUpgrade_UnifySameDomain.hxx` in my case - in my favorite editor and added that `#include` line. The Right(tm) way to fix it would probably be to either get a newer version of the opencascade source where that issue has been fixed, or apply a patch as part of the Formulae.

Been playing around with FreeCAD for ~30 minutes now and AFAICT everything seems to be working. I'm naively assuming that's good sign, and that this is all "just" a matter of updating all the dependency Formulae in homebrew-freecad to be more up-to-date with upstream.

neurodigitaltechnologies, 2021-11-11:
Thanks! I was asking about the patches or diff files for nglib, etc. you put the content of the diff but; how do you apply it? What’s the step by step to apply these patches?
Many thanks in advance.

vidmik, 2021-11-11:
Oh, that's also something I just changed directly in the homebrew-freecad tap file using an editor. Open up `/opt/homebrew/Library/Taps/freecad/homebrew-freecad/Formula/nglib.rb` and make those changes manually. Alternatively, you can store the patch in a file, say `~/nglib.rb.patch`, and do something like:

```# cd /opt/homebrew/Library/Taps/freecad/homebrew-freecad
# patch -p1 < ~/nglib.rb.patch
```

stepheneb, 2021-11-11:
@jbaehr,

Am now trying to build FreeCAD from source using the homebrew freecad formula to install dependencies.

Documented and more complete results here:

Building FreeCAD from source using homebrew freecad formula to install dependencies
https://gist.github.com/stepheneb/9bb11a53923df9795de236d438a3c155

Am stuck with errors building src/Base/CMakeFiles/FreeCADBase.dir/Builder3D.cpp.o

```
[  7%] Building CXX object src/Base/CMakeFiles/FreeCADBase.dir/Builder3D.cpp.o
In file included from /Users/stephen/dev/freecad/FreeCAD/src/Base/Builder3D.cpp:39:
In file included from /Users/stephen/dev/freecad/FreeCAD/src/Base/Tools.h:37:
In file included from /usr/local/opt/qt@5/lib/QtCore.framework/Headers/QString:1:
/usr/local/opt/qt@5/lib/QtCore.framework/Headers/qstring.h:78:7: error: definition of type 'QStringList' conflicts with type alias of the same name
class QStringList;
      ^
/usr/local/include/QtCore/qcontainerfwd.h:64:7: note: 'QStringList' declared here
using QStringList = QList<QString>;
      ^
In file included from /Users/stephen/dev/freecad/FreeCAD/src/Base/Builder3D.cpp:39:
In file included from /Users/stephen/dev/freecad/FreeCAD/src/Base/Tools.h:37:
In file included from /usr/local/opt/qt@5/lib/QtCore.framework/Headers/QString:1:
/usr/local/opt/qt@5/lib/QtCore.framework/Headers/qstring.h:81:29: error: redefinition of 'QVector' as different kind of symbol
template <typename T> class QVector;
                            ^
/usr/local/include/QtCore/qcontainerfwd.h:63:22: note: previous definition is here
template<typename T> using QVector = QList<T>;
                     ^
In file included from /Users/stephen/dev/freecad/FreeCAD/src/Base/Builder3D.cpp:39:
In file included from /Users/stephen/dev/freecad/FreeCAD/src/Base/Tools.h:37:
In file included from /usr/local/opt/qt@5/lib/QtCore.framework/Headers/QString:1:
/usr/local/opt/qt@5/lib/QtCore.framework/Headers/qstring.h:261:13: error: unknown type name 'QStringData'
    typedef QStringData Data;
            ^
/usr/local/opt/qt@5/lib/QtCore.framework/Headers/qstring.h:967:37: error: unknown type name 'QStringDataPtr'
    Q_DECL_CONSTEXPR inline QString(QStringDataPtr dd) : d(dd.ptr) {}
                                    ^
```

ipatch, 2021-11-11:
@vidmik 

i believe you can test most features of your build using the below steps (probably should add this to the README)

you can `cd` into the directory where you build FreeCAD depending if you built freecad as a mac app bundle or not, but give the below command a try

```
cd /path/to/FreeCAD
./FreeCAD -t 0
```

it should take a minute to run through the various tests. you can see some output that it show in the below two links i've created.

https://gist.github.com/ipatch/f26086fef67badf1db62c293bbf9579a

hope that helps.

vidmik, 2021-11-11:
Looks like the vast majority of the tests pass:

```
Ran 581 tests in 49.901s

FAILED (failures=1, errors=8)
```

I've also spent a few more hours playing around with FreeCAD and I have still have not run into any issues.

ipatch, 2021-11-14:
> With that patch applied to `nglib.rb`, and all the other changes to the Formulae I then ran into [tracker.dev.opencascade.org/view.php?id=32328](https://tracker.dev.opencascade.org/view.php?id=32328) which I worked around by simply including the additional header file:
> 
> ```
> --- ShapeUpgrade_UnifySameDomain.hxx.old	2021-11-10 15:36:27.000000000 -0800
> +++ ShapeUpgrade_UnifySameDomain.hxx.new	2021-11-10 15:36:34.000000000 -0800
> @@ -21,6 +21,7 @@
>  #include <Standard.hxx>
>  #include <Standard_Type.hxx>
> 
> +#include <TopoDS_Edge.hxx>
>  #include <TopoDS_Shape.hxx>
>  #include <Standard_Boolean.hxx>
>  #include <Standard_Transient.hxx>
> ```
> 
> With that fix applied the freecad build completed successfully and when I run `/opt/homebrew/bin/FreeCAD` the GUI starts up, but there are some `No module named "PySide2"` errors, which could well be caused by my changes to the various Formulae.

this is something i have run into when attempting to build freecad using `brew install freecad` however when building from a normal CLI, using the `cmake, make, make install` i generally do not run into this issue. either way i'll work on a patch to fix this, in hopes of getting `brew install freecad` going where it should be.

---

on a side note (forgive for hijacking this thread) i downloaded the lastest occ v7.6.0 source code and opened up the offending file and sure enough that fix you suggested in my quoted reply appears to not be there. so if i have time later this evening i'll try and get patch going for both the opencascade 7.6.0 formula and the existing opencascade 7.5.3 formula we have in this repo, and hopefully get all my changes pushed for getting the `freecad.rb` formula updated.

i also have lots of documentation that i need to compile down _no pun_ and update to hopefully make this whole process a lot better. everybody thanks for reporting issues, finding fixes, and having patience. like seeing the team work here. 🙌

neurodigitaltechnologies, 2021-11-14:
Hi @ipatch!
Wonderful news your active support!

So then, after your commits, are we able to compile Freecad for Apple Silicon on Monterey with "brew install freecad --HEAD --with-macos-app" after brew tap Freecad/freecad?

I'm just waiting to your signal to start compiling it!
Many thanks in advance.
Kindest,
Luis.

stepheneb, 2021-11-16:
See WIP pull request: #264, `fix "Another active Homebrew process ..."`

stepheneb, 2021-11-16:
@neurodigitaltechnologies , In my pull request, #264, I converted the form of the dependency paths from relative to absolute scoped to the tap ... which now allows this invocation to compile successfully. There is still a problem when the formula tries include the python dependencies from `site-packages` when packaging it all up into an app.

`brew install freecad --HEAD --with-macos-app --build-from-source `

Example of the change:

From: `depends_on "./boost@1.75.0"`
To: `depends_on "freecad/freecad/boost@1.75.0"`

stepheneb, 2021-11-16:
@vidmik 

Another change I made in #264 was fixing the generation of some of the `prefix_paths`.

This code: `(Formula["#{@tap}/opencascade@7.5.0"].opt_prefix + "/lib/cmake;"))`
generated this path: `/lib/cmake;`

Updated to: `(Formula["#{@tap}/opencascade@7.5.3"].opt_prefix/"lib/cmake;")`
generates: `/usr/local/opt/opencascade@7.5.3/lib/cmake;`

ipatch, 2021-12-10:
most if not all these issues should be fixed with PR #266 which should build a `FreeCAD.app` bundle on amd64 / x86_64 platforms. the CI even created bottles for catalina (the upstream homebrew  test-bot doesn't support mojave anymore 😢) and i haven't setup a Monterey vm yet to test and build FreeCAD and related deps yet.

i also removed the deprecated formula after updating the `freecad.rb` formula. with pr #277 

so the only known issue i'm presently running into is runtime issues related to pyside2 after the freecad gui launches which seems to be the same issue @vidmik pointed out in the below comment.

https://github.com/FreeCAD/homebrew-freecad/issues/237#issuecomment-965855376

hopefully i can update the `freecad.rb` formula in a day or two and resolve the runtime issues related to pyside2.

ipatch, 2022-08-17:
this should be resolved with the below commit,

https://github.com/FreeCAD/homebrew-freecad/commit/addab68821f9c96272c24c1540edfd0cbb486953

closed 2022-08-17 by ipatch