FreeCAD/homebrew-freecad #258: Problem with Pyside2 sha256 mismatch, building from sources on Apple Silicon Monterey

neurodigitaltechnologies, 2021-11-05:
I have this problem when trying to build it against Apple Silicon on Monterey:

The main issue is I cannot find where that especific sh256 is in any formula inside homebrew-freecad.
Any way to solve this, please?

**brew tap --HEAD Freecad/freecad
brew install Freecad/freecad --HEAD --build-from-source --verbose**

==> Downloading https://codereview.qt-project.org/gitweb?p=pyside/pyside-setup.git;a=snapshot;h=c7f4942dda0cf72032e7d8c601b1b32b22c4d8f0;sf=tgz
/opt/homebrew/Library/Homebrew/shims/shared/curl --disable --cookie /dev/null --globoff --show-error --user-agent Homebrew/3.3.2-50-geca16a2\ \(Macintosh\;\ arm64\ Mac\ OS\ X\ 12.0.1\)\ curl/7.77.0 --header Accept-Language:\ en --fail --retry 3 --location --remote-time --output /Users/luis/Library/Caches/Homebrew/downloads/6334d9b2233f86bb2d74403bc752db51aa5c934161844f434723d0682b83b0fc--pyside-setup-c7f4942.tar.gz.incomplete https://codereview.qt-project.org/gitweb\?p=pyside/pyside-setup.git\;a=snapshot\;h=c7f4942dda0cf72032e7d8c601b1b32b22c4d8f0\;sf=tgz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 5079k    0 5079k    0     0  2859k      0 --:--:--  0:00:01 --:--:-- 2871k
==> Verifying checksum for '6334d9b2233f86bb2d74403bc752db51aa5c934161844f434723d0682b83b0fc--pyside-setup-c7f4942.tar.gz'
**Error: freecad: SHA256 mismatch
Expected: 61b7ffb05ef0c42a0ad6b48e9b5a926e9e016cabf0a6881396ebe53e3a505f61
  Actual: ea19375d9f77accabc81d066cb4f745a2a8ed3fbbec7d46f39179f77c00f1b76**
    File: /Users/luis/Library/Caches/Homebrew/downloads/6334d9b2233f86bb2d74403bc752db51aa5c934161844f434723d0682b83b0fc--pyside-setup-c7f4942.tar.gz
To retry an incomplete download, remove the file above.

ipatch, 2021-11-05:
i don't have access to M1 machine, so it's difficult for to test issues specific to M1. since there are only bottles for amd64 / x86 processors at the moment. it might be that when you attempt to install freecad it's downloading the dependencies for freecad ie. `pyside2` there could be a invalid sha256sum for the pyside2 tarball. i'll investigate this and see what i can find.

vidmik, 2021-11-09:
I'm also seeing this issue. FWIW the "Actual" checksum seems to be different on every retry.

ipatch, 2021-11-09:
> I'm also seeing this issue. FWIW the "Actual" checksum seems to be different on every retry.
> 
> 
> 
> Edit: Confirmed, for whatever reason the contents downloaded from https://codereview.qt-project.org/gitweb?p=pyside/pyside-setup.git;a=snapshot;h=c7f4942dda0cf72032e7d8c601b1b32b22c4d8f0;sf=tgz differ for every download, both in size and actual data.
> 
> 
> 
> Edit2: The mtime (modification time) field in the tar header seems to be the blame, I'm guessing it will be set to the time when the download happens by the code review (Gerrit?) server.

thats some good detective work there. 👍 i bet if you look at the upstream pyside@2 formula theyll have the correct path to the targz file that uses the same shasum.

all that said, im presently afk for the next few hours but if you want to send a pr with an updated url i can def merge it in. thanks again for the detective work. 🙏 

---

im still banging my head with app bundling issues, ...getting closer though

ipatch, 2021-11-09:
did y'all attempt to install pyside2 from this tap again, after i merged the below PR?

https://github.com/FreeCAD/homebrew-freecad/commit/60dbd3da49393ff6431ac06c93fdcfe2d704580e

vidmik, 2021-11-09:
I'm new to homebrew and tap so ELI5: anything in particular I need to do to get the latest change? Simply running `brew install --only-dependencies freecad --with-packaging-utils` still leads to the same checksum mismatch problem. This is on macOS 12 (Monterey) if that matters..?

ipatch, 2021-11-10:
homebrew-core provide the pyside2 package as well. for the moment there are quite a few issues that need to be resolved in order to install freecad from using `brew install freecad` so probably not what you wanna here, but you could try to install the dependencies individually until i or somewhere can work through the issues.

vidmik, 2021-11-10:
Understood. Let me know if there's anything specific you want me to try.

vidmik, 2021-11-10:
FWIW `brew install freecad/freecad/pyside2 -s --verbose` completes successfully, and I do indeed have the latest/greatest master changes (385 commits).

IIUC the source url in the shiboken2.rb file still needs to be updated to something stable, correct? If so, it's not immediately clear to me why it's not using the same url as shiboken2@5.15.2.rb - after all they both appear to refer to 5.15.2. That said, I should probably back away from this slowly now because I'm clearly in over my head.

vidmik, 2021-11-10:
Because I'm notoriously bad at giving up on technical challenges I decided to update the url and checksum fields to the same ones used in `shiboken2@5.15.2.rb` and that does seem to get me past the checksum issue. 🎉 

Next up I'm running into the boost issue (#237).

neurodigitaltechnologies, 2021-11-10:
Hi @vidmik and @ipatch
The problem is that although you solve the thing related to shiboken, as you are identifying, then we have the thing with "boost on [https://github.com/FreeCAD/homebrew-freecad/issues/237](url).
I modified all .rb Formulas to switch from local versions of the formulas AKA (./$Formula) for home-core or "vanilla" versions. We must do this change recursively across all dependencies.
For example:

In Freecad.rb, changing ".boost@1.7.5" with "boost" will use the untouched and latest version from the core. 
This eventually works till we face nglib. It's not compiling due to a miss OCC Lib included in the CmakeList.txt I cannot solve. Anyone else is reaching this point?
Kindest!
Luis.

ipatch, 2021-12-10:
this issue should be resolved at this point. please comment or reopen if issues arrises again.

closed 2021-12-10 by ipatch