FreeCAD/homebrew-freecad #267: add netcdf.x.y.z?? -- vtk@8.2.0 appears to require netcdf/lib/libnetcdf.18.dylib

stepheneb, 2021-11-17:
I've made a bunch of progress getting `cmake --install .`  to complete successfully (outside of the formula but using all the dependencies it generates). Am now stuck because I'm pretty sure that vtk@8.2.0 needs `netcdf/lib/libnetcdf.18.dylib` -- but the current brew netcdf generates `libnetcdf.19.dylib`.

Here's the error in the console (I've added some extra logging in `src/tools/MakeMacBundleRelocatable.py`) to figure this out:

Building in a  `freecad-build` directory:

```
$ cmake --install .
...
-- INFO: /usr/local/Cellar/vtk@8.2.0/8.2.0_1/lib/libvtkParallelCore-8.2.1.dylib
-- INFO: /usr/local/Cellar/vtk@8.2.0/8.2.0_1/lib/libvtkFiltersModeling-8.2.1.dylib
-- INFO: /usr/local/Cellar/vtk@8.2.0/8.2.0_1/lib/libvtkFiltersParallel-8.2.1.dylib
-- INFO: /usr/local/Cellar/vtk@8.2.0/8.2.0_1/lib/libvtkIONetCDF-8.2.1.dylib
-- INFO: dk: /usr/local/opt/netcdf/lib/libnetcdf.18.dylib
-- INFO: /usr/local/opt/netcdf/lib/libnetcdf.18.dylib
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/otool-classic: can't open file: /usr/local/opt/netcdf/lib/libnetcdf.18.dylib (No such file or directory)
-- ERROR: Failed to resolve dependency in /usr/local/opt/netcdf/lib/libnetcdf.18.dylib
Traceback (most recent call last):
  File "/Users/stephen/dev/freecad/FreeCAD/src/Tools/MakeMacBundleRelocatable.py", line 393, in <module>
    main()
  File "/Users/stephen/dev/freecad/FreeCAD/src/Tools/MakeMacBundleRelocatable.py", line 376, in main
    build_deps_graph(graph, bundle_path, dir_filter, search_paths)
  File "/Users/stephen/dev/freecad/FreeCAD/src/Tools/MakeMacBundleRelocatable.py", line 241, in build_deps_graph
    deps = create_dep_nodes(list_install_names(k2), s_paths)
  File "/Users/stephen/dev/freecad/FreeCAD/src/Tools/MakeMacBundleRelocatable.py", line 112, in list_install_names
    output = check_output(["otool", "-L", path_macho])
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 223, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['otool', '-L', '/usr/local/opt/netcdf/lib/libnetcdf.18.dylib']' returned non-zero exit status 1
```
It seems that if there was a versioned `netcdf.x.y.z.rb` formula that generated this version of the dylib `libnetcdf.18.dylib` that might solve this issue.

Three question I don't know the answers to right now:
1. Is this the right way to fix this issue?
1. Which release of netcdf generates the dylib version `libnetcdf.18.dylib`?
2. How are versioned bottle artifacts produced and added as releases to homebrew-freecad?

To get this far I made these changes in `FreeCAD/src/MacAppBundle/CMakeLists.txt`:

https://github.com/FreeCAD/FreeCAD/compare/master...stepheneb:mac-app-bundle-paths-to-specific-dependencies?expand=1

I suspect there is a better way than adding these specific paths in `MacAppBundle/CMakeLists.txt` --  but it solved a bunch of problems in the `install` step.

ipatch, 2021-11-17:
homebrew provides a lot of little diagnostic commands a couple useful ones in this case are,

```
brew uses [NAME_OF_FORMULA]
brew deps [NAME_OF_FORMULA]
```

and there are several other commands for exploring all things homebrew try

```
brew commands
```

will print a list of all the commands available to homebrew

ipatch, 2021-12-10:
@stepheneb 

you still running into this issue with the latest updates i've made in the past couple of days?

stepheneb, 2021-12-30:
@ipatch I just tried the master branch and running this command invocation:

`$ brew install --HEAD freecad --with-no-macos-app`

Appeared to fail building Shiboken2-5.15.2

```
[homebrew-freecad (master)]$ pwd
/usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad
[homebrew-freecad (master)]$ git show --shortstat 
commit 04b498b7b7f4839b18e6ef4f6e29fef0c004e4ac (HEAD -> master, origin/master, origin/HEAD)
Author: Chris <chris.r.jones.1983@gmail.com>
Date:   Wed Dec 15 14:04:24 2021 -0600

    [no ci] update readme (#284)

 1 file changed, 26 insertions(+), 19 deletions(-)
```

```
[dev]$ brew install --HEAD freecad --with-no-macos-app
Running `brew update --preinstall`...
==> Auto-updated Homebrew!
Updated 3 taps (homebrew/core, homebrew/cask and homebrew/cask-drivers).
==> Updated Formulae
Updated 9 formulae.
==> Updated Casks
Updated 62 casks.

Warning: Treating freecad as a formula. For the cask, use homebrew/cask/freecad
Warning: tbb@2020 has been deprecated because it is not supported upstream!
==> Downloading https://ghcr.io/v2/homebrew/core/tbb/2020/manifests/2020_U3
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/9deaaa23013e8242ae24eb4bc001b5db8410770cd729797281680418ac7a6a3a--tbb@2020-2020_U3.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/tbb/2020/blobs/sha256:596f3f92c1765f24b9dc9cd866e8068c505428c9dcb9941df7b5f0ea4e10cde9
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/19c0c4b9949326f451dbb72b99bac92c3d516c06a0275878d37dace2511813ac--tbb@2020--2020_U3.big_sur.bottle.tar.gz
==> Downloading https://ghcr.io/v2/homebrew/core/coin3d/manifests/4.0.0_2
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/2fa174df5a41b6ac805e470b2a38eb7f0a2efba226b83f2ba3da4a93c3bf96e0--coin3d-4.0.0_2.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/coin3d/blobs/sha256:4cb262e52fb59b9c79d8aa11e32ec5498335ec536fdf7659c5249a4e17e4c3db
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/17ba86ce1f708a4114f8c6dac890da7ad71e866b3f23b354006ec1295cf7c737--coin3d--4.0.0_2.big_sur.bottle.tar.gz
==> Downloading https://ghcr.io/v2/homebrew/core/cython/manifests/0.29.26
######################################################################## 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/cython/blobs/sha256:9a8158098d2a41392113b7814d7350fe486654189c8157fcad0a856ee6eff7b9
==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:9a8158098d2a41392113b7814d7350fe486654189c8157fcad0
######################################################################## 100.0%
==> Downloading https://github.com/FreeCAD/homebrew-freecad/releases/download/matplotlib@3.4.3-3.4.3/matplotlib%403.4.3-3.4.3.big_sur.bo
==> Downloading from https://objects.githubusercontent.com/github-production-release-asset-2e65be/18778796/84368cc8-97d3-42e0-b1df-ae9f8
######################################################################## 100.0%
==> Downloading https://github.com/FreeCAD/homebrew-freecad/releases/download/med-file@4.1.0-4.1.0/med-file%404.1.0-4.1.0.big_sur.bottle
==> Downloading from https://objects.githubusercontent.com/github-production-release-asset-2e65be/18778796/5aa1d522-71eb-422f-a525-375bb
######################################################################## 100.0%
==> Downloading https://github.com/FreeCAD/homebrew-freecad/releases/download/opencascade@7.5.3-7.5.3/opencascade%407.5.3-7.5.3.big_sur.
==> Downloading from https://objects.githubusercontent.com/github-production-release-asset-2e65be/18778796/dfe0619b-8f8e-4cd1-8a22-95d63
######################################################################## 100.0%
==> Downloading https://github.com/FreeCAD/homebrew-freecad/releases/download/nglib@6.2.2104-6.2.2104_1/nglib%406.2.2104-6.2.2104_1.big_
==> Downloading from https://objects.githubusercontent.com/github-production-release-asset-2e65be/18778796/14507202-7c1f-4455-a263-3e8df
######################################################################## 100.0%
==> Downloading https://github.com/FreeCAD/homebrew-freecad/releases/download/shiboken2@5.15.2-5.15.2/shiboken2%405.15.2-5.15.2.big_sur.
==> Downloading from https://objects.githubusercontent.com/github-production-release-asset-2e65be/18778796/2392d3ca-49f1-4c5b-b15b-2e9bd
######################################################################## 100.0%
==> Downloading https://github.com/FreeCAD/homebrew-freecad/releases/download/pyside2@5.15.2-5.15.2/pyside2%405.15.2-5.15.2.big_sur.bott
==> Downloading from https://objects.githubusercontent.com/github-production-release-asset-2e65be/18778796/7a16d4b8-00ee-4fcf-a01c-ab0e2
######################################################################## 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/llvm/11/manifests/11.1.0_3
######################################################################## 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/llvm/11/blobs/sha256:8b9e7e448f801daad56d3180cccf190486bd785c551aeb37300a5296f1d9e81b
==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:8b9e7e448f801daad56d3180cccf190486bd785c551aeb37300
######################################################################## 100.0%
Warning: vtk@8.2 has been deprecated because it is a versioned formula!
==> Downloading https://ghcr.io/v2/homebrew/core/vtk/8.2/manifests/8.2.0_7-1
[... cut ...]
Problem with BOOST version installing `freecad --HEAD --with-macos-app` on BigSur https://github.com/FreeCAD/homebrew-freecad/issues/237
Freecad build fails on Big Sur during Install https://github.com/FreeCAD/homebrew-freecad/issues/163
homebrew build-from-source post-install step issue with homebrew-freecad-bundle.pth check https://github.com/FreeCAD/homebrew-freecad/issues/154
Please create pull requests instead of asking for help on Homebrew's GitHub,
Twitter or any other official channels.
```

ipatch, 2021-12-30:
i'm testing the command you ran above,

```
brew install --HEAD freecad --with-no-macos-app
```

so give me a second and i'll report back with my results, but looking at your output,

```
==> Installing freecad/freecad/freecad --HEAD --with-no-macos-app
==> pip3 install PySide2
==> cmake -DHOMEBREW_PREFIX=/usr/local -DUSE_PYTHON3=1 -DPYTHON_EXECUTABLE=/usr/local/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/local/opt/py
Last 15 lines from /Users/stephen/Library/Logs/Homebrew/freecad/02.cmake:
  version which was used when Shiboken was built.  Consider building shiboken
  with FORCE_LIMITED_API set to '1', so that only the Python major version
  matters.

  Built with: '3.9' Detected: '3.10'
Call Stack (most recent call first):
  /usr/local/opt/pyside@2/lib/cmake/Shiboken2-5.15.2/Shiboken2Config.cpython-39-darwin.cmake:45 (shiboken_check_if_built_and_target_python_are_compatible)
  /usr/local/opt/pyside@2/lib/cmake/Shiboken2-5.15.2/Shiboken2Config.cmake:5 (include)
  cMake/FreeCAD_Helpers/SetupShibokenAndPyside.cmake:13 (find_package)
  CMakeLists.txt:75 (SetupShibokenAndPyside)

-- Configuring incomplete, errors occurred!
See also "/tmp/freecad-20211229-1288-m0u8xp/Build/CMakeFiles/CMakeOutput.log".
See also "/tmp/freecad-20211229-1288-m0u8xp/Build/CMakeFiles/CMakeError.log".

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/freecad/homebrew-freecad/issues
```

...taking a shot in the dark here 🏹🦮🎯 it may be related to you having _python@3.10_ linked and you could be possibly building deps against that version whereas this tap is setup to use _python@3.9_

so you could try and link _python@3.9_ instead of _python@3.10_

```
brew link python@3.9
```

you might have to pass some flags to get _python@3.9_ linked ie. `--overwrite` or something like that.

stepheneb, 2021-12-30:
python@3.9 is linked but there still seems to be an issue with: `Built with: '3.9' Detected: '3.10'`

```
[dev]$ brew link python@3.9
Warning: Already linked: /usr/local/Cellar/python@3.9/3.9.9
To relink, run:
  brew unlink python@3.9 && brew link python@3.9
[dev]$ brew unlink python@3.9 && brew link python@3.9
Unlinking /usr/local/Cellar/python@3.9/3.9.9... 24 symlinks removed.
Linking /usr/local/Cellar/python@3.9/3.9.9... 24 symlinks created.
[dev]$ brew install --HEAD freecad --with-no-macos-app
Running `brew update --preinstall`...
==> Auto-updated Homebrew!
Updated Homebrew from 26694fd63 to 0b14b6d76.
Updated 2 taps (homebrew/core and homebrew/cask).
==> New Formulae
bash-language-server              grafana-agent                     mold                              zsh-fast-syntax-highlighting
fypp                              libgrape-lite                     yaml-language-server
==> Updated Formulae
Updated 49 formulae.
==> New Casks
fastmarks                                     kdrive                                        only-switch
==> Updated Casks
Updated 26 casks.

Warning: Treating freecad as a formula. For the cask, use homebrew/cask/freecad
==> Cloning https://github.com/freecad/FreeCAD.git
Updating /Users/stephen/Library/Caches/Homebrew/freecad--git
From https://github.com/freecad/FreeCAD
   4098c88171..bb5c3d2091  master     -> origin/master
==> Checking out branch master
Already on 'master'
Your branch is behind 'origin/master' by 12 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)
HEAD is now at bb5c3d2091 Use std::map without const
==> Installing freecad from freecad/freecad
==> pip3 install PySide2
==> cmake -DHOMEBREW_PREFIX=/usr/local -DUSE_PYTHON3=1 -DPYTHON_EXECUTABLE=/usr/local/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/local/opt/py
Last 15 lines from /Users/stephen/Library/Logs/Homebrew/freecad/02.cmake:
  version which was used when Shiboken was built.  Consider building shiboken
  with FORCE_LIMITED_API set to '1', so that only the Python major version
  matters.

  Built with: '3.9' Detected: '3.10'
Call Stack (most recent call first):
  /usr/local/opt/pyside@2/lib/cmake/Shiboken2-5.15.2/Shiboken2Config.cpython-39-darwin.cmake:45 (shiboken_check_if_built_and_target_python_are_compatible)
  /usr/local/opt/pyside@2/lib/cmake/Shiboken2-5.15.2/Shiboken2Config.cmake:5 (include)
  cMake/FreeCAD_Helpers/SetupShibokenAndPyside.cmake:13 (find_package)
  CMakeLists.txt:75 (SetupShibokenAndPyside)

-- Configuring incomplete, errors occurred!
See also "/tmp/freecad-20211230-8434-l5f0z5/Build/CMakeFiles/CMakeOutput.log".
See also "/tmp/freecad-20211230-8434-l5f0z5/Build/CMakeFiles/CMakeError.log".
```

stepheneb, 2021-12-30:
Interesting ... so even after uninstalling python 3.10 with brew there is some dependency pulling in python 3.10

Here's just the beginning of the verbose log output:

```
$ brew install --HEAD freecad --with-no-macos-app --verbose
Warning: Treating freecad as a formula. For the cask, use homebrew/cask/freecad
==> Downloading https://ghcr.io/v2/homebrew/core/python/3.10/manifests/3.10.1
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/ba44b4cfaec4b0996a8fc2b2def6f21371f8c859781b668a30ceb2e6f4bbf3be--python@3.10-3.10.1.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/python/3.10/blobs/sha256:c4f29a6bbdc7b48f6fd05fe8a9d9a136d9c4dfbcc0d2e42f7f7d5624206f0f5d
Already downloaded: /Users/stephen/Library/Caches/Homebrew/downloads/e88ca2820c3e1b5254129a9f6edabe6e29e7b283796213cd956c8b83d6908066--python@3.10--3.10.1.big_sur.bottle.tar.gz
==> Verifying checksum for 'e88ca2820c3e1b5254129a9f6edabe6e29e7b283796213cd956c8b83d6908066--python@3.10--3.10.1.big_sur.bottle.tar.gz'
==> Cloning https://github.com/freecad/FreeCAD.git
Updating /Users/stephen/Library/Caches/Homebrew/freecad--git
git config remote.origin.url https://github.com/freecad/FreeCAD.git
git config remote.origin.fetch \+refs/heads/master:refs/remotes/origin/master
git config remote.origin.tagOpt --no-tags
git config advice.detachedHead false
git fetch origin
==> Checking out branch master
git checkout -f master --
Already on 'master'
Your branch is up to date with 'origin/master'.
git reset --hard origin/master --
HEAD is now at bb5c3d2091 Use std::map without const
==> Installing freecad from freecad/freecad
==> Installing dependencies for freecad/freecad/freecad: python@3.10
==> Installing freecad/freecad/freecad dependency: python@3.10
==> Pouring python@3.10--3.10.1.big_sur.bottle.tar.gz
tar --extract --no-same-owner --file /Users/stephen/Library/Caches/Homebrew/downloads/e88ca2820c3e1b5254129a9f6edabe6e29e7b283796213cd956c8b83d6908066--python@3.10--3.10.1.big_sur.bottle.tar.gz --directory /private/tmp/d20211230-22772-10onh95
cp -pR /private/tmp/d20211230-22772-10onh95/python@3.10/. /usr/local/Cellar/python@3.10
chmod -Rf +w /private/tmp/d20211230-22772-10onh95
==> Finishing up
==> /usr/local/Cellar/python@3.10/3.10.1/bin/python3 -m ensurepip
==> /usr/local/Cellar/python@3.10/3.10.1/bin/python3 -m pip install -v --no-deps --no-index --upgrade --isolated --target=/usr/local/lib/python3.10/site-packages /usr/local/Cellar/python@3.10/3.10.1/Fr
==> Summary
🍺  /usr/local/Cellar/python@3.10/3.10.1: 3,132 files, 56MB
==> Installing freecad/freecad/freecad --HEAD --with-no-macos-app
...
```

stepheneb, 2021-12-30:
Ahh .. it is caused by an upgraded llvm:

```
$ brew deps --tree --installed llvm
llvm
└── python@3.10
    ├── gdbm
    ├── mpdecimal
    ├── openssl@1.1
    │   └── ca-certificates
    ├── readline
    ├── sqlite
    │   └── readline
    └── xz
```

stepheneb, 2021-12-30:
Here are the elements in the full freecad dependency tree that pull in python@3.10:

1. coin3d  < pyside@2 < llvm < python@3.10
2. freecad/freecad/pyside2@5.15.2 < freecad/freecad/shiboken2@5.15.2 < llvm < python@3.10

```
├── coin3d
│   ├── boost
│   │   └── icu4c
│   ├── pyside@2
│   │   ├── llvm
│   │   │   └── python@3.10

├── freecad/freecad/pyside2@5.15.2
│   ├── freecad/freecad/shiboken2@5.15.2
│   │   ├── llvm
│   │   │   └── python@3.10

├── freecad/freecad/shiboken2@5.15.2
│   ├── llvm
│   │   └── python@3.10
```

It's easy to resolve the freecad/freecad/shiboken2@5.15.2 dependency to use llvm@12

```::diff
$ git diff
diff --git a/Formula/shiboken2@5.15.2.rb b/Formula/shiboken2@5.15.2.rb
index 37f86c3..31f2458 100644
--- a/Formula/shiboken2@5.15.2.rb
+++ b/Formula/shiboken2@5.15.2.rb
@@ -15,7 +15,7 @@ class Shiboken2AT5152 < Formula
 
   depends_on "cmake" => :build
   depends_on "python@3.9" => :build
-  depends_on "llvm"
+  depends_on "llvm@12"
   depends_on "numpy"
   depends_on "qt@5"
```

But right now the formula adds a regular coin3d dependency, which pulls in pyside@2 < llvm < python@3.10

ipatch, 2021-12-31:
this may be another issue because you're building freecad to be launched from a CLI and not a mac app bundle.

https://github.com/FreeCAD/homebrew-freecad/blob/master/Formula/freecad.rb#L134

ipatch, 2021-12-31:
okay, i haven't messed around with this tap or building freecad from source in a while, so i checked out the `0.19.3` tag from the official freecad github source.

created a fresh build branch and a fresh build dir. kindof running through my cmake setup on my local catalina box and sure enough i'm running into the python 3.9 / 3.10 mismatch after using the below _MVP_ cmake setup

```
cmake \
        -DHOMEBREW_PREFIX="$bp" \
        -DCMAKE_PREFIX_PATH="$pthvtk;$pthqt;" \
        -DPYTHON_EXECUTABLE="$bp/bin/python3" \
        -DBUILD_QT5=1 \
        -DCMAKE_BUILD_TYPE=RelWithDebInfo \
        ../../freecad.src
```

i usually just start with a fresh build dir from time to time and then run `cmake` and then begin to bolt on args for my cmake command with that above cmake command i have run,

```
brew link python3.10
```

which unlinks my python@3.9 hombrew installation and links in the python@3.10 installation from homebrew-core.

and sure enough i get the below error message,

```
-- Found Freetype: /usr/local/lib/libfreetype.dylib (found version "2.11.1")
-- Coin3D libraries found
-- Could NOT find Spnav (missing: SPNAV_LIBRARY SPNAV_INCLUDE_DIR)
-- Shiboken2Config: Using default python: .cpython-39-darwin
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.10.1", minimum required is "3")
-- Found PythonLibs: /usr/local/Frameworks/Python.framework/Versions/3.10/lib/libpython3.10.dylib (found suitable version "3.10.1", minimum required is "3")
CMake Error at /usr/local/lib/cmake/Shiboken2-5.15.2/shiboken_helpers.cmake:468 (message):
  The detected Python minor version is not compatible with the Python minor
  version which was used when Shiboken was built.  Consider building shiboken
  with FORCE_LIMITED_API set to '1', so that only the Python major version
  matters.

  Built with: '3.9' Detected: '3.10'
Call Stack (most recent call first):
  /usr/local/lib/cmake/Shiboken2-5.15.2/Shiboken2Config.cpython-39-darwin.cmake:45 (shiboken_check_if_built_and_target_python_are_compatible)
  /usr/local/lib/cmake/Shiboken2-5.15.2/Shiboken2Config.cmake:5 (include)
  cMake/FreeCAD_Helpers/SetupShibokenAndPyside.cmake:25 (find_package)
  CMakeLists.txt:75 (SetupShibokenAndPyside)

-- Configuring incomplete, errors occurred!
See also "/opt/code/github/public/forks/freecad-git/builds/dec31.0.19.3/CMakeFiles/CMakeOutput.log".
See also "/opt/code/github/public/forks/freecad-git/builds/dec31.0.19.3/CMakeFiles/CMakeError.log".
```

so let me see if i can cook up a shiboken2@5.15.2 formula that uses python@3.10 from homebrew-core to bump this build a little further and see what crops up next.

thanks for reporting this issue. BTW. 🙏

ipatch, 2021-12-31:
https://github.com/FreeCAD/homebrew-freecad/runs/4676213903?check_suite_focus=true#step:13:146

ipatch, 2021-12-31:
seems like there may be a fix for this...

https://bugreports.qt.io/browse/PYSIDE-1436

ipatch, 2022-01-01:
https://bugreports.qt.io/browse/PYSIDE-1436?focusedCommentId=580150&page=com.atlassian.jira.plugin.system.issuetabpanels%3Acomment-tabpanel#comment-580150

ipatch, 2022-01-02:
once the 2nd ci job finishes i believe this should give us a minimum viable version of shiboken2 that works with python 3.10

i haven't fully tested it yet, so feel free to build it on your box. also i assume pyside2 will also require some patching to get working and possible pivy and coin as well ☹️

but i guess the below is a start,

https://github.com/FreeCAD/homebrew-freecad/pull/285

stepheneb, 2022-01-02:
Yes, it looks like pyside@2 also builds Shiboken2-5.15.2 and is generating an error.

I realize that `--with-no-macos-app` might not work but I expect any errors caused by that argument to present themselves after this part of the build process.

```
$ brew install --HEAD freecad --with-no-macos-app --verbose
...
-- Coin3D libraries found
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'pivy'
-- Found Spnav: /usr/local/lib/libspnav.dylib  
-- Shiboken2Config: Using default python: .cpython-39-darwin
-- Found PythonInterp: /usr/local/opt/python@3.10/bin/python3.10 (found suitable version "3.10.1", minimum required is "3") 
-- Found PythonLibs: /usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/libpython3.9.dylib (found suitable version "3.9.9", minimum required is "3") 
CMake Error at /usr/local/opt/pyside@2/lib/cmake/Shiboken2-5.15.2/shiboken_helpers.cmake:468 (message):
  The detected Python minor version is not compatible with the Python minor
  version which was used when Shiboken was built.  Consider building shiboken
  with FORCE_LIMITED_API set to '1', so that only the Python major version
  matters.

  Built with: '3.9' Detected: '3.10'
Call Stack (most recent call first):
  /usr/local/opt/pyside@2/lib/cmake/Shiboken2-5.15.2/Shiboken2Config.cpython-39-darwin.cmake:45 (shiboken_check_if_built_and_target_python_are_compatible)
  /usr/local/opt/pyside@2/lib/cmake/Shiboken2-5.15.2/Shiboken2Config.cmake:5 (include)
  cMake/FreeCAD_Helpers/SetupShibokenAndPyside.cmake:13 (find_package)
  CMakeLists.txt:75 (SetupShibokenAndPyside)
```

ipatch, 2022-01-03:
i reported the issues with pyside2 upstream below,

https://bugreports.qt.io/browse/PYSIDE-1763

ipatch, 2022-03-14:
another thing to consider,

https://github.com/Homebrew/homebrew-core/issues/76621#issuecomment-1044984628

looks like getting python 3.10 support working with freecad and friends will aid with the above ☝️ mentioned issue as well.

ipatch, 2022-08-17:
i feel pretty confident this issue has been resolved with the below commit,

https://github.com/FreeCAD/homebrew-freecad/commit/addab68821f9c96272c24c1540edfd0cbb486953

and `brew install freecad/freecad/freecad` should install the latest v0.20 release (stable) of freecad.

closed 2022-08-17 by ipatch