FreeCAD/homebrew-freecad #278: freecad mac app bundle ie. `FreeCAD.app` runtime issue, cannot find the runtime dependency pyside2

ipatch, 2021-12-10:
presently running `brew install freecad` should either install freecad from one of the provided bottles for catalina or big sur. if running a different OS ie. mojave or monterey then freecad will have to be build from source locally.

so depending how the dependencies are setup ie. `pyside2@5.15.2` freecad will fail to load the pyside2 runtime dep when launching the FreeCAD gui when double clicking the app bundle. ie. `FreeCAD.app`.

presently, either `pyside@2` or the tap provided `pyside2@5.15.2` need to be manually linked before running

```
brew install freecad -s -v
```

and the runtime issue should be resolved, but haven't fully tested.

the CI, cmake scripts or `freecad.rb` formula file need to be updated to support (finding) the install of pyside in order to copy the necessary dependencies into the `FreeCAD.app` bundle.

as mentioned in a prior comment ie.

https://github.com/FreeCAD/homebrew-freecad/issues/237#issuecomment-965855376

it may be possible to tweak the freecad gui launch script to specify the path to the pyside2 installation. once again, i haven't fully tested this locally on my machine or with the CI/CD (github actions yet)

ipatch, 2021-12-10:
error messages after launching the `FreeCAD.app` GUI

```
18:31:43  During initialization the error "No module named 'PySide2'" occurred in /Users/brewmaster/homebrew/Cellar/freecad/0.19/FreeCAD.app/Contents/Mod/Draft/InitGui.py
18:31:43  Please look into the log file for further information
18:31:45  During initialization the error "No module named 'PySide2'" occurred in /Users/brewmaster/homebrew/Cellar/freecad/0.19/FreeCAD.app/Contents/Mod/Arch/InitGui.py
18:31:45  Please look into the log file for further information
18:31:45  During initialization the error "No module named 'PySide2'" occurred in /Users/brewmaster/homebrew/Cellar/freecad/0.19/FreeCAD.app/Contents/Mod/AddonManager/InitGui.py
18:31:45  Please look into the log file for further information
18:31:46  During initialization the error "No module named 'PySide2'" occurred in /Users/brewmaster/homebrew/Cellar/freecad/0.19/FreeCAD.app/Contents/Mod/Tux/InitGui.py
18:31:46  Please look into the log file for further information
18:31:48  Unknown command 'Std_AddonMgr'
18:31:48  Traceback (most recent call last):
  File "<string>", line 2, in <module>
  File "/Users/brewmaster/homebrew/Cellar/freecad/0.19/FreeCAD.app/Contents/Mod/Start/StartPage/StartPage.py", line 29, in <module>
    from . import TranslationTexts
  File "/Users/brewmaster/homebrew/Cellar/freecad/0.19/FreeCAD.app/Contents/Mod/Start/StartPage/TranslationTexts.py", line 24, in <module>
    from PySide import QtGui
  File "/Users/brewmaster/homebrew/Cellar/freecad/0.19/FreeCAD.app/Contents/MacOS/PySide/__init__.py", line 2, in <module>
    from PySide2 import __version__
<class 'ModuleNotFoundError'>: No module named 'PySide2'
18:31:48  No module named 'PySide2'
```

ipatch, 2021-12-10:
https://github.com/FreeCAD/homebrew-freecad/issues/164#issuecomment-984944471

ipatch, 2021-12-10:
[image]

neurodigitaltechnologies, 2021-12-22:
Hi @ipatch,
Any updates so far on this issue, please?
Many thanks in advance for all your support and hard work on this.
All the best,
Luis.

ipatch, 2021-12-22:
i haven't done any work with the freecad tap or specifically this issue in a week or so. the holidays caught up with me so my time has been limited.

if you're trying to install freecad using the tap provided formula ie. `freecad.rb` then a possible work around would be the below

install all the freecad dependencies first

```
brew install freecad/freecad/freecad -v --only-dependenices
```

manually link pyside2@5.15.2 and shiboken2@5.15.2

```
brew link -v pyside2@5.15.2
brew link -v shiboken2@5.15.2
```

then build freecad from source using the tap provided formula

```
brew install freecad -v -s
```

vejmarie, 2021-12-31:
I am currently looking at it. My build is broken on Mac, and still try to figuring out from where it comes.

vejmarie, 2022-01-01:
This PR  #5316 in FreeCAD shall be helping a lot

ipatch, 2022-01-01:
https://github.com/FreeCAD/FreeCAD/pull/5316

ipatch, 2022-01-22:
- [ ] is there a way to check if the _brew_ process ie. can  the install process (using the CI) be aware that it is being run in a CI?
	- [ ] if so, would it be possible to _run_ additional steps to perform the necessary linking for **pyside** & **shiboken2**

clyang, 2022-04-19:
> i haven't done any work with the freecad tap or specifically this issue in a week or so. the holidays caught up with me so my time has been limited.
>
> if you're trying to install freecad using the [...]

I'm able to build app bundle by using this workaround. However, when I tried to run it, it shows the following error.

```
During initialization the error "Failed to wrap widget" occurred in /Applications/FreeCAD.app/Contents/Mod/Tux/InitGui.py
Please look into the log file for further information
```

Is there any solution to this issue?

clyang, 2022-04-19:
Thanks for the prompt response!

I compiled it on my M1 laptop. After the successful build, here is my steps:
```
1. cd /opt/homebrew/Cellar/freecad/0.19
2. open FreeCAD.app
```

My `brew doctor`
```
Please note that these warnings are just used to help the Homebrew maintainers
with debugging if you file an issue. If everything you use Homebrew for is
working fine: please don't worry or file an issue; just ignore this. Thanks!

Warning: Some installed formulae are deprecated or disabled.
You should find replacements for the following formulae:
  tbb@2020
  vtk@8.2
```

`brew config`
```
HOMEBREW_VERSION: 3.4.7
ORIGIN: https://github.com/Homebrew/brew
HEAD: 09e766f83c8a420cce9e2210c46a24ec3864a094
Last commit: 30 hours ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 9e6c2e2b31ec47136266ada0c32a2ecf0589c20e
Core tap last commit: 63 minutes ago
Core tap branch: master
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_CORE_GIT_REMOTE: https://github.com/Homebrew/homebrew-core
HOMEBREW_MAKE_JOBS: 8
Homebrew Ruby: 2.6.8 => /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby
CPU: octa-core 64-bit arm_firestorm_icestorm
Clang: 13.1.6 build 1316
Git: 2.32.0 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 7.79.1 => /usr/bin/curl
macOS: 12.3.1-arm64
CLT: 13.3.1.0.1.1648687083
Xcode: 13.3.1
Rosetta 2: false
```

ipatch, 2022-07-09:
little update on this issue. i think the simple solution to getting the cmake build process to properly locate the pyside2 installation is to prepend the pyside installation bin to the path while running cmake. i tested a couple of cases where i had the pyside bin directory prepended (top entry) in my `$USER` path and without. i get the following _snippet_ without the both entry...

```
-- Shiboken2:            [Shiboken2_DIR-NOTFOUND] []
-- PySide2:              []
--  IncludeDir:         Unable to find, python version mismatch?
-- PySide2Tools:        [/usr/local/opt/qt@5/bin/uic] [/usr/local/opt/qt@5/bin/rcc]
```

when i add pyside2 to my path with the following,

```
fish_add_path -p $bp/Cellar/pyside@2/5.15.5/bin
```

and...

```
brew unlink shiboken2@5.15.5
Unlinking /usr/local/Cellar/shiboken2@5.15.5/5.15.5... 0 symlinks removed.

brew unlink pyside@2
Unlinking /usr/local/Cellar/pyside@2/5.15.5... 0 symlinks removed.

i see the below in the cmake report output...

```
-- Shiboken2:           5.15.5 [/usr/local/Cellar/pyside@2/5.15.5/lib/cmake/Shiboken2-5.15.5] [/usr/local/Cellar/pyside@2/5.15.5/include/shiboken2;/usr/local/opt/python@3.10/Frameworks/Python.framework/Headers]
-- PySide2:             5.15.5 [/usr/local/Cellar/pyside@2/5.15.5/include/PySide2]
-- PySide2Tools:        [/usr/local/opt/qt@5/bin/uic] [/usr/local/opt/qt@5/bin/rcc]
```

not sure if this will resolve all the runtime related issues with pyside but seems like a step in the right direction.

ipatch, 2022-07-09:
and i think this will help us  with keg only deps in the freecad formula ie. pyside2 being keg only because of it's upstream pyside6 counterpart, so if this works then hopefully `brew install freecad` will just work without having to manually link the keg only deps of shiboken2 and pyside2 🤞

ipatch, 2022-07-12:
#335 

should help in getting closer to having `brew install freecad` to just work without any manual unlink/link commands

ipatch, 2022-07-28:
another little update on this. a quick way to see if shiboken and pyside are setup properly for a cmake build is to open a pythone repl

```
python3

# i'm using python3 provided by homebrew-core python@3.10
```

```python
import shiboken2
import PySide2
from PySide2 import QtCore
```

i was running into some edge cases yesterday where i was able to import PySide2 but was not able to load any of the accompanying qt modules.

i made certain that my shiboken2 and pyside2 installation are **not** linked.

i got a build going right now (non mac bundle) that seems to have picked up on my shiboken and pyside installations on a m1 mac. however i did have to add a `.pth` file for each package into my `site-packages` dir for python@3.10.

and if i understand everything correctly, running the `brew link [PKG]` it copies everything from the `site-packages` in the Cellar for shiboken and pyside to the `opt` site-packages for the python installation. hence why things worked when linking the above packages.

so entirely sure i can add the `.pth` files for shiboken2 and pyside to pick up the installations without having to link the packages but hopefully this will be a step closer. and this will eventually help with homebrew CI builds. 🤞

a little q/a about the `.pth` files and site-packages dir

https://stackoverflow.com/a/7901603/708807

ipatch, 2022-08-17:
feeling confident this issue should be resolved with the below commit

https://github.com/FreeCAD/homebrew-freecad/commit/addab68821f9c96272c24c1540edfd0cbb486953

if anybody experiences any issues feel to search the open issues and leave comment if something comes up or if your issue is completely separate, please open a new issue.

i believe adding the below lines to the freecad.rb formula resolved this issue at least in all my local testing.

https://github.com/FreeCAD/homebrew-freecad/blob/addab68821f9c96272c24c1540edfd0cbb486953/Formula/freecad.rb#L139-L141

ipatch, 2022-08-24:
did a `brew install freecad/freecad/freecad` to test the bottle built via the CI. it appears some more work will be required for the FreeCAD GUI cmd to pick up on the runtime dependencies of shiboken2 and PySide2.

the quick i found was to manually create `.pth` file pointing the `site-packages` directory for both the shiboken2 and PySide2 installations. after creating those two files.

one quick test is to run the below cmd.

```
python3
```

should open the the python3 REPL that is installed via homebrew and which freecad relies upon, and then attempt to import both shiboken2 and pyside2  with the below commands

```
import shiboken2
import PySide2
```

got my bottle installation of freecad v0.20.1 going.

obviously this far from perfect, and more work will be required to get all this ironed out for future bottled releases.

ipatch, 2025-01-12:
little update on this, attempting to build a FreeCAD.app bundle built around python v3.12 and at v5 (jan 12 2025) using the freecad v1.0.0 git tag. presently after running `cmake`, then `make`, then `make install` i'm getting the below error,

```
-- Making bundle relocatable...
-- INFO: Analyzing bundle dependencies...
-- ERROR: Unable to find LC_DYLD_LOAD entry: @rpath/libshiboken2.cpython-312-darwin.5.15.dylib
-- ERROR: Failed to resolve dependency in /Users/chris/homebrew/opt/pyside2@5.15.15_py312/lib/libpyside2.cpython-312-darwin.5.15.dylib
Traceback (most recent call last):
  File "/Users/chris/code/freecad-git/fcsrc/src/Tools/MakeMacBundleRelocatable.py", line 424, in <module>
    main()
    ~~~~^^
  File "/Users/chris/code/freecad-git/fcsrc/src/Tools/MakeMacBundleRelocatable.py", line 406, in main
    build_deps_graph(graph, bundle_path, dir_filter, search_paths)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/chris/code/freecad-git/fcsrc/src/Tools/MakeMacBundleRelocatable.py", line 261, in build_deps_graph
    deps = create_dep_nodes(list_install_names(k2), s_paths)
  File "/Users/chris/code/freecad-git/fcsrc/src/Tools/MakeMacBundleRelocatable.py", line 183, in create_dep_nodes
    raise LibraryNotFound(lib_name + " not found in given search paths")
LibraryNotFound: libshiboken2.cpython-312-darwin.5.15.dylib not found in given search paths
-- Installing: /Users/chris/code/freecad-git/installs/1.0.0/FreeCAD.app/Contents/share/examples/Schenkel.stp
-- Installing: /Users/chris/code/freecad-git/installs/1.0.0/FreeCAD.app/Contents/share/examples/draft_test_objects.FCStd
-- Installing: /Users/chris/code/freecad-git/installs/1.0.0/FreeCAD.app/Contents/share/examples/EngineBlock.FCStd
```

ipatch, 2025-01-13:
the below snippet the previous mentioned `CMakeLists.txt` file will require updating / overhaul of logic,

```
  set(FC_BUNDLE_PY312_DIR "${HOMEBREW_PREFIX}/opt/fc_bundle_py312/lib/python3.12/site-packages")
    file(GLOB HOMEBREW_PTH_FILES 
      "${PYTHON_DIR}/site-packages/homebrew*.pth"
      "${FC_BUNDLE_PY312_DIR}/*.pth")

    message(STATUS "--ipatch:------------------------------------")
    message(STATUS "HOMEBREW_PTH_FILES is set to: ${HOMEBREW_PTH_FILES}")
    # NOTE: iaptch, no pth files are detected due to non existent dir
    message(STATUS "--ipatch:------------------------------------")

    foreach(PTH_FILE ${HOMEBREW_PTH_FILES})
        file(READ ${PTH_FILE} ADDITIONAL_DIR)

        string(STRIP "${ADDITIONAL_DIR}" ADDITIONAL_DIR)
        string(FIND "${ADDITIONAL_DIR}" "${HOMEBREW_PREFIX}/Cellar" POSITION)
        string(LENGTH "${ADDITIONAL_DIR}" DIR_LENGTH)
        string(SUBSTRING "${ADDITIONAL_DIR}" ${POSITION} ${DIR_LENGTH}-${POSITION} DIR_TAIL)
        string(REGEX MATCHALL "^([/A-Za-z0-9_.@-]+)" CLEAR_TAIL ${DIR_TAIL})
        string(REGEX REPLACE "^${HOMEBREW_PREFIX}/Cellar/([A-Za-z0-9_]+).*$" "\\1" LIB_NAME ${CLEAR_TAIL})
        string(REGEX REPLACE ".*libexec(.*)/site-packages" "libexec/${LIB_NAME}\\1" NEW_SITE_DIR ${CLEAR_TAIL})

        install(DIRECTORY ${CLEAR_TAIL} DESTINATION ${CMAKE_INSTALL_PREFIX}/${NEW_SITE_DIR})

        #update the paths of the .pth files copied into the bundle
        get_filename_component(PTH_FILENAME ${PTH_FILE} NAME)
        install(CODE
            "file(WRITE
            ${CMAKE_INSTALL_LIBDIR}/${PYTHON_DIR_BASENAME}/site-packages/${PTH_FILENAME}
            \"../../../${NEW_SITE_DIR}/site-packages\"
            )"
            )
    endforeach(PTH_FILE)
```

yields something like,

```
╭─chris ssh mac-mini-fpa in /Users/chris/code/freecad-git/builds/1.0.0_py312
╰─λ fd --hidden --no-ignore --follow site-packages ../../installs/1.0.0/FreeCAD.app/                                    0 (4.178s)
../../installs/1.0.0/FreeCAD.app/Contents/Users/chris/homebrew/Cellar/coin3d@4.0.3_py312/4.0.3_1/lib/python3.12/site-packages/
../../installs/1.0.0/FreeCAD.app/Contents/lib/python3.12/site-packages/
╭─chris ssh mac-mini-fpa in /Users/chris/code/freecad-git/builds/1.0.0_py312
╰─λ l ../../installs/1.0.0/FreeCAD.app/Contents/lib/python3.12/site-packages/freecad-py-modules.pth                     0 (0.056s)
-rw-rw-rw-@ 1 chris  staff   108B Jan 12 18:13 ../../installs/1.0.0/FreeCAD.app/Contents/lib/python3.12/site-packages/freecad-py-modules.pth
╭─chris ssh mac-mini-fpa in /Users/chris/code/freecad-git/builds/1.0.0_py312
╰─λ cat ../../installs/1.0.0/FreeCAD.app/Contents/lib/python3.12/site-packages/freecad-py-modules.pth                   0 (0.017s)
../../..//Users/chris/homebrew/Cellar/coin3d@4.0.3_py312/4.0.3_1/lib/python3.12/site-packages//site-packages⏎
╭─chris ssh mac-mini-fpa in /Users/chris/code/freecad-git/builds/1.0.0_py312
╰─λ # TODO: update cmake logic for mac app bundling to properly copy the  necessary python modules into the app bundle  0 (0.010s)
╭─chris ssh mac-mini-fpa in /Users/chris/code/freecad-git/builds/1.0.0_py312
╰─λ # NOTE: maybe first just attempt to manually copy the modules                                                       0 (0.010s)
╭─chris ssh mac-mini-fpa in /Users/chris/code/freecad-git/builds/1.0.0_py312
╰─λ                                                                                                                     0 (0.010s)

```

which is no good.