FreeCAD/homebrew-freecad #294: Freecad not running on apple silicon

deveshsnair, 2022-02-05:
Hi I can't get to install freecad on apple silicon the software is not downloading through homebrew also
[image]
.

ipatch, 2022-02-05:
hello, 👋

that's a really old version of freecad. if you're looking a ready made version and you have an M1 based CPU your best is to run a prebuilt `FreeCAD.app` apple app bundle from one of the two sources, and run the bundle via rosetta or compile freecad from source with certain QT components disabled until the upstream source has support for QT 6.2 (web engine)

ipatch, 2022-02-05:
https://github.com/FreeCAD/homebrew-freecad/releases/tag/0.20-27223.dmg

👆 that's a really recent bundle i created about a week ago. you'll have to edit one file to get the addon manager going but most if not all things should _just work™️_

there's an official conda release 👇 of freecad for mac as well, you're mileage may vary,

https://github.com/FreeCAD/FreeCAD-Bundle/releases

deveshsnair, 2022-02-05:
Cool, thanks!

DEVESH S NAIR

ipatch, 2022-02-05:
closing issue. will/can reopen is need be. i hope one of those releases works out for you. 👍

closed 2022-02-05 by ipatch