FreeCAD/homebrew-freecad #304: Latest HEAD is r28503 but the latest homebrew-freecad release r28286? (who not use HEAD?)

luzpaz, 2022-03-28:
@ipatch just saw this get released: https://github.com/FreeCAD/homebrew-freecad/releases/tag/0.20.28286-git-master

Thank for doing that. But I'm wondering why you're not compiling the latest HEAD?

CC @PLchris would you consider testing this build since conda is stalled?

ipatch, 2022-03-28:
this release is produced from a `git clone` of the main freecad/freecad master branch on github.

that revision was the tip of the head when i decided to build the mac app bundle at that time. these releases are manually assembled on various vm's i've setup to test freecad with. i'm **not** using `brew install freecad` to generate these releases at this time, as there are still some issues that need to be ironed out with `brew install freecad` most noticeably the issue with versioned formula ie. **pyside@2** the default behavior for homebrew is to not link versioned formula so using the github actions to produce a `FreeCAD.app` bundle will generate a non working app bundle. however if a `$USER` runs `brew link [VERSIONED_FORMULA]` then `brew install freecad` should generate a working `FreeCAD.app` bundle and one ($USER) can copy such  bundle to their `/Applications` dir if they so choose.

ipatch, 2022-03-29:
> can you do a fresh git pull of freecad/freecad master and make a more up to date release

yeah i can, but give me a few. i'll try and publish an updated release by the end of the day. and for subsequent releases i can try and probably publish a new release ~ once a week. (looking like on sundays or something of that nature).

ipatch, 2022-03-31:
here you go.

https://github.com/FreeCAD/homebrew-freecad/releases/tag/0.20.28575-git-master

i'm gunna close this issue now. if a newer release is required just open another issue, or reopen and comment on this issue please.

luzpaz, 2022-03-31:
@plchris :point_up_2: 

Thanks @ipatch !!