FreeCAD/homebrew-freecad #316: qtwebengine arm support with qt v5.15.3 ...may be possible. 🤞

ipatch, 2022-05-12:
surfing 🏄‍♂️ through the homebrew discussions i came across the below post,

https://github.com/Homebrew/discussions/discussions/1880

which contains the following link.

https://bugreports.qt.io/browse/QTBUG-85279?focusedCommentId=562355&page=com.atlassian.jira.plugin.system.issuetabpanels%3Acomment-tabpanel#comment-562355

it may be possible to add qtwebengine support with the current qt v5.15.3

i know several people have mentioned no **native** start workbench support for apple m1.

not sure if the qt devs plan on adding support for v5.15.3 but i'll leave this here for the brave.

my access to an m1 box is spotty at best, but it seems it may be possible with a custom qt formula.

luzpaz, 2022-07-01:
CC @realthunder @looooo just a heads up

ipatch, 2022-08-17:
the latest qt v5.15.5 supports webengine for m1, if anyone has an issue, comment and i'll reopen.

closed 2022-08-17 by ipatch