FreeCAD/homebrew-freecad #322: Please make a v0.20beta release

luzpaz, 2022-05-29:
Which can be included in https://github.com/FreeCAD/FreeCAD/releases/tag/0.20beta1

ipatch, 2022-05-29:
sure give me a little bit. is there specific place you'd like me to upload the finished dmg?

luzpaz, 2022-05-29:
If you have permissions, then to https://github.com/FreeCAD/FreeCAD/releases/tag/0.20beta1  
If not just ping me a URL to download it from and I can upload it there. 

CC @donovaly (just keeping you in the loop)

ipatch, 2022-05-30:
don't have write permissions to that upstream releases page. feel free to attach the dmg i just uploaded

https://github.com/FreeCAD/homebrew-freecad/releases/tag/0.20beta1

luzpaz, 2022-05-30:
Thanks @ipatch!  
Added to https://github.com/FreeCAD/FreeCAD/releases/tag/0.20beta1

Closing

closed 2022-05-30 by luzpaz

JohnOCFII, 2022-06-02:
Just FYI -- using @ipatch HomeBrew Beta release (28901) on my M1 Mac, the app runs well via Rosetta (at least Sketcher and PartDesign).  The Start page is fully populated.  The Mac "App Package" is 1.08GB and launches in about 6.6 seconds.  Using the recent weekly Conda build (28929) the Start page is blank.  The Mac "App Package" is 2.46GB and launches in about 8 seconds.