FreeCAD/homebrew-freecad #325: Add/update FreeSpacenav libs (spacemouse support) in to downstream homebrew

luzpaz, 2022-06-07:
These libs are important for spacemouse support:

* https://github.com/FreeSpacenav/libspnav/ (this is in homebrew but needs to be bumped to v1.0)
* https://github.com/FreeSpacenav/spacenavd
* (lower priority) https://github.com/FreeSpacenav/spnavcfg

Can we get them added in to downstream homebrew? 

<table>
  <tr>
    <td valign="top">[image]
    </td>
    <td valign="top">[image]
    </td>
    <td valign="top">[image]
    </td>
  </tr>
</table>

ipatch, 2022-06-07:
it looks like `spacenavcfg` is built on top of `spacenavd` (please correct me if i'm wrong) and spacenavd is not macos compatible

https://github.com/FreeSpacenav/spacenavd/issues/70

luzpaz, 2022-06-07:
Sorry, I didn't do my homework on that.