FreeCAD/homebrew-freecad #327: FreeCAD 0.20 was released on 14 June 2022

ageeye, 2022-07-02:
Stable point still to 0.19.3. This should be update to 0.20.0.

ipatch, 2022-07-03:
@ageeye

i'm not opposed to you being a maintainer of this tap if you want to work on some of these issues. i don't think i have the group permissions to add you as a maintainer, but i'm sure if you put a request in the forums someone could add you. also feel free to submit a PR if you'd like.

ageeye, 2022-07-04:
@ipatch 

I will support you with PRs. Four-eye-principle is the best solution.

ipatch, 2022-07-04:
sure, sounds great. 🤓

ipatch, 2022-07-08:
started scaffolding out an update for this...

https://github.com/FreeCAD/homebrew-freecad/compare/master...ipatch:homebrew-freecad:ipatch.freecad.v0.20.0

ipatch, 2022-07-08:
when we make the switch for the 0.20 release in the freecad.rb formula we're either going to have to implement our own boost-python3 formula that relies on python 3.10.x and is synced with the same version as the boost in this case 1.79 or submit a PR for upstream boost-python3 to work with python 3.10.x or even better have it work with python 3.9 and 3.10

ageeye, 2022-07-08:
That's a good question. The last commit in the upstream of boost-python3 is less than a week old. I can't believe to stay at python@3.9 for no reason. 

@SMillerDev Can you say something about the background. Thanks already.

SMillerDev, 2022-07-08:
Homebrew core dependencies are slowly moving to 3.10 but still need more work to ensure we don't introduce conflicts with mixed dependencies. But in your tap you can do whatever you want.

ipatch, 2022-07-08:
if you look at certain homebrew-core formula you'll notice some of them are setup to build against both python@3.9 and python@3.10 if you would get boost-python3 to build against both version of python that'd help.

ageeye, 2022-07-09:
> Homebrew core dependencies are slowly moving to 3.10 but still need more work to ensure we don't introduce conflicts with mixed dependencies. But in your tap you can do whatever you want.

Thanks for your very fast feedback. Two more questions. The dependencies are purely limited to the formula of Homebrew-core? That doesn't look so galactic at first glance:

```
Level: 0, boost-python3
Level: 1, rdkit
Level: 1, libtorrent-rasterbar
Level: 1, graph-tool
Level: 1, eigenpy
Level: 1, openimageio
Level: 2, btfs
Level: 2, libtorrent-rakshasa
Level: 3, rtorrent
```

What would the process be? Would it all be individual PRs? Then it's probably more work than it looks.

ageeye, 2022-07-09:
> if you look at certain homebrew-core formula you'll notice some of them are setup to build against both python@3.9 and python@3.10 if you would get boost-python3 to build against both version of python that'd help.

That sounds interesting. Do you have ad hoc an example?

SMillerDev, 2022-07-09:
> Then it's probably more work than it looks.

It's a ton of work because any dependency that is also a dependency of something that depends on python@3.9 will need to be migrated at the same time.

ageeye, 2022-07-09:
This is what I analyzed above. I think we will first create our own formula.

ipatch, 2022-08-17:
with the below commit i'm pretty confident in saying `brew install freecad/freecad/freecad` should work on catalina mojave bigsur intel and i've run the cmd on a m1 box i have access to. that is running monterey, and so far so good. it should install the latest stable version of freecad v0.20

https://github.com/FreeCAD/homebrew-freecad/commit/addab68821f9c96272c24c1540edfd0cbb486953

closed 2022-08-17 by ipatch