FreeCAD/homebrew-freecad #328: Homebrew update QT@5 to 5.15.5

ageeye, 2022-07-02:
Since some time the original qt@5 formula have been using. Now is this update to 5.15.5. This means that the following formula have to update shiboken & pyside2.

```
  url "https://download.qt.io/official_releases/QtForPython/pyside2/PySide2-5.15.5-src/pyside-setup-opensource-src-5.15.5.zip"
  sha256 "d1c61308c53636823c1d0662f410966e4a57c2681b551003e458b2cc65902c41"
```

ageeye, 2022-07-06:
@ipatch Currently I have no good env to test it. It was great, if you have time to test my PR. Or I will test it on weekend with system of my wife.

ipatch, 2022-07-10:
@ageeye 

noticed you submitted a PR to update the qt@5 formula in homebrew-core to support / patch the qtwebeninge cmake thing with the latest qt v5.15.5 tarball.

when you're building freecad, does your cmake _summary report_ show _Qt5WebEngineWidgets at 5.15.10_ because i reinstalled qt@5 via homebrew core this afternoon (july 10, 2022) so not sure if that's what i'm suppose to be seeing via your recent patch of the upstream qt formula or is my install still inferring the version mistmatch?

below is a snippet of my _summary report_

```
==============
Summary report
==============

-- Build type:
-- Compiler:            /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ (12.0.0.12000032)
-- Flags:               -Wall -Wextra -Wpedantic -Wno-write-strings  -Wno-undefined-var-template
-- Standard:            Requires C++17
-- Python:              3.10.5 [/usr/local/Frameworks/Python.framework/Versions/3.10/bin/python3.10] [.abi3]
-- PCL:                 not enabled
-- pybind11:            not enabled
-- Boost:               1.79.0 (1.79.0)
-- XercesC:             3.2.3 [/usr/local/lib/libxerces-c.dylib] [/usr/local/include]
-- ZLIB:                1.2.11
-- PyCXX:               6.2.8 [/opt/code/github/public/forks/freecad-git/freecad.src/src]
-- OCC:                 7.6.2 [TKFillet;TKMesh;TKernel;TKG2d;TKG3d;TKMath;TKIGES;TKSTL;TKShHealing;TKXSBase;TKBool;TKBO;TKBRep;TKTopAlgo;TKGeomAlgo;TKGeomBase;TKOffset;TKPrim;TKSTEPBase;TKSTEPAttr;TKSTEP209;TKSTEP;TKHLR;TKFeat] [/usr/local/lib] [/usr/local/include/opencascade]
-- SMESH:               build internal
--  MEDFile:            4.1.1 [/usr/local/lib/libmedC.dylib;/usr/local/lib/libmed.dylib] [/usr/local/include]
--  HDF5:               1.12.2
--  VTK:                9.1.0
-- NETGEN:              not enabled
-- SWIG:                4.0.2
-- Eigen3               3.4.0
-- Qt5Core:             5.15.5
-- Qt5Network:          5.15.5
-- Qt5Xml:              5.15.5
-- Qt5XmlPatterns:      5.15.5
-- Qt5Widgets:          5.15.5
-- Qt5PrintSupport:     5.15.5
-- Qt5OpenGL:           5.15.5
-- Qt5Svg:              5.15.5
-- Qt5UiTools:          5.15.5
-- Qt5Concurrent:       5.15.5
-- Qt5WebEngineWidgets: 5.15.10
-- Designer plugin:     not built per user request (BUILD_DESIGNER_PLUGIN)
-- Shiboken2:           5.15.5 [/usr/local/opt/pyside@2/lib/cmake/Shiboken2-5.15.5] [/usr/local/opt/pyside@2/include/shiboken2;/usr/local/opt/python@3.10/Frameworks/Python.framework/Headers]
-- PySide2:             5.15.5 [/usr/local/opt/pyside@2/include/PySide2]
-- PySide2Tools:        [/usr/local/opt/qt@5/bin/uic] [/usr/local/opt/qt@5/bin/rcc]
-- Freetype:            2.12.1
-- OpenGL:              /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk/System/Library/Frameworks/OpenGL.framework
-- OpenGLU:             [/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk/System/Library/Frameworks/OpenGL.framework][/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk/System/Library/Frameworks/OpenGL.framework]
-- Coin3D:              4.0.0 [/usr/local/Cellar/coin3d/4.0.0_3/lib/libCoin.dylib] [/usr/local/Cellar/coin3d/4.0.0_3/include]
-- Pivy:                0.6.6
-- SPNAV:               [/usr/local/lib/libspnav.dylib] [/usr/local/include]
-- Matplotlib:          3.5.2
-- Rift:                not enabled (BUILD_VR)
-- Doxygen:             1.9.4
--  Language:           English
--  Coin3D_DOC:         not found
=================================================
Now run 'cmake --build /opt/code/github/public/forks/freecad-git/builds/build.0.20.release' to build FreeCAD
=================================================

-- Configuring done
-- Generating done
-- Build files have been written to: /opt/code/github/public/forks/freecad-git/builds/build.0.20.release
```

ageeye, 2022-07-11:
@ipatch 

Do you use revision 1 of qt@5? [This](https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/qt@5.rb#L240-L242) should fix it. Other solution is [this](https://github.com/ageeye/homebrew-freecad/blob/gift/Formula/qt5155.rb#L115).

ipatch, 2022-07-11:
> @ipatch
>
> Do you use revision 1 of qt@5? [This](https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/qt@5.rb?rgh-link-date=2022-07-11T04%3A45%3A02Z#L240-L242) should fix it. Other solution is [...]

yes on my personal catalina box i see the revision number you reference, as seen below, and the sha256 sums match for the bottles.

https://github.com/ipatch/homebrew-core/blob/master/Formula/qt%405.rb

closed 2022-07-12 by ipatch

ipatch, 2022-07-13:
@ageeye 

for reasons i don't know, my qt@5 installation use the latest version you submitted a while back with your PR to fix the version mismatch for qtwebengine is still showing a lot of references on my install of `qt@5` from the upstream bottle 🤷‍♂️

```
fd -HIiL -t f qtwebengine $bp/Cellar/qt@5/5.15.5_1/ | rg 5.15.10                                                                      0 (02:35.307)
Is a directory (os error 21)
/usr/local/Cellar/qt@5/5.15.5_1/include/QtWebEngineCore/5.15.10/QtWebEngineCore/private/qtwebenginecore-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/include/QtWebEngineCore/5.15.10/QtWebEngineCore/private/qtwebenginecoreglobal_p.h
/usr/local/Cellar/qt@5/5.15.5_1/include/QtWebEngine/5.15.10/QtWebEngine/private/qtwebengine-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/include/QtWebEngine/5.15.10/QtWebEngine/private/qtwebengineglobal_p.h
/usr/local/Cellar/qt@5/5.15.5_1/Frameworks/QtWebEngine.framework/Headers/5.15.10/QtWebEngine/private/qtwebengine-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/Frameworks/QtWebEngine.framework/Headers/5.15.10/QtWebEngine/private/qtwebengineglobal_p.h
/usr/local/Cellar/qt@5/5.15.5_1/Frameworks/QtWebEngine.framework/Versions/5/Headers/5.15.10/QtWebEngine/private/qtwebengine-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/Frameworks/QtWebEngine.framework/Versions/5/Headers/5.15.10/QtWebEngine/private/qtwebengineglobal_p.h
/usr/local/Cellar/qt@5/5.15.5_1/Frameworks/QtWebEngine.framework/Versions/Current/Headers/5.15.10/QtWebEngine/private/qtwebengine-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/Frameworks/QtWebEngine.framework/Versions/Current/Headers/5.15.10/QtWebEngine/private/qtwebengineglobal_p.h
/usr/local/Cellar/qt@5/5.15.5_1/include/QtWebEngineWidgets/5.15.10/QtWebEngineWidgets/private/qtwebenginewidgets-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/Frameworks/QtWebEngineWidgets.framework/Headers/5.15.10/QtWebEngineWidgets/private/qtwebenginewidgets-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/Frameworks/QtWebEngineWidgets.framework/Versions/5/Headers/5.15.10/QtWebEngineWidgets/private/qtwebenginewidgets-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/Frameworks/QtWebEngineWidgets.framework/Versions/Current/Headers/5.15.10/QtWebEngineWidgets/private/qtwebenginewidgets-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/Frameworks/QtWebEngineCore.framework/Headers/5.15.10/QtWebEngineCore/private/qtwebenginecore-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/Frameworks/QtWebEngineCore.framework/Headers/5.15.10/QtWebEngineCore/private/qtwebenginecoreglobal_p.h
/usr/local/Cellar/qt@5/5.15.5_1/Frameworks/QtWebEngineCore.framework/Versions/Current/Headers/5.15.10/QtWebEngineCore/private/qtwebenginecore-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/Frameworks/QtWebEngineCore.framework/Versions/Current/Headers/5.15.10/QtWebEngineCore/private/qtwebenginecoreglobal_p.h
/usr/local/Cellar/qt@5/5.15.5_1/Frameworks/QtWebEngineCore.framework/Versions/5/Headers/5.15.10/QtWebEngineCore/private/qtwebenginecore-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/Frameworks/QtWebEngineCore.framework/Versions/5/Headers/5.15.10/QtWebEngineCore/private/qtwebenginecoreglobal_p.h
/usr/local/Cellar/qt@5/5.15.5_1/lib/QtWebEngine.framework/Headers/5.15.10/QtWebEngine/private/qtwebengineglobal_p.h
/usr/local/Cellar/qt@5/5.15.5_1/lib/QtWebEngine.framework/Headers/5.15.10/QtWebEngine/private/qtwebengine-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/lib/QtWebEngine.framework/Versions/Current/Headers/5.15.10/QtWebEngine/private/qtwebengineglobal_p.h
/usr/local/Cellar/qt@5/5.15.5_1/lib/QtWebEngine.framework/Versions/Current/Headers/5.15.10/QtWebEngine/private/qtwebengine-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/lib/QtWebEngine.framework/Versions/5/Headers/5.15.10/QtWebEngine/private/qtwebengine-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/lib/QtWebEngine.framework/Versions/5/Headers/5.15.10/QtWebEngine/private/qtwebengineglobal_p.h
/usr/local/Cellar/qt@5/5.15.5_1/lib/QtWebEngineWidgets.framework/Headers/5.15.10/QtWebEngineWidgets/private/qtwebenginewidgets-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/lib/QtWebEngineWidgets.framework/Versions/Current/Headers/5.15.10/QtWebEngineWidgets/private/qtwebenginewidgets-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/lib/QtWebEngineWidgets.framework/Versions/5/Headers/5.15.10/QtWebEngineWidgets/private/qtwebenginewidgets-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/lib/QtWebEngineCore.framework/Headers/5.15.10/QtWebEngineCore/private/qtwebenginecoreglobal_p.h
/usr/local/Cellar/qt@5/5.15.5_1/lib/QtWebEngineCore.framework/Headers/5.15.10/QtWebEngineCore/private/qtwebenginecore-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/lib/QtWebEngineCore.framework/Versions/5/Headers/5.15.10/QtWebEngineCore/private/qtwebenginecoreglobal_p.h
/usr/local/Cellar/qt@5/5.15.5_1/lib/QtWebEngineCore.framework/Versions/5/Headers/5.15.10/QtWebEngineCore/private/qtwebenginecore-config_p.h
/usr/local/Cellar/qt@5/5.15.5_1/lib/QtWebEngineCore.framework/Versions/Current/Headers/5.15.10/QtWebEngineCore/private/qtwebenginecoreglobal_p.h
/usr/local/Cellar/qt@5/5.15.5_1/lib/QtWebEngineCore.framework/Versions/Current/Headers/5.15.10/QtWebEngineCore/private/qtwebenginecore-config_p.h
```

ipatch, 2022-07-13:
lulz from what i can tell after you made your @ageeye  change it seems that didn't dispatch a rebuild of the bottles from what i can see looking at the version history of the formula. 🧐 please correct me if i'm looking at this incorrectly.

ipatch, 2022-07-13:
https://github.com/Homebrew/homebrew-core/issues/105739