FreeCAD/homebrew-freecad #337: `shiboken2@5.15.2` install fails

TristanDebrunner, 2022-07-13:
I am trying to `brew install --only-dependencies freecad` as described in the README, but the install fails at `shiboken2@5.15.2`.

I am running MacOS Monterey on a MacBook Pro with M1 Max.

Output of `brew install freecad/freecad/shiboken2@5.15.2`
``` ==> Downloading https://download.qt.io/official_releases/QtForPython/pyside2/PySide2-5.15.2-src/pyside-setup-opensource-src-5.15.2.tar.xz
Already downloaded: /Users/tristan/Library/Caches/Homebrew/downloads/1250b0074c8d43c55c7efb1d9dcefaa1c469a48527b2cb08a59a6c9d2fb02e8a--pyside-setup-opensource-src-5.15.2.tar.xz
==> Installing shiboken2@5.15.2 from freecad/freecad
==> cmake -DBUILD_TYPE=Release -DBUILD_TESTS:BOOL=OFF -DPYTHON_EXECUTABLE=/opt/homebrew/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/bin/python3.9 -DPYTHON_LIBRARY=/opt/homebrew/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/libpython3.9.dylib -DPYTHON_INCLUDE_DIR=
==> make -j10 install
Last 15 lines from /Users/tristan/Library/Logs/Homebrew/shiboken2@5.15.2/02.make:
make[1]: *** [libshiboken/CMakeFiles/libshiboken.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 77%] Building CXX object ApiExtractor/CMakeFiles/apiextractor.dir/docparser.cpp.o
cd /tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/macbuild5.15.2/ApiExtractor && /opt/homebrew/Library/Homebrew/shims/mac/super/clang++ -DCMAKE_CXX_COMPILER=\"/opt/homebrew/Library/Homebrew/shims/mac/super/clang++\" -DHAVE_LIBXSLT -DHAVE_QTXMLPATTERNS -DQT_CORE_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_XMLPATTERNS_LIB -DQT_XML_LIB -I/tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/macbuild5.15.2/ApiExtractor/apiextractor_autogen/include -I/tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/sources/shiboken2/ApiExtractor -I/tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/macbuild5.15.2/ApiExtractor -I/tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/sources/shiboken2/ApiExtractor/parser -I/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/include/libxml2 -iframework /opt/homebrew/opt/qt@5/lib -isystem /opt/homebrew/opt/qt@5/lib/QtCore.framework/Headers -isystem /opt/homebrew/opt/qt@5/./mkspecs/macx-clang-arm64 -isystem /opt/homebrew/opt/qt@5/lib/QtXml.framework/Headers -isystem /opt/homebrew/opt/qt@5/lib/QtXmlPatterns.framework/Headers -isystem /opt/homebrew/opt/qt@5/lib/QtNetwork.framework/Headers -Wall -fvisibility=hidden -Wno-strict-aliasing -D QT_NO_CAST_FROM_ASCII -D QT_NO_CAST_TO_ASCII -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk   -fPIC -fPIC -std=gnu++11 -MD -MT ApiExtractor/CMakeFiles/apiextractor.dir/docparser.cpp.o -MF CMakeFiles/apiextractor.dir/docparser.cpp.o.d -o CMakeFiles/apiextractor.dir/docparser.cpp.o -c /tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/sources/shiboken2/ApiExtractor/docparser.cpp
[ 78%] Building CXX object ApiExtractor/CMakeFiles/apiextractor.dir/doxygenparser.cpp.o
cd /tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/macbuild5.15.2/ApiExtractor && /opt/homebrew/Library/Homebrew/shims/mac/super/clang++ -DCMAKE_CXX_COMPILER=\"/opt/homebrew/Library/Homebrew/shims/mac/super/clang++\" -DHAVE_LIBXSLT -DHAVE_QTXMLPATTERNS -DQT_CORE_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_XMLPATTERNS_LIB -DQT_XML_LIB -I/tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/macbuild5.15.2/ApiExtractor/apiextractor_autogen/include -I/tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/sources/shiboken2/ApiExtractor -I/tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/macbuild5.15.2/ApiExtractor -I/tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/sources/shiboken2/ApiExtractor/parser -I/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/include/libxml2 -iframework /opt/homebrew/opt/qt@5/lib -isystem /opt/homebrew/opt/qt@5/lib/QtCore.framework/Headers -isystem /opt/homebrew/opt/qt@5/./mkspecs/macx-clang-arm64 -isystem /opt/homebrew/opt/qt@5/lib/QtXml.framework/Headers -isystem /opt/homebrew/opt/qt@5/lib/QtXmlPatterns.framework/Headers -isystem /opt/homebrew/opt/qt@5/lib/QtNetwork.framework/Headers -Wall -fvisibility=hidden -Wno-strict-aliasing -D QT_NO_CAST_FROM_ASCII -D QT_NO_CAST_TO_ASCII -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk   -fPIC -fPIC -std=gnu++11 -MD -MT ApiExtractor/CMakeFiles/apiextractor.dir/doxygenparser.cpp.o -MF CMakeFiles/apiextractor.dir/doxygenparser.cpp.o.d -o CMakeFiles/apiextractor.dir/doxygenparser.cpp.o -c /tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/sources/shiboken2/ApiExtractor/doxygenparser.cpp
[ 80%] Building CXX object ApiExtractor/CMakeFiles/apiextractor.dir/qtdocparser.cpp.o
cd /tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/macbuild5.15.2/ApiExtractor && /opt/homebrew/Library/Homebrew/shims/mac/super/clang++ -DCMAKE_CXX_COMPILER=\"/opt/homebrew/Library/Homebrew/shims/mac/super/clang++\" -DHAVE_LIBXSLT -DHAVE_QTXMLPATTERNS -DQT_CORE_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_XMLPATTERNS_LIB -DQT_XML_LIB -I/tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/macbuild5.15.2/ApiExtractor/apiextractor_autogen/include -I/tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/sources/shiboken2/ApiExtractor -I/tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/macbuild5.15.2/ApiExtractor -I/tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/sources/shiboken2/ApiExtractor/parser -I/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/include/libxml2 -iframework /opt/homebrew/opt/qt@5/lib -isystem /opt/homebrew/opt/qt@5/lib/QtCore.framework/Headers -isystem /opt/homebrew/opt/qt@5/./mkspecs/macx-clang-arm64 -isystem /opt/homebrew/opt/qt@5/lib/QtXml.framework/Headers -isystem /opt/homebrew/opt/qt@5/lib/QtXmlPatterns.framework/Headers -isystem /opt/homebrew/opt/qt@5/lib/QtNetwork.framework/Headers -Wall -fvisibility=hidden -Wno-strict-aliasing -D QT_NO_CAST_FROM_ASCII -D QT_NO_CAST_TO_ASCII -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk   -fPIC -fPIC -std=gnu++11 -MD -MT ApiExtractor/CMakeFiles/apiextractor.dir/qtdocparser.cpp.o -MF CMakeFiles/apiextractor.dir/qtdocparser.cpp.o.d -o CMakeFiles/apiextractor.dir/qtdocparser.cpp.o -c /tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/sources/shiboken2/ApiExtractor/qtdocparser.cpp
[ 81%] Linking CXX static library libapiextractor.a
cd /tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/macbuild5.15.2/ApiExtractor && /opt/homebrew/Cellar/cmake/3.23.2/bin/cmake -P CMakeFiles/apiextractor.dir/cmake_clean_target.cmake
cd /tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/macbuild5.15.2/ApiExtractor && /opt/homebrew/Cellar/cmake/3.23.2/bin/cmake -E cmake_link_script CMakeFiles/apiextractor.dir/link.txt --verbose=1
/usr/bin/ar qc libapiextractor.a CMakeFiles/apiextractor.dir/apiextractor_autogen/mocs_compilation.cpp.o CMakeFiles/apiextractor.dir/apiextractor.cpp.o CMakeFiles/apiextractor.dir/abstractmetabuilder.cpp.o CMakeFiles/apiextractor.dir/abstractmetalang.cpp.o CMakeFiles/apiextractor.dir/fileout.cpp.o CMakeFiles/apiextractor.dir/graph.cpp.o CMakeFiles/apiextractor.dir/messages.cpp.o CMakeFiles/apiextractor.dir/propertyspec.cpp.o CMakeFiles/apiextractor.dir/reporthandler.cpp.o CMakeFiles/apiextractor.dir/sourcelocation.cpp.o CMakeFiles/apiextractor.dir/typeparser.cpp.o CMakeFiles/apiextractor.dir/typesystem.cpp.o CMakeFiles/apiextractor.dir/typesystemparser.cpp.o CMakeFiles/apiextractor.dir/include.cpp.o CMakeFiles/apiextractor.dir/typedatabase.cpp.o CMakeFiles/apiextractor.dir/clangparser/compilersupport.cpp.o CMakeFiles/apiextractor.dir/clangparser/clangparser.cpp.o CMakeFiles/apiextractor.dir/clangparser/clangbuilder.cpp.o CMakeFiles/apiextractor.dir/clangparser/clangdebugutils.cpp.o CMakeFiles/apiextractor.dir/clangparser/clangutils.cpp.o CMakeFiles/apiextractor.dir/parser/codemodel.cpp.o CMakeFiles/apiextractor.dir/parser/enumvalue.cpp.o CMakeFiles/apiextractor.dir/xmlutils.cpp.o CMakeFiles/apiextractor.dir/xmlutils_libxslt.cpp.o CMakeFiles/apiextractor.dir/xmlutils_qt.cpp.o CMakeFiles/apiextractor.dir/docparser.cpp.o CMakeFiles/apiextractor.dir/doxygenparser.cpp.o CMakeFiles/apiextractor.dir/qtdocparser.cpp.o
/usr/bin/ranlib libapiextractor.a
[ 81%] Built target apiextractor
make: *** [all] Error 2

Do not report this issue to Homebrew/brew or Homebrew/core!
```

ipatch, 2022-07-13:
try installing `shiboken2@5.15.5` instead of `shiboken2@5.15.2`

```
brew install shiboken2@5.15.5
```

unfortunately i haven't generated a shiboken bottle for that release/arch of macos so you'll have to build it from source for time being.

TristanDebrunner, 2022-07-14:
I tried all the versions except `freecad/freecad/shiboken2-python3_10@5.15.2`, and they have all failed. Here's the 5.15.5 output:
```
==> Downloading https://ghcr.io/v2/homebrew/core/llvm/manifests/14.0.6_1
######################################################################## 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/llvm/blobs/sha256:bb22d243ad001cd69c26ebe670684701ef03d9d3895773d30b907d90c645a744
==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:bb22d243ad001cd69c26ebe670684701ef03d9d3895773d30b907d90c645a744?se=2022-07-14T19%3A15%3A00Z&sig=FHhiQXe%2B%2Fe8ex1EyKv2hmeJJ193GBsR5w3YVGGFjdz4%3D&sp=r&spr=https&sr=b&sv=2019-12-12
######################################################################## 100.0%
==> Downloading https://download.qt.io/official_releases/QtForPython/pyside2/PySide2-5.15.5-src/pyside-setup-opensource-src-5.15.5.zip
Already downloaded: /Users/tristan/Library/Caches/Homebrew/downloads/7e478bd63671464e429850a09bc66eeda44c95ddd8ab0b66cfc8224ffd477cc9--pyside-setup-opensource-src-5.15.5.zip
==> Installing shiboken2@5.15.5 from freecad/freecad
==> Installing dependencies for freecad/freecad/shiboken2@5.15.5: llvm
==> Installing freecad/freecad/shiboken2@5.15.5 dependency: llvm
==> Pouring llvm--14.0.6_1.arm64_monterey.bottle.tar.gz
🍺  /opt/homebrew/Cellar/llvm/14.0.6_1: 5,851 files, 983.5MB
==> Installing freecad/freecad/shiboken2@5.15.5
==> Patching
Error: Failure while executing; `patch -g 0 -f -p1` exited with 1. Here's the output:
patching file sources/shiboken2/libshiboken/pep384impl.cpp
Hunk #1 FAILED at 707.
1 out of 1 hunk FAILED -- saving rejects to file sources/shiboken2/libshiboken/pep384impl.cpp.rej
patching file sources/shiboken2/libshiboken/sbknumpyarrayconverter.cpp
patch unexpectedly ends in middle of line
Hunk #1 FAILED at 116.
1 out of 1 hunk FAILED -- saving rejects to file sources/shiboken2/libshiboken/sbknumpyarrayconverter.cpp.rej
```

I tried to look at the files mentioned in the error message, but haven't been able to find them.

ipatch, 2022-07-15:
try with pr #344

TristanDebrunner, 2022-07-16:
That PR makes progress, I'm no longer erroring out on the patch step. The error is now in the make step, just like the original issue. I found this in the make log:
```
/tmp/shiboken2A5.15.5-20220715-48643-uc6jqc/pyside-setup-opensource-src-5.15.5/sources/shiboken2/libshiboken/sbknumpyarrayconverter.cpp:119:22: error: use of undeclared identifier 'NPY_ARRAY_UPDATEIFCOPY'
        if ((flags & NPY_ARRAY_UPDATEIFCOPY) != 0)
                     ^
1 error generated.
make[2]: *** [libshiboken/CMakeFiles/libshiboken.dir/sbknumpyarrayconverter.cpp.o] Error 1
```

I went back and looked at the log from the 5.15.2 install, and it has the same error:
```
/tmp/shiboken2A5.15.2-20220713-10029-10z34xd/pyside-setup-opensource-src-5.15.2/sources/shiboken2/libshiboken/sbknumpyarrayconverter.cpp:119:22: error: use of undeclared identifier 'NPY_ARRAY_UPDATEIFCOPY'
        if ((flags & NPY_ARRAY_UPDATEIFCOPY) != 0)
                     ^
1 error generated.
make[2]: *** [libshiboken/CMakeFiles/libshiboken.dir/sbknumpyarrayconverter.cpp.o] Error 1
```

Is there a numpy requirement that I may not be fulfilling? Not sure if this gives you useful info about my system:
```
$ brew list numpy
/opt/homebrew/Cellar/numpy/1.23.1/bin/f2py
/opt/homebrew/Cellar/numpy/1.23.1/bin/f2py3
/opt/homebrew/Cellar/numpy/1.23.1/bin/f2py3.10
/opt/homebrew/Cellar/numpy/1.23.1/bin/f2py3.9
/opt/homebrew/Cellar/numpy/1.23.1/lib/python3.10/ (862 files)
/opt/homebrew/Cellar/numpy/1.23.1/lib/python3.9/ (862 files)
```

plaes, 2022-07-16:
> The UPDATEIFCOPY array flag has been removed together with the enum NPY_ARRAY_UPDATEIFCOPY. The associated (and deprecated) PyArray_XDECREF_ERR was also removed. These were all deprecated in 1.14. They are replaced by WRITEBACKIFCOPY, that requires calling PyArray_ResoveWritebackIfCopy before the array is deallocated.

https://numpy.org/doc/stable/release/1.23.0-notes.html?highlight=npy_array_updateifcopy#expired-deprecations

ipatch, 2022-07-16:
> Apparently this symbol has been dropped from Numpy since 1.23:
> 
> > The UPDATEIFCOPY array flag has been removed together with the enum NPY_ARRAY_UPDATEIFCOPY. The associated (and deprecated) PyArray_XDECREF_ERR was also removed. These were all deprecated in 1.14. They are replaced by WRITEBACKIFCOPY, that requires calling PyArray_ResoveWritebackIfCopy before the array is deallocated.
> 
> 
> 
> https://numpy.org/doc/stable/release/1.23.0-notes.html?highlight=npy_array_updateifcopy#expired-deprecations

yeah thats what the patch file aims to resolve

- patch shiboken2 source tarball to be compatible with the numpy v1.23 api
- theres additional patch to allow the same tarball to build against python 3.10

both patches were inlined into the shiboken2@5.15.5

make sure youre attempting to install that version and not an older version of shiboken2

---

to complicate matters apparently some formatting is lost after ci modifies the shiboken2 formula to add the bottle blocks (that was fun edge case to figure out 😭)

so attempted to move the 2 patches out into separate files with separate commits to allow the shiboken2 formula file (tarball) ref the required patches and hopefully not run into ci issue above. also i used the github provided .patch for these two required patches.

IMHO, i have never run into such an issue patching software using the patch cmd and editing the formula files. not sure if there some edge case with the shiboken2 tarball but hopefully ill figure it out later this week. presently afk 📱

ipatch, 2022-07-17:
okay #345 should fix this, will require manually building of the bottles.

ipatch, 2022-07-18:
> Unfortunately, `brew install --only-dependencies freecad` still tries to use 5.15.2, which fails. Is there a way to force the use of 5.15.5 instead?

yeah, unfortunately until i or somebody else fixes the main `[freecad.rb](https://github.com/FreeCAD/homebrew-freecad/blob/6e973597e4063c32e7487c9d7dfa606a32517ab3/Formula/freecad.rb#L56)` formula file you'll have to manually build freecad using  cmake or install from a pre built binary package.

Vittorio1993, 2022-07-25:
> With the latest fixes, `brew install shiboken2@5.15.5` succeeds!
> 
> Unfortunately, `brew install --only-dependencies freecad` still tries to use 5.15.2, which fails. Is there a way to force the use of 5.15.5 instead?

Did you manage to build it manually?

TristanDebrunner, 2022-08-03:
Sorry, I had some other stuff come up that slowed down this project.

I was able to take the list of packages that `brew install --only-dependencies freecad` would've installed, and `brew install` them individually, replacing 5.15.2 with 5.15.5. I got through most of the build, but ran into issues with [`MakeMacBundleRelocatable.py`](https://github.com/FreeCAD/FreeCAD/blob/0.20/src/Tools/MakeMacBundleRelocatable.py) that I haven't fully resolved. That's not really a homebrew issue though.

ipatch, 2022-08-03:
> Sorry, I had some other stuff come up that slowed down this project.
> 
> I was able to take the list of packages that `brew install --only-dependencies freecad` would've installed, and `brew install` them individually, replacing 5.15.2 with 5.15.5. I got through most of the build, but ran into issues with [`MakeMacBundleRelocatable.py`](https://github.com/FreeCAD/FreeCAD/blob/0.20/src/Tools/MakeMacBundleRelocatable.py?rgh-link-date=2022-08-03T16%3A35%3A33Z) that I haven't fully resolved. That's not really a homebrew issue though.
> 
> UPDATE: Just saw #348. That seems to be the same problem I ran into

good to know that i'm not the only person running into that issue. 👍

ipatch, 2022-08-17:
> brew install --only-dependencies freecad

i believe me merging in the following commit should resolve the above mentioned issues,

https://github.com/FreeCAD/homebrew-freecad/commit/addab68821f9c96272c24c1540edfd0cbb486953

if you run into any more issues feel free to comment on this issue or even open a new & separate issue.

closed 2022-08-17 by ipatch