FreeCAD/homebrew-freecad #348: mac app bundling using cmake on m1

ipatch, 2022-07-28:
my `FreeCAD.app` mac app bundle crashes on launch after using cmake, make install to build the bundle. below is the crash report i see after the bundle crashes.

```shell
sw_vers

ProductName:    macOS
ProductVersion: 12.4
BuildVersion:   21F79
```

**crash report**

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               FreeCAD [83834]
Path:                  /Users/USER/*/FreeCAD.app/Contents/MacOS/FreeCAD
Identifier:            
Version:               0.20.0 (0.20.0)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               502

Date/Time:             2022-07-28 12:45:20.2686 -0500
OS Version:            macOS 12.4 (21F79)
Report Version:        12
Anonymous UUID:        5CC24BCD-B012-CF28-443B-7DF02AEE10CC

Time Awake Since Boot: 350000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_BAD_ACCESS (SIGKILL (Code Signature Invalid))
Exception Codes:       UNKNOWN_0x32 at 0x00000001034ac000
Exception Codes:       0x0000000000000032, 0x00000001034ac000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace CODESIGNING, Code 2 

VM Region Info: 0x1034ac000 is in 0x1034ac000-0x103654000;  bytes after start: 0  bytes before end: 1736703
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      mapped file                 1032f0000-1034ac000    [ 1776K] r--/rwx SM=COW  ...t_id=88272747
--->  mapped file                 1034ac000-103654000    [ 1696K] r-x/rwx SM=COW  ...t_id=8771e347
      VM_ALLOCATE (reserved)      103654000-1037a4000    [ 1344K] rw-/rwx SM=NUL  ...(unallocated)

Thread 0 Crashed:
0   dyld                          	       0x102a78008 dyld3::MachOFile::isMachO(Diagnostics&, unsigned long long) const + 20
1   dyld                          	       0x102a592cc dyld4::Loader::mapSegments(Diagnostics&, dyld4::RuntimeState&, char const*, unsigned long long, dyld4::Loader::CodeSignatureInFile const&, bool, dyld3::Array<dyld4::Loader::Region> const&, bool, bool, dyld4::Loader::FileValidationInfo const&) + 1096
2   dyld                          	       0x102a592cc dyld4::Loader::mapSegments(Diagnostics&, dyld4::RuntimeState&, char const*, unsigned long long, dyld4::Loader::CodeSignatureInFile const&, bool, dyld3::Array<dyld4::Loader::Region> const&, bool, bool, dyld4::Loader::FileValidationInfo const&) + 1096
3   dyld                          	       0x102a5eb88 invocation function for block in dyld4::JustInTimeLoader::makeJustInTimeLoaderDisk(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, bool, unsigned int) + 68
4   dyld                          	       0x102a5e528 dyld4::JustInTimeLoader::withRegions(dyld3::MachOAnalyzer const*, void (dyld3::Array<dyld4::Loader::Region> const&) block_pointer) + 292
5   dyld                          	       0x102a5eadc invocation function for block in dyld4::JustInTimeLoader::makeJustInTimeLoaderDisk(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, bool, unsigned int) + 480
6   dyld                          	       0x102a63d58 dyld4::SyscallDelegate::withReadOnlyMappedFile(Diagnostics&, char const*, bool, void (void const*, unsigned long, bool, dyld4::FileID const&, char const*) block_pointer) const + 132
7   dyld                          	       0x102a5e8c8 dyld4::JustInTimeLoader::makeJustInTimeLoaderDisk(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, bool, unsigned int) + 204
8   dyld                          	       0x102a586d0 invocation function for block in dyld4::Loader::getLoader(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&) + 1384
9   dyld                          	       0x102a58b40 invocation function for block in dyld4::Loader::forEachResolvedAtPathVar(dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, dyld4::ProcessConfig::PathOverrides::Type, bool&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) + 320
10  dyld                          	       0x102a45f98 dyld3::MachOFile::forEachLoadCommand(Diagnostics&, void (load_command const*, bool&) block_pointer) const + 168
11  dyld                          	       0x102a836d4 dyld3::MachOAnalyzer::forEachRPath(void (char const*, bool&) block_pointer) const + 128
12  dyld                          	       0x102a57b78 dyld4::Loader::forEachResolvedAtPathVar(dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, dyld4::ProcessConfig::PathOverrides::Type, bool&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) + 708
13  dyld                          	       0x102a577ec invocation function for block in dyld4::Loader::forEachPath(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) + 148
14  dyld                          	       0x102a4dee4 dyld4::ProcessConfig::PathOverrides::forEachPathVariant(char const*, dyld3::Platform, bool, bool&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) const + 672
15  dyld                          	       0x102a57740 dyld4::Loader::forEachPath(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) + 172
16  dyld                          	       0x102a57f60 dyld4::Loader::getLoader(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&) + 864
17  dyld                          	       0x102a5cb60 invocation function for block in dyld4::JustInTimeLoader::loadDependents(Diagnostics&, dyld4::RuntimeState&, dyld4::Loader::LoadOptions const&) + 380
18  dyld                          	       0x102a79264 invocation function for block in dyld3::MachOFile::forEachDependentDylib(void (char const*, bool, bool, bool, unsigned int, unsigned int, bool&) block_pointer) const + 148
19  dyld                          	       0x102a45f98 dyld3::MachOFile::forEachLoadCommand(Diagnostics&, void (load_command const*, bool&) block_pointer) const + 168
[... 142 lines cut ...]
  ],
  "experiments" : [

  ]
}
}

Model: Macmini9,1, BootROM 7459.121.3, proc 8:4:4 processors, 8 GB, SMC 
Graphics: Apple M1, Apple M1, Built-In
Display: LG HDR 4K, 3840 x 2160 (2160p/4K UHD 1 - Ultra High Definition), Main, MirrorOff, Online
Memory Module: LPDDR4
AirPort: Wi-Fi, wl0: Mar 23 2022 19:57:59 version 18.60.27.0.7.8.129 FWID 01-570be953
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: Magic Keyboard with Numeric Keypad
USB Device: USB30Bus
Thunderbolt Bus: Mac mini, Apple Inc.
Thunderbolt Bus: Mac mini, Apple Inc.
```

@ageeye 

you care to weigh in on this. i know you've been building app bundles with cmake on an m1?

ipatch, 2022-07-28:
https://gitlab.kitware.com/cmake/cmake/-/issues/21854

ipatch, 2022-07-29:
it seems there is a similar take on how mac app bundling is done, but the below approach uses a shell script instead of python but seems the project is making a m1 app bundle that works with m1 based macs.

https://github.com/apfelkraut/darktable/blob/9b672a2ab28a083c73ec5855d97b545d935d1f50/packaging/macosx/3_make_hb_darktable_package.sh

ipatch, 2022-07-31:
progress...

[image]

ipatch, 2022-08-01:
@ageeye 

you care to share cmake commands you're using to build a self contained mac app bundle for `FreeCAD.app` ?

TristanDebrunner, 2022-08-03:
> progress...

I'm running into the same problem. How did you get the app to open? Manually using `codesign`?

ipatch, 2022-08-03:
> > progress...
> 
> I'm running into the same problem. How did you get the app to open? Manually using `codesign`?

by not updating the loader paths for the bins and libs, but thus renders the app no relocatable.

TristanDebrunner, 2022-08-03:
ah. Is there a cmake option to skip that step?

ipatch, 2022-08-03:
> ah. Is there a cmake option to skip that step?

not that i'm aware of.

MatthiasWM, 2022-09-16:
Assuming that `src/Tools/MakeMacBundleRelocatable.py` runs all the way through (as of today, it does not), running 
```
find FreeCAD.app/Contents/lib/ -exec codesign --force -s -  \{\} \;
```
will codesign all changed libraries after they have been relocated and their paths have been fixed. It is brute force, but at least for development on the local machine, it makes `FreeCAD.app` relocatable.

I patched `src/Tools/MakeMacBundleRelocatable.py` to find all the paths to Qt frameworks, but it's a horrible hack and maybe even linking the wrong libraries. But in any case, the resulting bundle runs and can be moved around.
```
--- a/src/Tools/MakeMacBundleRelocatable.py
+++ b/src/Tools/MakeMacBundleRelocatable.py
@@ -161,3 +162,8 @@ def create_dep_nodes(install_names, search_paths):
             if not path:
                 path = install_path
 
+        qt_path = []
+        #qt_path.append("/opt/homebrew/Cellar/qt@5/5.15.5_2/lib/" + lib_name + ".framework")
+        qt_path.append("/opt/homebrew/Cellar/qt/6.3.1_3/lib/" + lib_name + ".framework")
+        if not path:
+            path = get_path(lib_name, qt_path)
```

ipatch, 2022-09-16:
@MatthiasWM 

you might want to doublecheck that. it looks like you commented out the path to qt@5 which freecad uses, and then you added the path to qt ie. qt@6 which freecad currently does not support.

MatthiasWM, 2022-09-20:
> you might want to doublecheck that. it looks like you commented out the path to qt@5 which freecad uses, and then you added the path to qt ie. qt@6 which freecad currently does not support.

Yes, you are right of course. I published the wrong patch. The Qt5 link is the correct one.