FreeCAD/homebrew-freecad #361: Freecad formula failing on M1 mac

TristanDebrunner, 2022-08-17:
When I run the most recent homebrew formula on my MacBook Pro with M1 MAX, is fails with the error below. I don't have any other hardware to test this on, so it might not be M1 specific.
```
$ brew install --only-dependencies freecad
Error: Failed to load formula: freecad
freecad: /opt/homebrew/Library/Taps/freecad/homebrew-freecad/Formula/freecad.rb:15: syntax error, unexpected <<, expecting end
<<<<<<< HEAD
^~
/opt/homebrew/Library/Taps/freecad/homebrew-freecad/Formula/freecad.rb:18: syntax error, unexpected and, expecting ')'
...update file to work with m1 and 0.20 release (#358))
...                            ^~~
/opt/homebrew/Library/Taps/freecad/homebrew-freecad/Formula/freecad.rb:21: syntax error, unexpected tIDENTIFIER, expecting ')'
  depends_on "hdf5" => :build
  ^~~~~~~~~~
```

The same error occurs without the `--only-dependencies` flag

ipatch, 2022-08-17:
i believe your using an outdated version of the tap possibly brew as well. make sure your homebrew-freecad tap is update. you can run `brew update` and should pull in all the latest changes.

you can also run,

```
cd (brew --repo freecad/freecad)
```

and then run,

```
git rev-list --count HEAD
```

and you should see something like `481` as the output.

also make sure your tap is on the master branch.

and looking at your error message it looks you might have make some changes to your local repo/tap of homebrew-freecad hence the `<<<<<<< HEAD` usually is an indicator of a failed merge (git) or a merge/rebase that hasn't been completed.

probably the quickest solution to get you back on track would be to run the below cmd in the root of the homebrew-freecad repo which the cmd i posted above should get you there. and then git reset the repo with the below cmd.

```
git reset --hard origin
```

note that will wipe out any local changes you've made so you might want to stash them with a `git stash` if their important to you.

and that cmd also implies your `origin` is set to,

```
https://github.com/freecad/homebrew-freecad.git
```

the capitalization / `.git` at the end of the url shouldn't make a difference. hope all that helps.

dirkolbrich, 2022-08-18:
Updating an already tapped freecad/freecad also gives errors. Before that updating was no problem.

```zsh
➜  ~ brew update
error: could not apply 6d2c863... freecad.rb update file to work with m1 and 0.20 release (#358)
hint: Resolve all conflicts manually, mark them as resolved with
hint: "git add/rm <conflicted_files>", then run "git rebase --continue".
hint: You can instead skip this commit: run "git rebase --skip".
hint: To abort and get back to the state before "git rebase", run "git rebase --abort".
Could not apply 6d2c863... freecad.rb update file to work with m1 and 0.20 release (#358)
No stash entries found.
Updated 1 tap (freecad/freecad).
```

dirkolbrich, 2022-08-18:
There seems to be some syntax error introduced with the latest commit:

```zsh
➜  ~ brew info freecad/freecad/freecad
Error: Failed to load formula: freecad/freecad/freecad
freecad/freecad/freecad: /opt/homebrew/Library/Taps/freecad/homebrew-freecad/Formula/freecad.rb:15: syntax error, unexpected <<, expecting end
<<<<<<< HEAD
^~
/opt/homebrew/Library/Taps/freecad/homebrew-freecad/Formula/freecad.rb:18: syntax error, unexpected and, expecting ')'
...update file to work with m1 and 0.20 release (#358))
...                            ^~~
/opt/homebrew/Library/Taps/freecad/homebrew-freecad/Formula/freecad.rb:21: syntax error, unexpected tIDENTIFIER, expecting ')'
  depends_on "hdf5" => :build
  ^~~~~~~~~~
Warning: Treating freecad/freecad/freecad as a cask.
==> freecad: 0.20.0
https://www.freecad.org/
Not installed
From: https://github.com/freecad/homebrew-freecad/blob/HEAD/Casks/freecad.rb
==> Name
FreeCAD
==> Description
3D parametric modler
==> Artifacts
FreeCAD.app -> FreeCAD_0.20_release.app (App)
==> Analytics
install: 655 (30 days), 1,932 (90 days), 7,076 (365 days)
```

ipatch, 2022-08-18:
@dirkolbrich 

i made a asciicast that will hopefully help you with getting your repo / homebrew tap setup properly.

https://asciinema.org/a/515723

and also converted the asciicast to a gif so you watch within the repo.
[image]

TristanDebrunner, 2022-08-18:
I don't think I've done anything that would modify the brew repo, but that does seem to have been the issue. 🤷  When I `cd`ed in it was mid rebase. The `git reset` seems to have fixed it

dirkolbrich, 2022-08-19:
Me neither, haven‘t touched any homebrew related repo manually apart from the daily `brew update`.

ipatch, 2022-08-19:
and i'll leave this issue open for another day or two in case somebody else ran into this issue, then i'll close it.

closed 2022-08-27 by ipatch