FreeCAD/homebrew-freecad #367: Failing to build from HEAD

dannystaple, 2022-11-04:
I'm a bit of a newbie with homebrew tap installs, but it looks like a patch failed during the build. I am on an X86 mac, with MacOS Monterey 12.6

My commands were:
```
> brew tap freecad/freecad
> brew install --HEAD freecad
```
The install failed with the following (redacted all but the relevant lines for now):
```
==> Downloading https://raw.githubusercontent.com/FreeCAD/homebrew-freecad/a4b71
Already downloaded: /Users/danielstaple/Library/Caches/Homebrew/downloads/cb41ac94e83f4c532bb20e708cf594b3ac72b0f836f7ecce36d40f99787d3dd9--freecad-0.20.1-macos-10.15-sdk.patch
==> Downloading https://raw.githubusercontent.com/FreeCAD/homebrew-freecad/a4b71
Already downloaded: /Users/danielstaple/Library/Caches/Homebrew/downloads/4fa80a175f519cda0d7e502602499c920e22a6290a361b67b4f3f582bccb8dc5--freecad-0.20.1-macos-app-bundling.patch
.
.
.
.
==> Cloning https://github.com/freecad/FreeCAD.git
Updating /Users/danielstaple/Library/Caches/Homebrew/freecad@0.20.1--git
From https://github.com/freecad/FreeCAD
   92819be7fd..7c3e5d3b34  master     -> origin/master
==> Checking out branch master
Already on 'master'
Your branch is behind 'origin/master' by 2 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)
HEAD is now at 7c3e5d3b34 [Typo]Gui: fix file indentation + improve tootip
.
.
.
==> Installing freecad@0.20.1 from freecad/freecad
==> Patching
==> Applying freecad-0.20.1-macos-10.15-sdk.patch
patching file src/Base/UnitsSchemaInternal.h
patching file src/Base/UnitsSchemaMKS.h
==> Applying freecad-0.20.1-macos-app-bundling.patch
patching file src/MacAppBundle/CMakeLists.txt
Hunk #1 FAILED at 114.
Hunk #2 FAILED at 132.
2 out of 2 hunks FAILED -- saving rejects to file src/MacAppBundle/CMakeLists.txt.rej
```

I am guessing that the sdk patch CMakeLists no longer match the underlying file in the FreeCAD repo head?
Trying to find the rejects file, but not found this yet:

```> ls /Users/danielstaple/Library/Caches/Homebrew/freecad@0.20.1--git/src/MacAppBundle/
CMakeLists.txt  DiskImage/      FreeCAD.app/
```

Looking at the content of `/Users/danielstaple/Library/Caches/Homebrew/downloads/4fa80a175f519cda0d7e502602499c920e22a6290a361b67b4f3f582bccb8dc5--freecad-0.20.1-macos-app-bundling.patch` the failed hunk:

```
--
 src/MacAppBundle/CMakeLists.txt | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

diff --git a/src/MacAppBundle/CMakeLists.txt b/src/MacAppBundle/CMakeLists.txt
index 07876170dec0..3dfb655d6bac 100644
--- a/src/MacAppBundle/CMakeLists.txt
+++ b/src/MacAppBundle/CMakeLists.txt
@@ -114,7 +114,7 @@ install(CODE "execute_process(COMMAND chmod -R a+w ${CMAKE_INSTALL_LIBDIR})")
 get_filename_component(APP_PATH ${CMAKE_INSTALL_PREFIX} PATH)
 
 execute_process(
-    COMMAND find /usr/local/Cellar/icu4c -name pkgconfig
+    COMMAND find ${HOMEBREW_PREFIX}/Cellar/icu4c -name pkgconfig
     RESULT_VARIABLE CMD_ERROR
     OUTPUT_VARIABLE CONFIG_ICU)
 set(ENV{PKG_CONFIG_PATH} "$ENV{PKG_CONFIG_PATH}:${CONFIG_ICU}")
@@ -132,6 +132,6 @@ install(CODE
     execute_process(
         COMMAND python3
         ${CMAKE_SOURCE_DIR}/src/Tools/MakeMacBundleRelocatable.py
-        ${APP_PATH} ${HOMEBREW_PREFIX}${MACPORTS_PREFIX}/lib ${ICU_PREFIX}/lib/ /usr/local/opt ${CONFIG_NGLIB} ${Qt5Core_DIR}/../../.. ${XCTEST_PATH} ${WEBKIT_FRAMEWORK_DIR}
+        ${APP_PATH} ${HOMEBREW_PREFIX}${MACPORTS_PREFIX}/lib ${HOMEBREW_PREFIX}/opt/icu4c/lib ${HOMEBREW_PREFIX}/opt/llvm/lib ${HOMEBREW_PREFIX}/opt/med-file@4.1.1/lib ${HOMEBREW_PREFIX}/opt/gcc/lib/gcc/12 /usr/local/opt ${CONFIG_NGLIB} ${Qt5Core_DIR}/../../.. ${XCTEST_PATH} ${WEBKIT_FRAMEWORK_DIR}
     )"
 )
 ```

In `/Users/danielstaple/Library/Caches/Homebrew/freecad@0.20.1--git/src/MacAppBundle/CMakeLists.txt` I see:
```
add_subdirectory(Build)
add_subdirectory(3rdParty)
add_subdirectory(Base)
add_subdirectory(App)
add_subdirectory(Main)
add_subdirectory(Mod)
add_subdirectory(Ext)
add_subdirectory(Doc)

if(BUILD_GUI)
    add_subdirectory(Gui)
    if(UNIX AND NOT APPLE)
        add_subdirectory(XDGData)
    endif()
endif(BUILD_GUI)

if(BUILD_TEMPLATE)
    add_subdirectory(Tools/_TEMPLATE_)
endif(BUILD_TEMPLATE)

if(FREECAD_CREATE_MAC_APP)
    add_subdirectory(MacAppBundle)
endif(FREECAD_CREATE_MAC_APP)
```

There's not enough for a line 132 there.

dannystaple, 2022-11-04:
There's significantly more at https://github.com/FreeCAD/FreeCAD/blob/master/src/MacAppBundle/CMakeLists.txt. Hmm - I'm going to see if I can get brew to clear those caches - as that looks very wrong to me.

dannystaple, 2022-11-04:
`brew cleanup` did a lot of stuff, but `/Users/danielstaple/Library/Caches/Homebrew/freecad@0.20.1--git/src` is still there.. Hmmm.

dannystaple, 2022-11-04:
Still there after `brew cleanup --prune=1 -s`. So untapping, and then rm -rf that freecad@0.20.1--git, and other freecad patch files. Lets retry adding the tap and buidling after that.

dannystaple, 2022-11-04:
Also doing a brew doctor before trying, see fi there's any underlying issues.

dannystaple, 2022-11-04:
Between the rm -rf's and cleanup, missed soemthign as I still see:
```Already downloaded: /Users/danielstaple/Library/Caches/Homebrew/downloads/4753540cc1748fc29cd0744b6a540f8f5cee0c0f8db1f97168e0071849005e4e--med-file@4.1.1-4.1.1-1.bottle_manifest.json
``` for all these bits.

And still failed an install:
```
Hunk #1 FAILED at 114.
Hunk #2 FAILED at 132.
2 out of 2 hunks FAILED -- saving rejects to file src/MacAppBundle/CMakeLists.txt.rej
```

ipatch, 2022-11-04:
thanks for reporting the issue. there was a rather large commit that went through upstream editing a file related to mac app bundling, so if i had to guess it could be related to that. i'll poke around here in a bit and see what i can come up with.

ipatch, 2022-11-04:
more than likely related to

https://github.com/FreeCAD/FreeCAD/pull/7491

ipatch, 2022-11-04:
#368

dannystaple, 2022-11-04:
Ok - trying this out again. First - naively tried `brew cleanup --prune=1 -s && brew install --HEAD freecad` and still see the same error.
I then visited `$(brew --repository)/Library/Taps/freecad/homebrew-freecad` and performed a git pull - it was already on the most recent master. I performed a `git log` there and I can see:

```
Author: Chris <chris.r.jones.1983@gmail.com>
Date:   Fri Nov 4 13:02:23 2022 -0500

    [no ci] checkin patch file to fix `brew install freecad --HEAD` (#368)
```

Am I still missing a cleanup or update step?

ipatch, 2022-11-04:
it's kindof a 2 step process. i have to checkin the patch file, then update formula file. see the change i made with,

pr #369

ipatch, 2022-11-04:
i'm about to test locally to make sure the patching / configuring (cmake) goes through

ipatch, 2022-11-04:
tested locally with both commands posted above. from the tip of the `HEAD` and from `0.20.1` tarball, the `brew install` process completed the cmake checks and began the build/compile process.

i did have to unlink python@3.11 which was tripping the boost cmake check.

hope this helps.

dannystaple, 2022-11-04:
Hello, this built and installed with a `brew update` and `brew install freecad/freecad/freecad --head`. I'm happy to call it closed.

closed 2022-11-04 by dannystaple

ipatch, 2022-11-05:
> Hello, this built and installed with a `brew update` and `brew install freecad/freecad/freecad --head`. I'm happy to call it closed.

awesome 🎉🥳 glad to hear that, and thanks for the followup for letting me know, and closing the issue. 👍