FreeCAD/homebrew-freecad #373: Documentation on building dev version

dannystaple, 2022-11-05:
I am trying to use https://wiki.freecadweb.org/Compile_on_MacOS and this to build. 
It looks like that nglib and vtk must be installed with Brew, and the PREFIX_PATH isn't quite right.

Do you have any docs on getting this to build for FreeCAD development?

dannystaple, 2022-11-05:
With the version installed lastnight, during startup (with /usr/local/Cellar/freecad@0.20.1/HEAD-caa953a/MacOS/FreeCAD)  
I see:
```shell
During initialization the error "No module named 'PySide2'" occurred in /usr/local/Cellar/freecad@0.20.1/HEAD-caa953a/Mod/Draft/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide2'" occurred in /usr/local/Cellar/freecad@0.20.1/HEAD-caa953a/Mod/Arch/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide2'" occurred in /usr/local/Cellar/freecad@0.20.1/HEAD-caa953a/Mod/AddonManager/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide2'" occurred in /usr/local/Cellar/freecad@0.20.1/HEAD-caa953a/Mod/Tux/InitGui.py
Please look into the log file for further information
``` 

but FreeCAD does start.

Trying to compile I see:

```
-- libshiboken built for Release
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'shiboken2'
FATAL_ERROR,==================================
Shiboken2 Python module not found.
==================================

Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'PySide2'
FATAL_ERROR,================================
```

brew info for these shows me:

```
>  brew info pyside2@5.15.5 
==> freecad/freecad/pyside2@5.15.5: stable 5.15.5 (bottled) [keg-only]
Python bindings for Qt5 and greater
https://code.qt.io/cgit/pyside/pyside-setup.git/tree/README.pyside2.md?h=5.15.2
/usr/local/Cellar/pyside2@5.15.5/5.15.5 (235 files, 42.0MB)
  Poured from bottle on 2022-11-04 at 14:02:46
From: https://github.com/freecad/homebrew-freecad/blob/HEAD/Formula/pyside2@5.15.5.rb
License: GFDL-1.3-only and GPL-2.0-only and GPL-3.0-only and LGPL-2.1-only and LGPL-3.0-only
==> Dependencies
Build: cmake ✔, python@3.10 ✔
Required: freecad/freecad/shiboken2@5.15.5 ✔, qt@5 ✔, sphinx-doc ✔
==> Requirements
Build: Xcode ✔
==> Caveats
    if qt6, qt@6 is linked then this formula will fail to build from source
    this formula requires manually linking after install

pyside2@5.15.5 is keg-only, which means it was not symlinked into /usr/local,
because this is an alternate version of another formula.

For compilers to find pyside2@5.15.5 you may need to set:
  export LDFLAGS="-L/usr/local/opt/pyside2@5.15.5/lib"
  export CPPFLAGS="-I/usr/local/opt/pyside2@5.15.5/include"

For pkg-config to find pyside2@5.15.5 you may need to set:
  export PKG_CONFIG_PATH="/usr/local/opt/pyside2@5.15.5/lib/pkgconfig"
```

and 

```
> brew info shiboken2@5.15.5 
==> freecad/freecad/shiboken2@5.15.5: stable 5.15.5 (bottled), HEAD [keg-only]
GeneratorRunner plugin that outputs C++ code for CPython extensions
https://code.qt.io/cgit/pyside/pyside-setup.git/tree/README.shiboken2-generator.md?h=5.15.2
/usr/local/Cellar/shiboken2@5.15.5/5.15.5_1 (65 files, 2.6MB)
  Poured from bottle on 2022-11-04 at 14:02:44
From: https://github.com/freecad/homebrew-freecad/blob/HEAD/Formula/shiboken2@5.15.5.rb
License: GFDL-1.3-only and GPL-2.0-only and GPL-3.0-only and LGPL-2.1-only and LGPL-3.0-only
==> Dependencies
Build: cmake ✔, python@3.10 ✔
Required: llvm ✔, numpy ✔, qt@5 ✔
==> Options
--HEAD
        Install HEAD version
==> Caveats
    this formula is keg-only due to freecad/freecad/shiboken2

shiboken2@5.15.5 is keg-only, which means it was not symlinked into /usr/local,
because this is an alternate version of another formula.

If you need to have shiboken2@5.15.5 first in your PATH, run:
  echo 'export PATH="/usr/local/opt/shiboken2@5.15.5/bin:$PATH"' >> /Users/danielstaple/.bash_profile

For compilers to find shiboken2@5.15.5 you may need to set:
  export LDFLAGS="-L/usr/local/opt/shiboken2@5.15.5/lib"
  export CPPFLAGS="-I/usr/local/opt/shiboken2@5.15.5/include"

For pkg-config to find shiboken2@5.15.5 you may need to set:
  export PKG_CONFIG_PATH="/usr/local/opt/shiboken2@5.15.5/lib/pkgconfig"
```
Let's add the exports in and try that build again.. See if that removes the shiboken2 errors.

```shell
-- Shiboken2Config: Using default python: .cpython-310-darwin
-- Found PythonInterp: /usr/local/Frameworks/Python.framework/Versions/3.10/bin/python3.10 (found suitable version "3.10.8", minimum required is "3") 
-- Found PythonLibs: /usr/local/opt/python@3.10/Frameworks/Python.framework/Versions/3.10/lib/libpython3.10.dylib (Required is at least version "3") 
-- SHIBOKEN_PYTHON_INCLUDE_DIRS computed to value: '/usr/local/opt/python@3.10/Frameworks/Python.framework/Versions/3.10/include/python3.10
'
-- SHIBOKEN_PYTHON_LIBRARIES computed to value: '-undefined dynamic_lookup'
-- libshiboken built for Release
-- PYTHON_CONFIG_SUFFIX: .cpython-310-darwin
-- libshiboken built for Release
```

In my attempts to build I added the following:
```
export PATH="/usr/local/opt/shiboken2@5.15.5/bin:$PATH"
export PKG_CONFIG_PATH="/usr/local/opt/shiboken2@5.15.5/lib/pkgconfig:/usr/local/opt/pyside2@5.15.5/lib/pkgconfig:$PKG_CONFIG_PATH"
export LDFLAGS="-L/usr/local/opt/shiboken2@5.15.5/lib -L/usr/local/opt/pyside2@5.15.5/lib $LDFLAGS"
export CPPFLAGS="-I/usr/local/opt/shiboken2@5.15.5/include -I/usr/local/opt/pyside2@5.15.5/include $CPPFLAGS"
```

And into the cmake command:
```
  -DPySide2_DIR="/usr/local/opt/pyside2@5.15.5/lib/cmake/PySide2-5.15.5" \
  -DShiboken2_DIR="/usr/local/opt/shiboken2@5.15.5/lib/cmake/Shiboken2-5.15.5" \
```

Going to keep plugging at this - but not got it figured out yet.

ipatch, 2022-11-05:
i see as you've found out that doing the `brew install freecad/freecad/freecad` does install a version of freecad, but doesn't lend itself to making local changes to the freecad source unfortunately the way brew is setup. brew is more geared for installing packages as opposed to developing packages. all that said, i do use deps installed by brew to manually build freecad, and then manually invoke cmake to configure freecad source followed by a make then make install.

and as you've seen, freecad has a runtime deps of shiboken and pyside. and unfortunately the way brew is setup to handle versioned formula doesn't lend itself well for working with versioned deps. the quickest way to get cmake to find the shiboken2 and pyside2 deps is to run the below commands

```
brew link freecad/freecad/shiboken2@5.15.5 -v
brew link freecad/freecad/pyside2@5.15.5 -v
```

then use something along the lines of the below cmake invocation to setup your freecad source.

```
cmake \
-DHOMEBREW_PREFIX=$bp \
-DPYTHON_EXECUTABLE="$bp/opt/python@3.10/bin/python3" \
-DPYTHON_INCLUDE_DIR="$bp/opt/python@3.10/Frameworks/Python.framework/Headers" \
-DCMAKE_PREFIX_PATH="$pthmedfile;$pthswig;$pthqt;$pthshiboken2;$pthpyside2;$pthvtk82;$pthocc;$pthicu4c" \
-DCMAKE_BUILD_TYPE=RelWithDebInfo \
-DBUILD_FEM_NETGEN=0 \
-DFREECAD_USE_3DCONNEXION=1 \
-DBUILD_QT5=1 \
-DBUILD_ENABLE_CXX_STD=C++17 \
-Wno-dev \
-DFREECAD_CREATE_MAC_APP=1 \
-DCMAKE_INSTALL_PREFIX="$code/freecad-git/installs/$(basename $PWD)" \
-DBUILD_ARCH=1 \
-DBUILD_DRAFT=1 \
-DBUILD_TUX=1 \
-DBUILD_OPENSCAD=1 \
-DBUILD_TECHDRAW=1 \
-DBUILD_SPREADSHEET=1 \
-DBUILD_DRAWING=1 \
\
../../freecad-src/
```

obviously that cmake invocation will have to be adapted for your dev environment, and the local env vars defined in that invocation are specific for my build env as well. i tend to lean on manually specifying the prefix paths for the required deps, but if the dep is linked with a `brew link` the freecad cmake scripts tend to do a decent job in finding the library however there are some exceptions.

take a look at the freecad formula file itself, i've left quite bit of little notes in there.

chennes, 2022-11-05:
You should drop the BUILD_QT5 flag: it doesn't exist anymore. You can either omit it completely, or use FREECAD_QT_VERSION=5 instead (though that is the default, so it is not needed).

dannystaple, 2022-11-06:
Good news - I got the configure step completed.
I set:
```
export PYTHONPATH="$PYTHONPATH:/usr/local/opt/pyside2@5.15.5/lib/python3.10/site-packages/:/usr/local/opt/shiboken2@5.15.5/lib/python3.10/site-packages/"
```
And used:
```
```

From a git cone of master (of branch thereof), will I need to apply the patches from this repo? Is there a recommended/suggested way you do this mid build (I don't want to create diffs that I might accidentally check in)?

ipatch, 2024-04-29:
@dannystaple 

hey you should be able to just run `brew install freecad@0.20.2_py310` and everything should _just work_™️ i was able to successfully build homebrew bottle artifacts for the last three macos versions provided by the github ci.

i'll close this issue in a day or two if i don't here back or if i hear that things worked out.

cheers
chris

ipatch, 2024-05-01:
please leave a comment if you are still having an issue, and will reopen. thanks.

closed 2024-05-01 by ipatch