FreeCAD/homebrew-freecad #374: installing 0.20.1 on Apple M1

christiangda, 2022-12-17:
Hi 
I recently updated my MacBook Pro to the latest OS release 13.1 (macOS Ventura). After that, I tried to install FreeCad's latest version available on the brew, following the instruction available on this link: [https://github.com/FreeCAD/homebrew-freecad](https://github.com/FreeCAD/homebrew-freecad)

The commands I used are:

```bash
brew tap freecad/freecad
brew install freecad -v
```

 and also I tried

```bash
brew install --HEAD freecad
```

both of them generate the following error:

```bash
...
[ 95%] Building Fortran object src/CMakeFiles/medfwrap.dir/MEDiteratorsF.f.o
cd /tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/build/tools/mdump && /opt/homebrew/Library/Homebrew/shims/mac/super/clang -DH5_USE_16_API -DMED_MESGERR_FOR_TOOLS -I/opt/homebrew/Cellar/hdf5/1.12.2_2/include -I/opt/homebrew/opt/libaec/include -I/tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/build/include -I/tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/include -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk -fPIE -MD -MT tools/mdump/CMakeFiles/mdump2.dir/mdump2.c.o -MF CMakeFiles/mdump2.dir/mdump2.c.o.d -o CMakeFiles/mdump2.dir/mdump2.c.o -c /tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/tools/mdump/mdump2.c
cd /tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/build/src && /opt/homebrew/bin/gfortran -DH5_USE_16_API -Dmedfwrap_EXPORTS -I/opt/homebrew/Cellar/hdf5/1.12.2_2/include -I/opt/homebrew/opt/libaec/include -I/tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/build/include -I/tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/include -O3 -DNDEBUG -O3 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk -fPIC -c /tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/src/MEDiteratorsF.f -o CMakeFiles/medfwrap.dir/MEDiteratorsF.f.o
/tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/build/python/med/medfamily_modulePYTHON_wrap.cxx:178:11: fatal error: 'Python.h' file not found
# include <Python.h>
          ^~~~~~~~~~
/tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/build/python/med/medenum_modulePYTHON_wrap.cxx:178:11: fatal error: 'Python.h' file not found
# include <Python.h>
          ^~~~~~~~~~
/tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/build/python/med/medenumtest_modulePYTHON_wrap.cxx:178:11: fatal error: 'Python.h' file not found
# include <Python.h>
          ^~~~~~~~~~
/tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/build/python/med/medequivalence_modulePYTHON_wrap.cxx:178:11: fatal error: 'Python.h' file not found
# include <Python.h>
          ^~~~~~~~~~
[ 95%] Linking CXX shared library libmedfwrap.dylib
cd /tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/build/src && /opt/homebrew/Cellar/cmake/3.25.1/bin/cmake -E cmake_link_script CMakeFiles/medfwrap.dir/link.txt --verbose=1
/opt/homebrew/Library/Homebrew/shims/mac/super/clang++ -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk -dynamiclib -Wl,-headerpad_max_install_names -compatibility_version 11.0.0 -current_version 11.1.1 -o libmedfwrap.11.1.1.dylib -install_name @rpath/libmedfwrap.11.dylib CMakeFiles/medfwrap.dir/MEDiteratorsF.f.o 2.3.6/fi/CMakeFiles/_fi236.dir/entmaaf.f.o 2.3.6/fi/CMakeFiles/_fi236.dir/jntf.f.o 2.3.6/fi/CMakeFiles/_fi236.dir/gauss.f.o 2.3.6/fi/CMakeFiles/_fi236.dir/exit.f.o 2.3.6/fi/CMakeFiles/_fi236.dir/resf_i.f.o 2.3.6/fi/CMakeFiles/_fi236.dir/interf.f.o 2.3.6/fi/CMakeFiles/_fi236.dir/fichf.f.o 2.3.6/fi/CMakeFiles/_fi236.dir/maaf.f.o 2.3.6/fi/CMakeFiles/_fi236.dir/prof.f.o 2.3.6/fi/CMakeFiles/_fi236.dir/resf_r.f.o 2.3.6/fi/CMakeFiles/_fi236.dir/famf.f.o 2.3.6/fi/CMakeFiles/_fi236.dir/equf.f.o 2.3.6/fi/CMakeFiles/_fi236.dir/scalaire.f.o fi/CMakeFiles/_fi.dir/medsupport.f.o fi/CMakeFiles/_fi.dir/medequivalence.f.o fi/CMakeFiles/_fi.dir/medlink.f.o fi/CMakeFiles/_fi.dir/medinterp.f.o fi/CMakeFiles/_fi.dir/medstructelement.f.o fi/CMakeFiles/_fi.dir/medfilter.f.o fi/CMakeFiles/_fi.dir/medfile.f.o fi/CMakeFiles/_fi.dir/medlocalization.f.o fi/CMakeFiles/_fi.dir/medprofile.f.o fi/CMakeFiles/_fi.dir/medjoint.f.o fi/CMakeFiles/_fi.dir/medparameter.f.o fi/CMakeFiles/_fi.dir/medfamily.f.o fi/CMakeFiles/_fi.dir/medfield.f.o fi/CMakeFiles/_fi.dir/medflibrary.f.o fi/CMakeFiles/_fi.dir/medmesh.f.o 2.3.6/cfi/CMakeFiles/_cfi236.dir/equcf.c.o 2.3.6/cfi/CMakeFiles/_cfi236.dir/fichcf.c.o 2.3.6/cfi/CMakeFiles/_cfi236.dir/intercf.c.o 2.3.6/cfi/CMakeFiles/_cfi236.dir/jntcf.c.o 2.3.6/cfi/CMakeFiles/_cfi236.dir/rescf.c.o 2.3.6/cfi/CMakeFiles/_cfi236.dir/entmaacf.c.o 2.3.6/cfi/CMakeFiles/_cfi236.dir/famcf.c.o 2.3.6/cfi/CMakeFiles/_cfi236.dir/maacf.c.o 2.3.6/cfi/CMakeFiles/_cfi236.dir/procf.c.o 2.3.6/cfi/CMakeFiles/_cfi236.dir/scalairecf.c.o 2.3.6/cfi/CMakeFiles/_cfi236.dir/gausscf.c.o 2.3.6/cfi/CMakeFiles/_cfi236.dir/exitcf.c.o 2.3.6/cfi/CMakeFiles/_cfi236.dir/famcf231.c.o 2.3.6/cfi/CMakeFiles/_cfi236.dir/famcf232.c.o 2.3.6/cfi/CMakeFiles/_cfi236.dir/MEDversionedApiF.cxx.o cfi/CMakeFiles/_cfi.dir/librarycf.c.o cfi/CMakeFiles/_cfi.dir/medlocalizationcf.c.o cfi/CMakeFiles/_cfi.dir/medprofilecf.c.o cfi/CMakeFiles/_cfi.dir/medinterpcf.c.o cfi/CMakeFiles/_cfi.dir/medlinkcf.c.o cfi/CMakeFiles/_cfi.dir/medequivalencecf.c.o cfi/CMakeFiles/_cfi.dir/medfieldcf.c.o cfi/CMakeFiles/_cfi.dir/meshcf.c.o cfi/CMakeFiles/_cfi.dir/medfamilycf.c.o cfi/CMakeFiles/_cfi.dir/medstructelementcf.c.o cfi/CMakeFiles/_cfi.dir/medsupportcf.c.o cfi/CMakeFiles/_cfi.dir/medjointcf.c.o cfi/CMakeFiles/_cfi.dir/medparametercf.c.o cfi/CMakeFiles/_cfi.dir/medfiltercf.c.o cfi/CMakeFiles/_cfi.dir/filecf.c.o cfi/CMakeFiles/_cfi.dir/MEDversionedApi3F.cxx.o   -L/opt/homebrew/Cellar/gcc/12.2.0/lib/gcc/current/gcc/aarch64-apple-darwin21/12  -L/opt/homebrew/Cellar/gcc/12.2.0/lib/gcc/current/gcc  -L/opt/homebrew/Cellar/gcc/12.2.0/lib/gcc/current  -L/usr/local/lib  -Wl,-rpath,/tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/build/src libmedC.11.1.1.dylib /opt/homebrew/opt/hdf5/lib/libhdf5.dylib /opt/homebrew/lib/libsz.dylib /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk/usr/lib/libz.tbd /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk/usr/lib/libdl.tbd /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk/usr/lib/libm.tbd -lgfortran -lemutls_w -lgcc -lquadmath -lemutls_w -lgcc -lgcc
[ 95%] Linking C executable medconforme
cd /tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/build/tools/medconforme && /opt/homebrew/Cellar/cmake/3.25.1/bin/cmake -E cmake_link_script CMakeFiles/medconforme.dir/link.txt --verbose=1
1 error generated.
[ 95%] Building C object tools/medimport/CMakeFiles/medimportengine.dir/MAJ_version.c.o
cd /tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/build/tools/medimport && /opt/homebrew/Library/Homebrew/shims/mac/super/clang -DH5_USE_16_API -DMED_MESGERR_FOR_TOOLS -Dmedimportengine_EXPORTS -I/opt/homebrew/Cellar/hdf5/1.12.2_2/include -I/opt/homebrew/opt/libaec/include -I/tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/tools/medimport/include -I/tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/build/include -I/tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/include -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk -fPIC -MD -MT tools/medimport/CMakeFiles/medimportengine.dir/MAJ_version.c.o -MF CMakeFiles/medimportengine.dir/MAJ_version.c.o.d -o CMakeFiles/medimportengine.dir/MAJ_version.c.o -c /tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/tools/medimport/MAJ_version.c
/opt/homebrew/Library/Homebrew/shims/mac/super/clang -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names CMakeFiles/medconforme.dir/medconforme.c.o -o medconforme  -Wl,-rpath,/tmp/med-fileA4.1.1-20221217-96942-1lccosc/med-4.1.1_SRC/build/src ../../src/libmedC.11.1.1.dylib /opt/homebrew/opt/hdf5/lib/libhdf5.dylib /opt/homebrew/lib/libsz.dylib /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk/usr/lib/libz.tbd /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk/usr/lib/libdl.tbd /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk/usr/lib/libm.tbd
make[2]: *** [python/CMakeFiles/_medenumtest.dir/med/medenumtest_modulePYTHON_wrap.cxx.o] Error 1
make[1]: *** [python/CMakeFiles/_medenumtest.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 95%] Building C object tools/mdump/CMakeFiles/mdump3.dir/MED23meshIterators.c.o
[... cut ...]
double-conversion		highway				libnghttp2			metis				qmk				zeromq
doxygen				htop				libogg				minikube			qt				zimg
eigen				hunspell			libomp				mpdecimal			qt@5				zlib
ffmpeg				hwloc				libpng				mpfr				rapidjson			zstd
ffmpeg@4			icu4c				libpthread-stubs		mpg123				rav1e
flac				imagemagick			libraqm				ncurses				readline
fontconfig			imath				libraw				netcdf				rubberband
freeimage			isl				librist				netpbm				rust
``` 

There is a way to tell brew package `med-file@4.1.1` where to find the python3.10 headers?

ipatch, 2022-12-17:
thanks for the thorough error/bug report. i'll try and take a deeper look into this probably on monday. i'll have to setup a ventura test unbox unfortunately because the newest box/vm i have setup is big sur.

christiangda, 2022-12-18:
hi @ipatch , thank you very much,  please let me know if you need something from my side to be tested once you are working on this.  Sorry because I can't contribute much more on this, I don't know too much about how brew inject necessary flags on build time.

ipatch, 2022-12-19:
i believe i was able to reproduce this error on the m1 box i have access to. i used the below command to reproduce the error you mentioned above on the m1 box,

```
brew install freecad/freecad/med-file -v -s 
```

the way the directory / tap structure is setup is that the `med-file` alias should point to the `med-file@4.1.1` formula file.

food for thought, i fired a x86 big sur vm this evening and am unable to reproduce this error 😕.

i'll continue to further work through this issue. and hopefully submit a PR that applies a fix.

ipatch, 2022-12-20:
i just tested pr #375 on the m1 box running macos ventura i have access, and running the install command i mentioned above seemed to install med-file as should on the box.

if you feel comfortable with git skills you can pull in the PR locally and use it now, or you can wait for me to run it through the github runners tomorrow to make sure nothing crazy happens, and allow the runners to rebuild the bottles for the older versions of macos.

hope this helps.

ipatch, 2022-12-21:
this is where i'm at right now.

https://github.com/gromgit/homebrew-core-mojave/issues/9

another way to see that issue is from the below job, hopefully this will get resolved sooner than later.

https://github.com/FreeCAD/homebrew-freecad/actions/runs/3737189355/jobs/6357290707#step:17:571

christiangda, 2023-01-22:
> i just tested pr #375 on the m1 box running macos ventura i have access to, and running the install command i mentioned above seemed to install med-file as it should on the box.
> 
> if you feel comfortable with your git skills you can pull in the PR locally and use it now, or you can wait for me to run it through the github runners tomorrow to make sure nothing crazy happens, and allow the runners to rebuild the bottles for the older versions of macos.
> 
> hope this helps.

Hi @ipatch sorry for my very late answer!

I tried the following with your remote branch

```bash
git clone git@github.com:ipatch/homebrew-freecad.git
cd homebrew-freecad
git switch issue.374
```

then

```bach
brew install --build-from-source ./Formula/freecad@0.20.1.rb
```

and I got the following errors

```bash
Error: Failed to load cask: ./Formula/freecad@0.20.1.rb
Cask 'freecad@0.20.1' is unreadable: wrong constant name #<Class:0x000000013fc3e700>
Warning: Treating ./Formula/freecad@0.20.1.rb as a formula.
==> Fetching dependencies for freecad@0.20.1: freecad/freecad/med-file, freecad/freecad/shiboken2@5.15.5, freecad/freecad/pyside2@5.15.5, opencascade, orocos-kdl and xerces-c
==> Fetching freecad/freecad/med-file@4.1.1
==> Downloading https://raw.githubusercontent.com/archlinux/svntogit-community/458b52e0d43ebbcf67f9025aad66c76454573a06/trunk/hdf5-1.12.patch
Already downloaded: /Users/christian/Library/Caches/Homebrew/downloads/1bedb86bef42de206c897730e953c014c727d5c52c55f4354fdb56d681d31ca5--hdf5-1.12.patch
==> Downloading https://files.salome-platform.org/Salome/other/med-4.1.1.tar.gz
Already downloaded: /Users/christian/Library/Caches/Homebrew/downloads/7dcf99218d0dc407132afdc6ebe9e83793e2808f13744a3cb004cb0bbc8335c8--med-4.1.1.tar.gz
==> Fetching freecad/freecad/shiboken2@5.15.5
==> Downloading https://raw.githubusercontent.com/FreeCAD/homebrew-freecad/8944b8b362c7fd87c515efb07eb0fb022e946610/patches/libshiboken-numpy-1.23.compat.patch
Already downloaded: /Users/christian/Library/Caches/Homebrew/downloads/ed6d1a59a76d10fd011fc6a95bd26ef96d7d91262ac5666e48592f259e6259ab--libshiboken-numpy-1.23.compat.patch
==> Downloading https://raw.githubusercontent.com/FreeCAD/homebrew-freecad/8944b8b362c7fd87c515efb07eb0fb022e946610/patches/libshiboken2-python10-compat.patch
Already downloaded: /Users/christian/Library/Caches/Homebrew/downloads/35da8d9458288cd8048d657eebe1815741c26d6750ab2f8a9edf499244351925--libshiboken2-python10-compat.patch
==> Downloading https://download.qt.io/official_releases/QtForPython/pyside2/PySide2-5.15.5-src/pyside-setup-opensource-src-5.15.5.zip
Already downloaded: /Users/christian/Library/Caches/Homebrew/downloads/7e478bd63671464e429850a09bc66eeda44c95ddd8ab0b66cfc8224ffd477cc9--pyside-setup-opensource-src-5.15.5.zip
==> Fetching freecad/freecad/pyside2@5.15.5
==> Downloading https://download.qt.io/official_releases/QtForPython/pyside2/PySide2-5.15.5-src/pyside-setup-opensource-src-5.15.5.zip
Already downloaded: /Users/christian/Library/Caches/Homebrew/downloads/7e478bd63671464e429850a09bc66eeda44c95ddd8ab0b66cfc8224ffd477cc9--pyside-setup-opensource-src-5.15.5.zip
==> Fetching opencascade
==> Downloading https://ghcr.io/v2/homebrew/core/opencascade/manifests/7.6.3_1
Already downloaded: /Users/christian/Library/Caches/Homebrew/downloads/ec6ceed2e910d4d3a20740dcc196c74de0e20b265ef77489fc4fac2328106e23--opencascade-7.6.3_1.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/opencascade/blobs/sha256:ee40f04033b6e18e85244585165ae9fe4f607819f90832b987c33ec7dcc0bcf7
Already downloaded: /Users/christian/Library/Caches/Homebrew/downloads/775dc7c5197966fd092a1d0a15fed2b14abc94fe81a7fc47217342b517fc672c--opencascade--7.6.3_1.arm64_ventura.bottle.tar.gz
==> Fetching orocos-kdl
==> Downloading https://ghcr.io/v2/homebrew/core/orocos-kdl/manifests/1.5.1
Already downloaded: /Users/christian/Library/Caches/Homebrew/downloads/c3dace39ce0285564caf38a41219e45e00d574ba10a3bf22adcb42857cda7c56--orocos-kdl-1.5.1.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/orocos-kdl/blobs/sha256:49449b3bff3c05cad2d5ba7d7271d86a9b76a32ac0272e88a7544abc080c1e72
Already downloaded: /Users/christian/Library/Caches/Homebrew/downloads/f16672e5a47c035c2f3b079692c7d1f8dc41761735c1c536ef0f774fe75a2e06--orocos-kdl--1.5.1.arm64_ventura.bottle.tar.gz
==> Fetching xerces-c
==> Downloading https://ghcr.io/v2/homebrew/core/xerces-c/manifests/3.2.4_1-1
Already downloaded: /Users/christian/Library/Caches/Homebrew/downloads/3ace7ff6ac268af682c2bee1179fea7f5d2156ba96e474eb20f159ec45d761f8--xerces-c-3.2.4_1-1.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/xerces-c/blobs/sha256:99006e9ad984212dc5016d5aa9f6ae8021d50f56fec9e13947d9779d9decc1de
Already downloaded: /Users/christian/Library/Caches/Homebrew/downloads/5498169a0125d861716f05a473e601eaf88326f86cf060edcdb5ef54a089a826--xerces-c--3.2.4_1.arm64_ventura.bottle.1.tar.gz
==> Fetching freecad@0.20.1
==> Downloading https://raw.githubusercontent.com/FreeCAD/homebrew-freecad/a4b71def99b5fe907550729038752aaf6fa1b9bf/patches/freecad-0.20.1-macos-app-bundling.patch
Already downloaded: /Users/christian/Library/Caches/Homebrew/downloads/4fa80a175f519cda0d7e502602499c920e22a6290a361b67b4f3f582bccb8dc5--freecad-0.20.1-macos-app-bundling.patch
==> Downloading https://raw.githubusercontent.com/FreeCAD/homebrew-freecad/a4b71def99b5fe907550729038752aaf6fa1b9bf/patches/freecad-0.20.1-macos-10.15-sdk.patch
[... cut ...]
make[1]: *** [python/CMakeFiles/_medfield.dir/all] Error 2
1 error generated.
make[2]: *** [python/CMakeFiles/_medfile.dir/med/medfile_modulePYTHON_wrap.cxx.o] Error 1
make[1]: *** [python/CMakeFiles/_medfile.dir/all] Error 2
make: *** [all] Error 2

Do not report this issue to Homebrew/brew or Homebrew/core!

These open issues may also help:
installing 0.20.1 on Apple M1 (errors installing med-file@4.1.1) https://github.com/FreeCAD/homebrew-freecad/issues/374
```

Let me know if this is what you were expecting

ipatch, 2023-01-23:
check out this, i am not able to reproduce using my PR

https://asciinema.org/a/553545

christiangda, 2023-01-25:
Hi, @ipatch thank you for your help and detailed video.

I follows these instructions and looks like now it is working

```bash
python3 --version
Python 3.10.6
```

```bash 
brew update && brew upgrade
mkdir freecad
cd freecad/
git clone git@github.com:FreeCAD/homebrew-freecad.git
cd homebrew-freecad/
git fetch origin pull/375/head:issue.374
git switch issue.374
```

```bash
brew reinstall Formula/med-file@4.1.1.rb
Error: Failed to load cask: Formula/med-file@4.1.1.rb
Cask 'med-file@4.1.1' is unreadable: wrong constant name #<Class:0x0000000133960170>
Warning: Treating Formula/med-file@4.1.1.rb as a formula.
==> Fetching med-file@4.1.1
==> Downloading https://raw.githubusercontent.com/archlinux/svntogit-community/458b52e0d43ebbcf67f9025aad66c76454573a06/trunk/hdf5-1.12.patch
Already downloaded: /Users/christian/Library/Caches/Homebrew/downloads/1bedb86bef42de206c897730e953c014c727d5c52c55f4354fdb56d681d31ca5--hdf5-1.12.patch
==> Downloading https://files.salome-platform.org/Salome/other/med-4.1.1.tar.gz
Already downloaded: /Users/christian/Library/Caches/Homebrew/downloads/7dcf99218d0dc407132afdc6ebe9e83793e2808f13744a3cb004cb0bbc8335c8--med-4.1.1.tar.gz
==> Reinstalling med-file@4.1.1
==> Patching
==> Applying hdf5-1.12.patch
patching file 'config/cmake_files/medMacros.cmake'
patching file 'src/ci/MEDfileCompatibility.c'
patching file 'src/hdfi/_MEDattributeNumWrByName.c'
patching file 'src/hdfi/_MEDattributeNumWr.c'
patching file 'src/hdfi/_MEDcheckAttributeStringFunc.c'
patching file 'src/hdfi/_MEDchecknSublinkFunc.c'
patching file 'src/hdfi/_MEDdatagroupExist.c'
patching file 'src/hdfi/_MEDdatasetExist.c'
patching file 'src/hdfi/_MEDfileCreate.c'
patching file 'src/hdfi/_MEDfileOpen.c'
patching file 'src/hdfi/_MEDlinkObjs.c'
patching file 'src/hdfi/_MEDmemFileOpen.c'
patching file 'src/hdfi/_MEDparFileCreate.c'
patching file 'src/hdfi/_MEDparFileOpen.c'
patching file 'tools/medimport/2.3.6/_MEDconvertStringDatasets.c'
==> cmake .. -DMEDFILE_BUILD_PYTHON=ON -DMEDFILE_BUILD_TESTS=OFF -DMEDFILE_INSTALL_DOC=OFF -DPYTHON_EXECUTABLE=/opt/homebrew/opt/python@3.10/bin/python3.10" -DPYTHON_LIBRARY=/opt/homebrew
==> make
==> make install
🍺  /opt/homebrew/Cellar/med-file@4.1.1/4.1.1: 105 files, 9.7MB, built in 31 seconds
==> Running `brew cleanup med-file@4.1.1`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
```

```bash
 brew info med-file

==> freecad/freecad/med-file@4.1.1: stable 4.1.1
Modeling and Data Exchange standardized format library
https://www.salome-platform.org/
/opt/homebrew/Cellar/med-file@4.1.1/4.1.1 (105 files, 9.7MB) *
  Built from source on 2023-01-25 at 21:10:46
From: https://github.com/Freecad/homebrew-freecad/blob/HEAD/Formula/med-file@4.1.1.rb
License: GPL-3.0-only
==> Dependencies
Build: cmake ✔, freecad/freecad/swig@4.0.2 ✔
Required: gcc ✔, hdf5 ✔, libaec ✔, python@3.10 ✔
```
[... 37 lines cut ...]
Already downloaded: /Users/christian/Library/Caches/Homebrew/downloads/57542c99e41561dcc0cec6e338e6c04c8ab05e5ef18a0fb97cb4dfe5da8c811c--FreeCAD-0.20.1.tar.gz
==> Reinstalling freecad@0.20.1
==> Installing dependencies for freecad@0.20.1: freecad/freecad/pyside2@5.15.5, opencascade, orocos-kdl and xerces-c
==> Installing freecad@0.20.1 dependency: freecad/freecad/pyside2@5.15.5
==> cmake -DPYTHON_EXECUTABLE=/opt/homebrew/opt/python@3.10/Frameworks/Python.framework/Versions/3.10/bin/python3.10 -DPYTHON_LIBRARY=/opt/homebrew/opt/python@3.10/Frameworks/Python.frame
==> make -j10
Last 15 lines from /Users/christian/Library/Logs/Homebrew/pyside2@5.15.5/02.make:
(printsupport) [1832ms] Detecting inconsistencies in typesystem (7705)...           [OK]
(widgets) [1840ms] Checking inconsistencies in function modifications...       [OK]
(widgets) [1842ms] Writing log files...                                        [OK]
(printsupport) [1854ms] Checking inconsistencies in function modifications...       [OK]
(printsupport) [1855ms] Writing log files...                                        [OK]
(widgets) [1865ms] Running Source generator...                                 [OK]
(printsupport) [1879ms] Running Source generator...                                 [OK]
(printsupport) [1898ms] Running Header generator...                                 [OK]
Done, (printsupport) 1900ms, 4 warnings (1128 known issues)
[  2%] Built target QtPrintSupport_generator
[OK]
(widgets) [2300ms] Running Header generator...                                 [OK]
Done, (widgets) 2342ms, 16 warnings (1126 known issues)
[  2%] Built target QtWidgets_generator
make: *** [all] Error 2

Do not report this issue to Homebrew/brew or Homebrew/core!
```

ipatch, 2023-01-25:
i don't understand when you ran the below commands

```
mkdir freecad
cd freecad/
git clone git@github.com:FreeCAD/homebrew-freecad.git
cd homebrew-freecad/
git fetch origin pull/375/head:issue.374
git switch issue.374
```

if you already ran `brew tap freecad/freecad/freecad` it should be located in your brew prefix repo dir, so no need to clone again and create a separate directory with a fresh clone.

should be able to `cd` into the directory that you tapped freecad when you ran `brew tap freecad/freecad`

take a look at the below command,

```
brew tap-info freecad/freecad 
```

and you can cd into the the directory listed, and then make your changes in there. just make sure to create a new branch. and if you run `brew doctor` it should let you know you're not on the default branch.

stefan-nu, 2023-01-30:
Just to confirm. I get the same error when I try to build freecad on Mac M1.

```
brew reinstall freecad/freecad/med-file@4.1.1

...
patching file 'src/hdfi/_MEDparFileOpen.c'
patching file 'tools/medimport/2.3.6/_MEDconvertStringDatasets.c'
==> cmake .. -DMEDFILE_BUILD_PYTHON=ON -DMEDFILE_BUILD_TESTS=OFF -DMEDFILE_INSTALL_DOC=OFF -DCMAKE_PREFIX_PATH=/opt/homebrew/opt/hdf5/lib;/opt/homebrew/opt/gcc/lib -DCMAKE_INSTALL_
==> make
Last 15 lines from /Users/stiopa/Library/Logs/Homebrew/med-file@4.1.1/02.make:
[ 96%] Building C object src/ci/CMakeFiles/_ci_static.dir/_MEDfieldnEntityType41.c.o
cd /tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/build/src/ci && /opt/homebrew/Library/Homebrew/shims/mac/super/clang -DH5_USE_16_API -DMED3_USESTATIC -I/opt/homebrew/Cellar/hdf5/1.12.2_2/include -I/opt/homebrew/opt/libaec/include -I/tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/build/include -I/tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/include -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk -fPIC -MD -MT src/ci/CMakeFiles/_ci_static.dir/_MEDfieldnEntityType41.c.o -MF CMakeFiles/_ci_static.dir/_MEDfieldnEntityType41.c.o.d -o CMakeFiles/_ci_static.dir/_MEDfieldnEntityType41.c.o -c /tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/src/ci/_MEDfieldnEntityType41.c
[ 96%] Building C object src/ci/CMakeFiles/_ci_static.dir/_MEDfieldnGeometryType41.c.o
cd /tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/build/src/ci && /opt/homebrew/Library/Homebrew/shims/mac/super/clang -DH5_USE_16_API -DMED3_USESTATIC -I/opt/homebrew/Cellar/hdf5/1.12.2_2/include -I/opt/homebrew/opt/libaec/include -I/tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/build/include -I/tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/include -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk -fPIC -MD -MT src/ci/CMakeFiles/_ci_static.dir/_MEDfieldnGeometryType41.c.o -MF CMakeFiles/_ci_static.dir/_MEDfieldnGeometryType41.c.o.d -o CMakeFiles/_ci_static.dir/_MEDfieldnGeometryType41.c.o -c /tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/src/ci/_MEDfieldnGeometryType41.c
1 error generated.
make[2]: *** [python/CMakeFiles/_medfield.dir/med/medfield_modulePYTHON_wrap.cxx.o] Error 1
make[1]: *** [python/CMakeFiles/_medfield.dir/all] Error 2
[ 96%] Building C object src/ci/CMakeFiles/_ci_static.dir/MEDfieldnGeometryType.c.o
cd /tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/build/src/ci && /opt/homebrew/Library/Homebrew/shims/mac/super/clang -DH5_USE_16_API -DMED3_USESTATIC -I/opt/homebrew/Cellar/hdf5/1.12.2_2/include -I/opt/homebrew/opt/libaec/include -I/tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/build/include -I/tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/include -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk -fPIC -MD -MT src/ci/CMakeFiles/_ci_static.dir/MEDfieldnGeometryType.c.o -MF CMakeFiles/_ci_static.dir/MEDfieldnGeometryType.c.o.d -o CMakeFiles/_ci_static.dir/MEDfieldnGeometryType.c.o -c /tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/src/ci/MEDfieldnGeometryType.c
[ 96%] Building C object src/ci/CMakeFiles/_ci_static.dir/_MEDfieldGeometryType41.c.o
cd /tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/build/src/ci && /opt/homebrew/Library/Homebrew/shims/mac/super/clang -DH5_USE_16_API -DMED3_USESTATIC -I/opt/homebrew/Cellar/hdf5/1.12.2_2/include -I/opt/homebrew/opt/libaec/include -I/tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/build/include -I/tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/include -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk -fPIC -MD -MT src/ci/CMakeFiles/_ci_static.dir/_MEDfieldGeometryType41.c.o -MF CMakeFiles/_ci_static.dir/_MEDfieldGeometryType41.c.o.d -o CMakeFiles/_ci_static.dir/_MEDfieldGeometryType41.c.o -c /tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/src/ci/_MEDfieldGeometryType41.c
[ 96%] Building C object src/ci/CMakeFiles/_ci_static.dir/MEDfieldGeometryType.c.o
cd /tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/build/src/ci && /opt/homebrew/Library/Homebrew/shims/mac/super/clang -DH5_USE_16_API -DMED3_USESTATIC -I/opt/homebrew/Cellar/hdf5/1.12.2_2/include -I/opt/homebrew/opt/libaec/include -I/tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/build/include -I/tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/include -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk -fPIC -MD -MT src/ci/CMakeFiles/_ci_static.dir/MEDfieldGeometryType.c.o -MF CMakeFiles/_ci_static.dir/MEDfieldGeometryType.c.o.d -o CMakeFiles/_ci_static.dir/MEDfieldGeometryType.c.o -c /tmp/med-fileA4.1.1-20230130-95548-liw45/med-4.1.1_SRC/src/ci/MEDfieldGeometryType.c
[ 96%] Built target _ci_static
make: *** [all] Error 2
```

mrly16, 2023-02-04:
Thanks @ipatch and @christiangda.

I have successfully solved the <med-file> problem and encountered the same <pyside2> problem with the following code snippet.

```
cd /opt/homebrew/Library/Taps/freecad/homebrew-freecad 
git fetch origin pull/375/head:issue.374 
git switch issue.374g
```

oursland, 2023-03-11:
I'm also encountering the build failure and the `issue.374` branch is not resolving the issue, at least for `med-file@4.1.1`.  What I can discern is that the installation for `python@3.10`, for whatever reason, cannot find `Python.h` despite it being on the include path.  When I use the brew debug flag and break to a shell, copying and pasting the failed command suddenly succeeds.  Is there an environment issue with `python@3.10`?

I've tried replacing all references to `python3.10` with `python3.11` in the `Formula` directory, which resolved the `Python.h`-related build failures.  I think this suggests something broken when building with `python3.10`.  However, Python 3.11 is when `Py_TYPE()` was converted from a macro to a static function, necessitating the change to `Py_SET_TYPE()` for assignment calls.

After making the function call changes, I'm now encountering a Python error about freeing an un-malloced pointer address when generating `QtCore.py`.

micagoytia, 2023-04-15:
What is the performance difference between freeCAD and Rosetta that you guys are gaining when doing the migrations!? (Do you loose performance?)

oursland, 2023-04-17:
@alber70g If you're eager to build FreeCAD on macOS for ARM, take a look at my guide: https://github.com/oursland/FreeCAD-Build-Notes

This uses `conda` to manage the build environment, as the rapidly changing upstream dependencies in Homebrew has made getting a development environment up and running quite a painful experience.

ipatch, 2023-06-30:
this issue should be resolved now. just tested with the master branch on the freecad repo using an m1 mac that i have access to.

if there is something i have missed, feel free to leave comment below and i am not opposed to reopening the issue.

closed 2023-06-30 by ipatch