FreeCAD/homebrew-freecad #411: home brew install failed with "Could not identify shiboken version"

peggyxia-alj, 2023-10-22:
encountered the following error when using brew install freecad on mac.

==> cmake -DCMAKE_PREFIX_PATH=/opt/homebrew/opt/qt@5/lib -DPYTHON_EXECUTABLE=/opt/homebrew/opt/python@3.10/Frameworks/Python.framework/Versions/3.10/bin/python3 -DCMAKE_INSTALL_RPATH=/opt/homebrew/Cellar/shiboken2@5.15.5/5.15.5_1/lib ../sources/shiboken2
Last 15 lines from /Users/yuxia/Library/Logs/Homebrew/shiboken2@5.15.5/01.cmake:
-- Check for working CXX compiler: /opt/homebrew/Library/Homebrew/shims/mac/super/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using Qt 5
-- Found PythonInterp: /opt/homebrew/opt/python@3.10/Frameworks/Python.framework/Versions/3.10/bin/python3  
-- Found PythonLibs: /opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/lib/libpython3.11.dylib (found version "3.11.6") 
-- PYTHON_ARCH:             
-- CLANG: /opt/homebrew/opt/llvm, /opt/homebrew/opt/llvm/lib/libclang.dylib detected by LLVM_INSTALL_DIR
-- Looking for C++ include clang-c/Index.h
-- Looking for C++ include clang-c/Index.h - found
CMake Error at CMakeLists.txt:51 (message):
  Could not identify shiboken version.  Error:

ipatch, 2023-10-22:
from the looks of things it looks like you're on a m1 mac. i haven't done a build of freecad using m1 in a while. i'm trying to get things updated at the moment. for the time being if you are just wanting to use freecad i'd recommend using the conda version to suite your needs, but if your attempting to build freecad please give me some time to fix the outdated dependencies.

ipatch, 2023-10-23:
i got that same error this afternoon. from what i could tell it may be due to your homebrew installation using `python@3.11` as the default python. as presently the upstream `qt@5` is still using `python@3.10` so we gotta keep things at `python@3.10` until `qt@5` is updated to use `python@3.11`.

ipatch, 2023-10-25:
@peggyxia-alj 

when you get chance try installing `shiboken2@5.15.5` i just tested with the m1 box i have access to and think the pr #419  i just pushed should resolve the issue you reported earlier.

closed 2023-10-29 by ipatch