FreeCAD/homebrew-freecad #433: brew install --only-dependencies freecad fails in step freecad/freecad/pyside2@5.15.5

superfloh247, 2023-11-24:
I followed this documentation

https://wiki.freecad.org/Compile_on_MacOS

this step fails brew install --only-dependencies freecad 

M1 Mac, MacOS 14.1.1, Xcode 15.0.1

[02.make.txt.gz](https://github.com/FreeCAD/homebrew-freecad/files/13461977/02.make.txt.gz)

ipatch, 2023-11-24:
hey thanks for reporting the issue. 👍️ yes, the `pyside2@5.15.5` formula in this tap is running into build issues / failures. it is something i have been trying to address haven't submitted a PR to fix the build errors. but hopefully i can have something drafted by the end of the week. 🤞

ipatch, 2023-11-24:
also i downloaded your gzip make / build log the output is below, makes it easier for the spiders 🕷️ and people searching for error messages.

<details>
<summary> freecad pyside2@5.15.5 build issue </summary>

```
2023-11-24 19:54:50 +0000

make
-j10

2801 Willing Ave

/opt/homebrew/Cellar/cmake/3.27.8/bin/cmake -S/tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/sources/pyside2 -B/tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/macbuild5.15.5 --check-build-system CMakeFiles/Makefile.cmake 0
/opt/homebrew/Cellar/cmake/3.27.8/bin/cmake -E cmake_progress_start /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/macbuild5.15.5/CMakeFiles /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/macbuild5.15.5//CMakeFiles/progress.marks
/Applications/Xcode.app/Contents/Developer/usr/bin/make  -f CMakeFiles/Makefile2 all
/Applications/Xcode.app/Contents/Developer/usr/bin/make  -f libpyside/CMakeFiles/pyside2.dir/build.make libpyside/CMakeFiles/pyside2.dir/depend
/Applications/Xcode.app/Contents/Developer/usr/bin/make  -f plugins/CMakeFiles/uiplugin.dir/build.make plugins/CMakeFiles/uiplugin.dir/depend
/Applications/Xcode.app/Contents/Developer/usr/bin/make  -f PySide2/QtCore/CMakeFiles/QtCore_generator.dir/build.make PySide2/QtCore/CMakeFiles/QtCore_generator.dir/depend
/Applications/Xcode.app/Contents/Developer/usr/bin/make  -f PySide2/QtGui/CMakeFiles/QtGui_generator.dir/build.make PySide2/QtGui/CMakeFiles/QtGui_generator.dir/depend
/Applications/Xcode.app/Contents/Developer/usr/bin/make  -f PySide2/QtWidgets/CMakeFiles/QtWidgets_generator.dir/build.make PySide2/QtWidgets/CMakeFiles/QtWidgets_generator.dir/depend
/Applications/Xcode.app/Contents/Developer/usr/bin/make  -f PySide2/QtPrintSupport/CMakeFiles/QtPrintSupport_generator.dir/build.make PySide2/QtPrintSupport/CMakeFiles/QtPrintSupport_generator.dir/depend
/Applications/Xcode.app/Contents/Developer/usr/bin/make  -f PySide2/QtSql/CMakeFiles/QtSql_generator.dir/build.make PySide2/QtSql/CMakeFiles/QtSql_generator.dir/depend
/Applications/Xcode.app/Contents/Developer/usr/bin/make  -f PySide2/QtNetwork/CMakeFiles/QtNetwork_generator.dir/build.make PySide2/QtNetwork/CMakeFiles/QtNetwork_generator.dir/depend
/Applications/Xcode.app/Contents/Developer/usr/bin/make  -f PySide2/QtTest/CMakeFiles/QtTest_generator.dir/build.make PySide2/QtTest/CMakeFiles/QtTest_generator.dir/depend
/Applications/Xcode.app/Contents/Developer/usr/bin/make  -f PySide2/QtConcurrent/CMakeFiles/QtConcurrent_generator.dir/build.make PySide2/QtConcurrent/CMakeFiles/QtConcurrent_generator.dir/depend
cd /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/macbuild5.15.5 && /opt/homebrew/Cellar/cmake/3.27.8/bin/cmake -E cmake_depends "Unix Makefiles" /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/sources/pyside2 /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/sources/pyside2/libpyside /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/macbuild5.15.5 /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/macbuild5.15.5/libpyside /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/macbuild5.15.5/libpyside/CMakeFiles/pyside2.dir/DependInfo.cmake "--color="
cd /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/macbuild5.15.5 && /opt/homebrew/Cellar/cmake/3.27.8/bin/cmake -E cmake_depends "Unix Makefiles" /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/sources/pyside2 /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/sources/pyside2/PySide2/QtWidgets /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/macbuild5.15.5 /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/macbuild5.15.5/PySide2/QtWidgets /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/macbuild5.15.5/PySide2/QtWidgets/CMakeFiles/QtWidgets_generator.dir/DependInfo.cmake "--color="
cd /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/macbuild5.15.5 && /opt/homebrew/Cellar/cmake/3.27.8/bin/cmake -E cmake_depends "Unix Makefiles" /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/sources/pyside2 /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/sources/pyside2/PySide2/QtCore /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/macbuild5.15.5 /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/macbuild5.15.5/PySide2/QtCore /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/macbuild5.15.5/PySide2/QtCore/CMakeFiles/QtCore_generator.dir/DependInfo.cmake "--color="
cd /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/macbuild5.15.5 && /opt/homebrew/Cellar/cmake/3.27.8/bin/cmake -E cmake_depends "Unix Makefiles" /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/sources/pyside2 /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/sources/pyside2/PySide2/QtGui /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/macbuild5.15.5 /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside-setup-opensource-src-5.15.5/macbuild5.15.5/PySide2/QtGui /tmp/pyside2A5.15.5-20231124-96353-wx3b0v/pyside [... cut ...] /homebrew/opt/gettext/bin:/opt/homebrew/opt/glib/bin:/opt/homebrew/opt/jpeg-turbo/bin:/opt/homebrew/opt/giflib/bin:/opt/homebrew/opt/libtiff/bin:/opt/homebrew/opt/webp/bin:/opt/homebrew/opt/qt@5/bin:/opt/homebrew/opt/shiboken2@5.15.5/bin:/opt/homebrew/opt/docutils/bin:/opt/homebrew/opt/pygments/bin:/opt/homebrew/opt/python-tabulate/bin:/opt/homebrew/opt/sphinx-doc/bin:/opt/homebrew/opt/python@3.12/libexec/bin:/opt/homebrew/opt/python@3.10/libexec/bin:/usr/bin:/bin:/usr/sbin:/sbin
```

</details>

ipatch, 2023-11-25:
the below pull request is the first step in fixing the issue. then i'll need to update the freecad formula to use the new version of pyside2. it'll take a couple of hours for the self-hosted runners to build each version pyside ie. ~ 1hr per version.

https://github.com/FreeCAD/homebrew-freecad/pull/435

ipatch, 2023-11-29:
i'm still working on this. feel free if you can install the new version of pyside2 on your _apple silicon_ mac. unfortunately i can not presently build bottles for _apple silicon_ with a standard `/opt/local` or /usr/local` prefix. thus rendering the bottles pretty much useless. but one can attempt to install the new formula with,

```
brew install pyside2@5.15.11 -v
```

superfloh247, 2023-11-29:
brew on apple silicon uses /opt/homebrew

or is my installation somewhat broken?

superfloh247, 2023-11-29:
Error: freecad/freecad/pyside2@5.15.11 5.15.11 did not build

I don't see any errors in all of the log files

ipatch, 2023-11-30:
both,

```
brew install shiboken2@5.15.11 -v;
brew install pyside2@5.15.11 -v;
```

should be working now. 🤞

ipatch, 2023-11-30:
wait for #446 to be merged and hopefully everything with `pyside2@5.15.11` should be good.

ipatch, 2023-11-30:
should ...should be fixed now🤞

superfloh247, 2023-12-01:
I'm probably doing it wrong, it still fails

brew update && brew autoremove && brew cleanup --prune=all && brew outdated

brew upgrade --no-quarantine --verbose

brew install --only-dependencies freecad

brew doctor - everything is fine

ipatch, 2023-12-01:
i haven't updated the freecad formula itself, as i had to update the dependencies of freecad. that's why your still seeing an error related to `pyside2@5.15.5` hopefully i'll get the actual freecad formula updated to the latest release in a day or two. 🤞

superfloh247, 2023-12-01:
yay :)

/opt/homebrew/Cellar/pyside2@5.15.11/5.15.11_1: 296 files, 54.8MB, built in 5 minutes 53 seconds

ipatch, 2023-12-10:
hey i'm in the process of updating the freecad formula file itself now. i've pushed a rough draft to myfork of this repo, but am still running into issues using `brew install` as it's finding / using my python@3.12 and i need the `brew install` step to find and use `python@3.11` as defined in the updated formula. anyways more info in the link below,

https://github.com/orgs/Homebrew/discussions/4968

stefandiaconu, 2024-04-14:
For me still fails to install pyside@2.15.5

I use `brew install freecad/freecad/pyside2@5.15.5` and I get this message:

```
==> Fetching freecad/freecad/pyside2@5.15.5
==> Downloading https://download.qt.io/official_releases/QtForPython/pyside2/PySide2-5.15.5-
==> Downloading from https://mirrors.20i.com/pub/qt.io/official_releases/QtForPython/pyside2
##################################################################################### 100.0%
==> Installing pyside2@5.15.5 from freecad/freecad
==> cmake -DPYTHON_EXECUTABLE=/opt/homebrew/opt/python@3.10/Frameworks/Python.framework/Vers
==> make -j8
Last 15 lines from /Users/myusername/Library/Logs/Homebrew/pyside2@5.15.5/02.make:
(printsupport) [2292ms] Detecting inconsistencies in typesystem (7705)...           [OK]
(widgets) [2295ms] Checking inconsistencies in function modifications...       [OK]
(widgets) [2296ms] Writing log files...                                        [OK]
(printsupport) [2314ms] Checking inconsistencies in function modifications...       [OK]
(printsupport) [2314ms] Writing log files...                                        [OK]
(widgets) [2321ms] Running Source generator...                                 [OK]
(printsupport) [2338ms] Running Source generator...                                 [OK]
(printsupport) [2357ms] Running Header generator...                                 [OK]
Done, (printsupport) 2359ms, 4 warnings (1215 known issues)
[  2%] Built target QtPrintSupport_generator
[OK]
(widgets) [2760ms] Running Header generator...                                 [OK]
Done, (widgets) 2802ms, 16 warnings (1211 known issues)
[  2%] Built target QtWidgets_generator
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/freecad/homebrew-freecad/issues

These open issues may also help:
brew install --only-dependencies freecad fails in step freecad/freecad/pyside2@5.15.5 https://github.com/FreeCAD/homebrew-freecad/issues/433
```

I use terminal on macOS Sonoma 14.4.1 running on M1 chip.

I did started the process from the beginning few times but still the same issue.

I can see that shiboken2@5.15.5 does install but no pyside@2.15.5.

I did `brew install --only-dependencies freecad` but still fails at the same step pyside@2.15.5.

Any idea what could be?

ipatch, 2024-04-14:
@stefandiaconu 

hey thanks for the follow up. believe or not, i have been attempting to resolve errors within this repo. i just have limited resources to debug certain issues at the moment. i'm in the process of updating the deps for one of the virtual machines (mojave) so i can debug the `brew install freecad` cmd and why it crashes on launch. you can see my progress in the below PR,

https://github.com/FreeCAD/homebrew-freecad/pull/488

if you're eager to run freecad on your mac i'd suggest giving the conda version a try as mentioned in the main readme for the repo. and if you're trying to just build freecad from source then try installing the deps (formula) in this repo that the `py310` suffix. i am able to build freecad using the deps along with cmake to produce a working freecad install. unfortunately at the moment using `brew install freecad` will produce a binary that crashes on launch. and the m1 systems i have access i can not unfortunately use lldb to examine a core dump at the moment. hence why i am updating my mojave vm to hopefully go through the `brew install freecad` steps and then hopefully everything will start moving a little quicker in the repo. but if you are eager to build freecad yourself take a look at the above mentioned PR and you'll see most of the cmake args i use to build freecad, and it does work if you build freecad outside of `brew install`.

happy hacking.

ipatch, 2024-05-01:
please comment if you are still having an issue, and i will reopen.

closed 2024-05-01 by ipatch