FreeCAD/homebrew-freecad #434: "brew install --only-dependencies freecad" error building PySide2 QtSensors

jlmuir, 2023-11-24:
I followed the instructions at [Compile on MacOS](https://wiki.freecad.org/Compile_on_MacOS)

```
$ brew tap freecad/freecad
$ brew install eigen
$ brew install --only-dependencies freecad
```

but the build failed with `make: *** [all] Error 2` partway through the last command.

Environment:
* Chip: M1 Pro
* OS: Sonoma 14.1.1
* Xcode: 15.0.1

The end of the output of the last command is as follows:

```
[ 35%] Built target QtLocation
[ 35%] Linking CXX shared module ../QtSensors.cpython-310-darwin.so
cd /tmp/pyside2A5.15.5-20231124-92097-h2kpnc/pyside-setup-opensource-src-5.15.5/macbuild5.15.5/PySide2/QtSensors && /opt/homebrew/Cellar/cmake/3.27.8/bin/cmake -E cmake_link_script CMakeFiles/QtSensors.dir/link.txt --verbose=1
/opt/homebrew/Library/Homebrew/shims/mac/super/clang++ -std=c++17 -Wall -fvisibility=hidden -Wno-strict-aliasing -O3 -DNDEBUG -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.sdk -bundle -Wl,-headerpad_max_install_names -o ../QtSensors.cpython-310-darwin.so CMakeFiles/QtSensors.dir/PySide2/QtSensors/qsensorgesturemanager_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qsensorgestureplugininterface_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qsensorgesturerecognizer_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qaccelerometer_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qaccelerometerfilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qaccelerometerreading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qaltimeter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qaltimeterfilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qaltimeterreading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qambientlightfilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qambientlightreading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qambientlightsensor_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qambienttemperaturefilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qambienttemperaturereading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qambienttemperaturesensor_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qcompass_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qcompassfilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qcompassreading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qdistancefilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qdistancereading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qdistancesensor_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qgyroscope_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qgyroscopefilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qgyroscopereading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qholsterfilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qholsterreading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qholstersensor_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qhumidityfilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qhumidityreading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qhumiditysensor_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qirproximityfilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qirproximityreading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qirproximitysensor_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qlidfilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qlidreading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qlidsensor_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qlightfilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qlightreading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qlightsensor_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qmagnetometer_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qmagnetometerfilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qmagnetometerreading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qorientationfilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qorientationreading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qorientationsensor_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qpressurefilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qpressurereading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qpressuresensor_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qproximityfilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qproximityreading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qproximitysensor_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qrotationfilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qrotationreading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qrotationsensor_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qsensor_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qsensorfilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qsensorreading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qoutputrange_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qsensorbackend_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qsensorbackendfactory_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qsensormanager_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qsensorchangesinterface_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qsensorplugininterface_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qtapfilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qtapreading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qtapsensor_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qtiltfilter_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qtiltreading_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qtiltsensor_wrapper.cpp.o CMakeFiles/QtSensors.dir/PySide2/QtSensors/qtsensors_module_wrapper.cpp.o -iframework /opt/homebrew/opt/qt@5/lib  -Wl,-rpath,/tmp/pyside2A5.15.5-20231124-92097-h2kpnc/pyside-setup-opensource-src-5.15.5/macbuild5.15.5/libpyside ../../libpyside/libpyside2.cpython-310-darwin.5.15.5.dylib /opt/homebrew/opt/qt@5/lib/QtSensors.framework/QtSensors /opt/homebrew/opt/shiboken2@5.15.5/lib/libshiboken2.cpython-310-darwin.5.15.5.dylib -undefined dynamic_lookup /opt/homebrew/opt/qt@5/lib/QtCore.framework/QtCore
[ 35%] Built target QtSensors
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/freecad/homebrew-freecad/issues

These open issues may also help:
brew install --only-dependencies freecad fails in step freecad/freecad/pyside2@5.15.5 https://github.com/FreeCAD/homebrew-freecad/issues/433
```

ipatch, 2023-11-24:
hey thanks for reporting this issue. 🙏 i'm going to close this issue as this same issue was reported about an hour earlier, and you're likely experiencing build failures due to the `pyside2@5.15.5` build failure.

please see the below issue for updates

ipatch, 2023-11-24:
#433

closed 2023-11-24 by ipatch