FreeCAD/homebrew-freecad #475: "Python.h no such file or directory" error on compilation of freecad/freecad/med-file@4.1.1

kaczuch, 2024-03-01:
When trying to install freecad with:
`brew install freecad`
or just med-file dependency with:
`brew install med-file`
I got error when building
`
[ 96%] Building CXX object python/CMakeFiles/_medenumtest.dir/med/medenumtest_modulePYTHON_wrap.cxx.o
cd /tmp/med-fileA4.1.1-20240302-33450-jfibmu/med-4.1.1/build/python && /home/linuxbrew/.linuxbrew/Homebrew/Library/Homebrew/shims/linux/super/g++-13 -DH5_BUILT_AS_DYNAMIC_LIB -DH5_USE_16_API -DH5_USE_18_API -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGEFILE64_SOURCE -D_LARGEFILE_SOURCE -D_POSIX_C_SOURCE=200809L -D_medenumtest_EXPORTS -I/home/linuxbrew/.linuxbrew/include/python3.12 -I/tmp/med-fileA4.1.1-20240302-33450-jfibmu/med-4.1.1/include -I/tmp/med-fileA4.1.1-20240302-33450-jfibmu/med-4.1.1/build/include -O3 -DNDEBUG -fPIC -MD -MT python/CMakeFiles/_medenumtest.dir/med/medenumtest_modulePYTHON_wrap.cxx.o -MF CMakeFiles/_medenumtest.dir/med/medenumtest_modulePYTHON_wrap.cxx.o.d -o CMakeFiles/_medenumtest.dir/med/medenumtest_modulePYTHON_wrap.cxx.o -c /tmp/med-fileA4.1.1-20240302-33450-jfibmu/med-4.1.1/build/python/med/medenumtest_modulePYTHON_wrap.cxx
/tmp/med-fileA4.1.1-20240302-33450-jfibmu/med-4.1.1/build/python/med/medenum_modulePYTHON_wrap.cxx:168:11: fatal error: Python.h: No such file or directory
  168 | # include <Python.h>
      |           ^~~~~~~~~~
compilation terminated.
`
My system is Kubuntu 22.04
I don't know how brew manages its dependencies, but I've tried to reinstall python `with brew reinstall python`.

ipatch, 2024-03-02:
i've run into this issue on linuxbrew as well. the homebrew-freecad formula file of med is built against the upstream homebrew version of python@3.11. i believe homebrew recently switched to python@3.12 as the main / primary version of python. and in the past when i have run into similar issues it is because python@3.12 is linked in your cellar (if memory serves me correctly) so maybe try unlinking whatever version of python you have linked in your cellar, and make sure to have `python@3.11` linked.

if the issue still persists i'll try and replicate locally on my system.

ipatch, 2024-03-05:
https://github.com/FreeCAD/homebrew-freecad/pull/478

the above pr should resolve the `Python.h` error mentioned above. tested it with the github hosted runners. also tested it with my local linuxbrew install with python@3.12 linked in my Cellar, and the install is able to build the necessary python module.

going to close this issue now, but if the issue is still persisting please comment below.

closed 2024-03-05 by ipatch