FreeCAD/homebrew-freecad #547: freecad@0.21.2_py310_qt6: setup new freecad formula that builds and links against upstream qt@6

ipatch, 2024-07-21:
need to setup a new formula that builds and links against qt@6 upstream homebrew-core formula.

ipatch, 2024-07-21:
currently running into the below issue because upstream pyside@6 has the force limited api set within the formula.

https://forum.freecad.org/viewtopic.php?t=77963&start=40

ipatch, 2024-07-21:
a workaround for the above issue is to install a versioned formula of pyside that sets the below cmake var,

```
-DFORCE_LIMITED_API=no
```

ipatch, 2024-07-21:
it appears a patch may be required to fix the below error,

```
[ 57%] Building CXX object src/Mod/Surface/App/CMakeFiles/Surface.dir/Blending/FeatureBlendCurve.cpp.o
[ 57%] Built target Import
[ 57%] Building CXX object src/Mod/Robot/App/CMakeFiles/Robot.dir/TrajectoryObject.cpp.o
/Users/ipatch/code/freecad-git/freecad-src/src/Mod/MeshPart/App/MeshFlatteningPy.cpp:39:10: fatal error: 'pybind11/eigen.h' file not found
#include <pybind11/eigen.h>
         ^~~~~~~~~~~~~~~~~~
[ 58%] Built target Sketche
```

using the patch supplied within this repo does bump my build further without encourting this fatal error message.

ipatch, 2024-07-21:
the current error i'm getting when using my versioned version of pyside@6 and applying several patches to the freecad source, and using the below cmake command to configure my build directory,

```
cmake \
                                                 -DFREECAD_QT_VERSION=6 \
                                                 -DCMAKE_FIND_USE_SYSTEM_ENVIRONMENT_PATH=FALSE \
                                                 -DCMAKE_FIND_USE_CMAKE_SYSTEM_PATH=FALSE \
                                                 -DCMAKE_PREFIX_PATH=\
                                             "$pthymlcpp;$pthxercesc;$pthzlib;$pthocc;$pthqt6;$pthboost;$pthvtk;$pthmed;$ptheigen;$pthswig;$pthcups;$pthhdf5mpi;$pthfreetype;$pthexpat;$pthdoubleconversion;$pthlz4;$pthlzma;$pthlibjpeg;$pthpkgconfig;$pthutf8cpp;$pthdoxygen;$pthpugixml;$pthglew;$pthlibpng;$pthlibtiff;$pthcoin;$pthfmt;$pthpybind;$pthhbfcpyside6;$pthvlknhdrs;" \
                                                 -DOPENGL_INCLUDE_DIR=/System/Library/Frameworks/OpenGL.framework \
                                                 -DOPENGL_gl_LIBRARY=/System/Library/Frameworks/OpenGL.framework \
                                                 -DINSTALL_TO_SITEPACKAGES=0 \
                                                 -DOPENGL_GLU_INCLUDE_DIR=/System/Library/Frameworks/OpenGL.framework \
                                                 -DOPENGL_glu_LIBRARY=/System/Library/Frameworks/OpenGL.framework \
                                                 -DHDF5_DIR=$bp/opt/hdf5-mpi \
                                                 -DQT_DEBUG_FIND_PACKAGE=ON \
                                                 -DFREECAD_USE_PYBIND11=1 \
                                                 -DHOMEBREW_PREFIX=$bp \
                                                 -L \
                                                  ../../freecad-src
```

---

error message,

```
[ 87%] Generating qrc_Assembly.cpp
[ 87%] Generating qrc_Assembly_translation.cpp
In file included from /Users/ipatch/code/freecad-git/freecad-src/src/Mod/TechDraw/Gui/TaskSelectLineAttributes.cpp:27:
/Users/ipatch/code/freecad-git/freecad-src/src/Mod/TechDraw/App/LineGenerator.h:84:45: error: implicit instantiation of undefined template 'std::vector<std::vector<std::string>>'
    std::vector< std::vector<std::string> > m_lineDefs;
                                            ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/usr/include/c++/v1/iosfwd:259:28: note: template is declared here
class _LIBCPP_TEMPLATE_VIS vector;
                           ^
In file included from /Users/ipatch/code/freecad-git/freecad-src/src/Mod/TechDraw/Gui/TaskSelectLineAttributes.cpp:27:
/Users/ipatch/code/freecad-git/freecad-src/src/Mod/TechDraw/App/LineGenerator.h:85:32: error: implicit instantiation of undefined template 'std::vector<std::string>'
    std::vector< std::string > m_lineDescs;
                               ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/usr/include/c++/v1/iosfwd:259:28: note: template is declared here
class _LIBCPP_TEMPLATE_VIS vector;
                           ^
^Cmake[2]: *** Deleting file `src/Mod/Assembly/Gui/qrc_Assembly_translation.cpp'
```

ipatch, 2024-07-21:
it seems the above error message is related to the macos toolchain (possibly) and workaround / fix is to add the missing include for `<vector>` to the `LineGenerator.h` file. see the below link for more information.

https://stackoverflow.com/questions/49246160/xcode-c-vectors-implicit-instantiation-of-undefined-template

ipatch, 2024-07-23:
doing work within this branch of my local fork

https://github.com/ipatch/homebrew-freecad/tree/freecad-0.21.2_qt6

ipatch, 2024-07-23:
current cmake issue i'm presently running into,

```
CMake Error at src/3rdParty/CMakeLists.txt:11 (message):
  The OndselSolver git submodule is not available.  Please run

          git submodule update --init

CMake Error at src/3rdParty/CMakeLists.txt:14 (add_subdirectory):
  The source directory

    /Volumes/STORAGE/ipatch/homebrew/tmp/freecadA0.21.2_py312_qt6-20240722-585-oxws0d/FreeCAD-ec586668d9794afdfe574887f16c215f744ee8c0/src/3rdParty/OndselSolver

  does not contain a CMakeLists.txt file.

```

ipatch, 2024-07-23:
either need a new formula file for the ondselsolver or i need to figure out how i can add it as a resource to the current formula 🤔

ipatch, 2024-07-23:
attempted to add the ondselsolver as a resource but unfortunately that currently is not working,

```
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
CMake Error at src/3rdParty/CMakeLists.txt:11 (message):
  The OndselSolver git submodule is not available.  Please run

          git submodule update --init

CMake Error at src/3rdParty/CMakeLists.txt:14 (add_subdirectory):
  The source directory

    /Volumes/STORAGE/ipatch/homebrew/tmp/freecadA0.21.2_py312_qt6-20240722-9304-25alia/FreeCAD-ec586668d9794afdfe574887f16c215f744ee8c0/src/3rdParty/OndselSolver

  does not contain a CMakeLists.txt file.

-- /Volumes/STORAGE/ipatch/homebrew/tmp/freecadA0.21.2_py312_qt6-20240722-9304-25alia/build/src/Base/TypePy.cpp
```

ipatch, 2024-07-23:
setup a tap formula for the ondselsolver see new pr #549

ipatch, 2024-07-25:
it seems using the below cmake var does not prevent the cmake process from searching for the ondsel solver

```
FREECAD_USE_EXTERNAL_ONDSELSOLVER:BOOL=OFF
```

from what i can tell because i'm still getting the above cmake warnings/errors.

ipatch, 2024-07-25:
```
/Volumes/STORAGE/ipatch/homebrew/tmp/freecadA0.21.2_py312_qt6-20240725-45539-m01701/FreeCAD-ec586668d9794afdfe574887f16c215f744ee8c0/src/3rdParty/OndselSolver/OndselSolver/FullVector.h:183:57: warning: unused parameter 'tol' [-Wunused-parameter]
        inline void FullVector<T>::conditionSelfWithTol(double tol)
                                                               ^
In file included from /Volumes/STORAGE/ipatch/homebrew/tmp/freecadA0.21.2_py312_qt6-20240725-45539-m01701/FreeCAD-ec586668d9794afdfe574887f16c215f744ee8c0/src/Base/Builder3D.cpp:45:
In file included from /Volumes/STORAGE/ipatch/homebrew/tmp/freecadA0.21.2_py312_qt6-20240725-45539-m01701/FreeCAD-ec586668d9794afdfe574887f16c215f744ee8c0/src/Base/Tools.h:37:
In file included from /Users/ipatch/homebrew/opt/qt/lib/QtCore.framework/Headers/QString:1:
In file included from /Users/ipatch/homebrew/opt/qt/lib/QtCore.framework/Headers/qstring.h:22:
In file included from /Users/ipatch/homebrew/opt/qt/include/QtCore/qanystringview.h:9:
/Users/ipatch/homebrew/opt/qt/include/QtCore/qutf8stringview.h:278:19: error: unknown type name 'Q_IMPLICIT'
    [[nodiscard]] Q_IMPLICIT operator std::basic_string_view<storage_type>() const noexcept
                  ^
/Users/ipatch/homebrew/opt/qt/include/QtCore/qutf8stringview.h:303:16: error: no matching function for call to 'compareStrings'
        return QtPrivate::compareStrings(QBasicUtf8StringView<false>(lhs.data(), lhs.size()),
               ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/ipatch/homebrew/opt/qt@5/include/QtCore/qstringalgorithms.h:62:58: note: candidate function not viable: no known conversion from 'QBasicUtf8StringView<false>' to 'QStringView' for 1st argument
Q_REQUIRED_RESULT Q_CORE_EXPORT Q_DECL_PURE_FUNCTION int compareStrings(QStringView   lhs, QStringView   rhs, Qt::CaseSensitivity cs = Qt::CaseSensitive) noexcept;
                                                         ^
/Users/ipatch/homebrew/opt/qt@5/include/QtCore/qstringalgorithms.h:63:58: note: candidate function not viable: no known conversion from 'QBasicUtf8StringView<false>' to 'QStringView' for 1st argument
Q_REQUIRED_RESULT Q_CORE_EXPORT Q_DECL_PURE_FUNCTION int compareStrings(QStringView   lhs, QLatin1String rhs, Qt::CaseSensitivity cs = Qt::CaseSensitive) noexcept;
                                                         ^
/Users/ipatch/homebrew/opt/qt@5/include/QtCore/qstringalgorithms.h:64:58: note: candidate function not viable: no known conversion from 'QBasicUtf8StringView<false>' to 'QLatin1String' for 1st argument
Q_REQUIRED_RESULT Q_CORE_EXPORT Q_DECL_PURE_FUNCTION int compareStrings(QLatin1String lhs, QStringView   rhs, Qt::CaseSensitivity cs = Qt::CaseSensitive) noexcept;
                                                         ^
/Users/ipatch/homebrew/opt/qt@5/include/QtCore/qstringalgorithms.h:65:58: note: candidate function not viable: no known conversion from 'QBasicUtf8StringView<false>' to 'QLatin1String' for 1st argument
Q_REQUIRED_RESULT Q_CORE_EXPORT Q_DECL_PURE_FUNCTION int compareStrings(QLatin1String lhs, QLatin1String rhs, Qt::CaseSensitivity cs = Qt::CaseSensitive) noexcept;
```

cmake appears to be picking qt@5 installation

ipatch, 2024-08-02:
sucessfully built freecat from git main this afternoon using cmake with homebrew provided libaries, the below cmake command allowed me build fc from the below commit,

```
 cmake \                                                                                                             1 (0.001s)
        -DHOMEBREW_PREFIX=$bp \
        -DCMAKE_FIND_USE_SYSTEM_ENVIRONMENT_PATH=FALSE \
        -DCMAKE_FIND_USE_CMAKE_SYSTEM_PATH=FALSE \
        -DOPENGL_INCLUDE_DIR=$pthmacfrmwkgl/Headers \
        -DOPENGL_gl_LIBRARY=$pthmacfrmwkgl/OpenGL.tbd \
        -DOPENGL_GLU_INCLUDE_DIR=$pthmacfrmwkgl/Headers \
        -DOPENGL_glu_LIBRARY=$pthmacfrmwkgl/Libraries/libGLU.tbd \
        -DFREECAD_USE_PYBIND11=1 \
        -DCMAKE_AR=/usr/bin/ar \
        -DCMAKE_PREFIX_PATH="$pthqt6;$pthdoxygen;$pthymlcpp;$pthxercesc;$pthzlib;$pthocc;$pthboost;$pthvtk;$pthhdf5mpi;\
    $pthmed;$ptheigen;$pthcups;$pthcoin3dpy312;$pthpybind;$pthfmt;$pthswig;$pthfreetype;$pthglew;$pthpugixml;$pthpyside6;\
    $pthninja;$pthmpi;" \
        -G Ninja \
        -DCMAKE_BUILD_TYPE=RelWithDebInfo \
        -DCMAKE_INSTALL_PREFIX=$HOME/.local/fc-git \
        -L \
        ../../fcsrc
```

that above cmake does require setting up env vars for the path to the libaries and setting a env var of `bp` to the output of `$(brew --prefix)`

also i still have to apply the xerces-c patch, and the pybind11 patch contained within this repo or build errors will occur.

also i set the compilers to use the open-mpi compiler wrappers ie.

```
 export CC=$bp/opt/open-mpi/bin/mpicc     
 export CXX=$bp/opt/open-mpi/bin/mpic++   
 ```
 
 all that said, freecad does launch, but
 
 1. can not create a body because i'm getting the below error,
 
 ```
       File "<string>", line 1, in <module>
    <class 'RuntimeError'>: {'sclassname': 'N4Base12RuntimeErrorE', 'sErrMsg': 'GeoFeatureGroupExtension can only be applied to GeoFeatures', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'GeoFeatureGroupExtension can only be applied to GeoFeatures', 'btranslatable': False, 'breported': False}
    "
```

2. any workbench that is using PySide fails due to trying to load a py module of the name `PySide` which opening a python repl ie.

```
$bp/opt/python@3.12/bin/python3
```

and running,

```
import PySide
```

will produce the below error,

```
Python 3.12.4 (main, Jun  6 2024, 18:26:44) [Clang 15.0.0 (clang-1500.1.0.2.5)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import PySide
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'PySide'
```

i can import the python module with the name `PySide6`

so further investigation is required to figure out how to get PySide working when building and linking against qt6

ipatch, 2024-08-02:
## useful links, upstream related to using qt6 with freecad

https://github.com/FreeCAD/FreeCAD/issues/13303

chennes, 2024-08-02:
For the record, FREECAD_USE_EXTERNAL_ONDSELSOLVER is considered an advanced option designed for developers who are working on OndselSolver itself. In most cases self-compilers should be using `git submodule update --init` to get OndselSolver as part of the main FreeCAD source tree.

ipatch, 2024-08-02:
running `cmake -L` in my build directory isn't even listing the `FREECAD_USE_EXTERNAL_ONDSELSOLVER` option being available.

```
-- Generating done (1.7s)
-- Build files have been written to: /Users/chris/code/freecad-git/builds/git002
-- Cache values
3DCONNEXIONCLIENT_FRAMEWORK:FILEPATH=3DCONNEXIONCLIENT_FRAMEWORK-NOTFOUND
BUILD_ADDONMGR:BOOL=ON
BUILD_ASSEMBLY:BOOL=ON
BUILD_BIM:BOOL=ON
BUILD_CLOUD:BOOL=OFF
BUILD_DESIGNER_PLUGIN:BOOL=OFF
BUILD_DRAFT:BOOL=ON
BUILD_DRAWING:BOOL=OFF
BUILD_DYNAMIC_LINK_PYTHON:BOOL=ON
BUILD_ENABLE_CXX_STD:STRING=C++17
BUILD_FEM:BOOL=ON
BUILD_FEM_NETGEN:BOOL=OFF
BUILD_FLAT_MESH:BOOL=ON
BUILD_FORCE_DIRECTORY:BOOL=OFF
BUILD_GMOCK:BOOL=ON
BUILD_GUI:BOOL=ON
BUILD_HELP:BOOL=ON
BUILD_IDF:BOOL=ON
BUILD_IMPORT:BOOL=ON
BUILD_INSPECTION:BOOL=ON
BUILD_JTREADER:BOOL=OFF
BUILD_MATERIAL:BOOL=ON
BUILD_MEASURE:BOOL=ON
BUILD_MESH:BOOL=ON
BUILD_MESH_PART:BOOL=ON
BUILD_OPENSCAD:BOOL=ON
BUILD_PART:BOOL=ON
BUILD_PART_DESIGN:BOOL=ON
BUILD_PATH:BOOL=ON
BUILD_PLOT:BOOL=ON
BUILD_POINTS:BOOL=ON
BUILD_REVERSEENGINEERING:BOOL=ON
BUILD_ROBOT:BOOL=ON
BUILD_SANDBOX:BOOL=OFF
BUILD_SHOW:BOOL=ON
BUILD_SKETCHER:BOOL=ON
BUILD_SPREADSHEET:BOOL=ON
BUILD_START:BOOL=ON
BUILD_SURFACE:BOOL=ON
BUILD_TECHDRAW:BOOL=ON
BUILD_TEMPLATE:BOOL=OFF
BUILD_TEST:BOOL=ON
BUILD_TESTING:BOOL=ON
BUILD_TUX:BOOL=ON
BUILD_VR:BOOL=OFF
BUILD_WEB:BOOL=ON
BUILD_WITH_CONDA:BOOL=OFF
Boost_DATE_TIME_LIBRARY_RELEASE:STRING=/Users/chris/homebrew/Cellar/boost/1.85.0/lib/libboost_date_time-mt.dylib
Boost_FILESYSTEM_LIBRARY_RELEASE:STRING=/Users/chris/homebrew/Cellar/boost/1.85.0/lib/libboost_filesystem-mt.dylib
Boost_INCLUDE_DIR:PATH=/Users/chris/homebrew/Cellar/boost/1.85.0/include
Boost_PROGRAM_OPTIONS_LIBRARY_RELEASE:STRING=/Users/chris/homebrew/Cellar/boost/1.85.0/lib/libboost_program_options-mt.dylib
Boost_REGEX_LIBRARY_RELEASE:STRING=/Users/chris/homebrew/Cellar/boost/1.85.0/lib/libboost_regex-mt.dylib
Boost_SYSTEM_LIBRARY_RELEASE:STRING=/Users/chris/homebrew/Cellar/boost/1.85.0/lib/libboost_system-mt.dylib
Boost_THREAD_LIBRARY_RELEASE:STRING=/Users/chris/homebrew/Cellar/boost/1.85.0/lib/libboost_thread-mt.dylib
CCACHE_PROGRAM:FILEPATH=CCACHE_PROGRAM-NOTFOUND
CMAKE_BUILD_TYPE:STRING=RelWithDebInfo
CMAKE_INSTALL_PREFIX:PATH=/Users/chris/.local/fc-git
CMAKE_OSX_ARCHITECTURES:STRING=
CMAKE_OSX_DEPLOYMENT_TARGET:STRING=
CMAKE_OSX_SYSROOT:PATH=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.2.sdk
COIN3D_DOC_FOUND:BOOL=NO
E57_BUILD_SHARED:BOOL=OFF
E57_CLANG_FORMAT_EXE:FILEPATH=E57_CLANG_FORMAT_EXE-NOTFOUND
E57_DEBUG:BOOL=ON
[... 55 lines cut ...]
Qt6LinguistTools_DIR:PATH=/Users/chris/homebrew/opt/qt6/lib/cmake/Qt6LinguistTools
Qt6Network_DIR:PATH=/Users/chris/homebrew/opt/qt6/lib/cmake/Qt6Network
Qt6OpenGLWidgets_DIR:PATH=/Users/chris/homebrew/opt/qt6/lib/cmake/Qt6OpenGLWidgets
Qt6OpenGL_DIR:PATH=/Users/chris/homebrew/opt/qt6/lib/cmake/Qt6OpenGL
Qt6PrintSupport_DIR:PATH=/Users/chris/homebrew/opt/qt6/lib/cmake/Qt6PrintSupport
Qt6SvgWidgets_DIR:PATH=/Users/chris/homebrew/opt/qt6/lib/cmake/Qt6SvgWidgets
Qt6Svg_DIR:PATH=/Users/chris/homebrew/opt/qt6/lib/cmake/Qt6Svg
Qt6Test_DIR:PATH=/Users/chris/homebrew/opt/qt6/lib/cmake/Qt6Test
Qt6UiPlugin_DIR:PATH=/Users/chris/homebrew/opt/qt6/lib/cmake/Qt6UiPlugin
Qt6UiTools_DIR:PATH=/Users/chris/homebrew/opt/qt6/lib/cmake/Qt6UiTools
Qt6WidgetsTools_DIR:PATH=/Users/chris/homebrew/opt/qt6/lib/cmake/Qt6WidgetsTools
Qt6Widgets_DIR:PATH=/Users/chris/homebrew/opt/qt6/lib/cmake/Qt6Widgets
Qt6Xml_DIR:PATH=/Users/chris/homebrew/opt/qt6/lib/cmake/Qt6Xml
Qt6_DIR:PATH=/Users/chris/homebrew/opt/qt6/lib/cmake/Qt6
SHIBOKEN_FOUND:BOOL=ON
SHIBOKEN_INCLUDE_DIR:PATH=
Shiboken6_DIR:PATH=/Users/chris/homebrew/opt/pyside6_py312/lib/cmake/Shiboken6
Tiff_DIR:PATH=Tiff_DIR-NOTFOUND
VTK_DIR:PATH=/Users/chris/homebrew/opt/vtk/lib/cmake/vtk-9.3
WrapOpenGL_AGL:FILEPATH=WrapOpenGL_AGL-NOTFOUND
fmt_DIR:PATH=/Users/chris/homebrew/opt/fmt/lib/cmake/fmt
meddotH:FILEPATH=/Users/chris/homebrew/opt/med-file@4.1.1/include/med.h
pugixml_DIR:PATH=/Users/chris/homebrew/opt/pugixml/lib/cmake/pugixml
yaml-cpp_DIR:PATH=/Users/chris/homebrew/opt/yaml-cpp/lib/cmake/yaml-cpp
```

ipatch, 2024-08-13:
i think the issue related to PySide can be seen in the below `freecadcmd` repl session, i think a possilbe patch file will be required to properly set things up.

```
╰─λ freecadcmd                                                                                                          0 (0.021s)
FreeCAD 0.21.2, Libs: 0.21.2RUnknown
(c) Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

[FreeCAD Console mode <Use Ctrl-D (i.e. EOF) to exit.>]
>>> sys.path
['/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Material', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Import', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Part', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Tux', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/PartDesign', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Measure', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Points', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/AddonManager', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Arch', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Idf', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Inspection', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Robot', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Surface', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/OpenSCAD', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Start', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Sketcher', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Show', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/MeshPart', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Path', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/TechDraw', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Web', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Spreadsheet', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Fem', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Test', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Plot', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/ReverseEngineering', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Draft', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Mesh', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/lib', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Ext', '', '/Users/chris/homebrew/Cellar/python@3.10/3.10.14_1/Frameworks/Python.framework/Versions/3.10/lib/python310.zip', '/Users/chris/homebrew/Cellar/python@3.10/3.10.14_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10', '/Users/chris/homebrew/Cellar/python@3.10/3.10.14_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/lib-dynload', '/Users/chris/homebrew/Cellar/python@3.10/3.10.14_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages', '/Users/chris/homebrew/Cellar/pyside2@5.15.11_py310/5.15.11/lib/python3.10/site-packages', '/Users/chris/homebrew/Cellar/coin3d_py310/4.0.2/lib/python3.10/site-packages', '/Users/chris/homebrew/Cellar/numpy@1.26.4_py310/1.26.4/lib/python3.10/site-packages', '/Users/chris/homebrew/Cellar/fc_bundle/0.21.1/libexec/vendor/lib/python3.10/site-packages', '/Users/chris/Library/Application Support/FreeCAD/Macro/', '/Users/chris/Library/Application Support/FreeCAD/Macro', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Macro', '/Users/chris/Library/Application Support/FreeCAD/Mod']
>>> import sys
>>> sys.path.apped('    ^C
KeyboardInterrupt
>>> import readline
>>> import rlcompleter
>>> readline.parse_and_bind("tab: complete")
>>> sys.path.append('/Users/chris/homebrew/freecad@0.21.2_py310/MacOS/PySide')
>>> import Draft
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Draft/Draft.py", line 53, in <module>
    from draftutils.utils import ARROW_TYPES as arrowtypes
  File "/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Draft/draftutils/utils.py", line 39, in <module>
    import PySide.QtCore as QtCore
ModuleNotFoundError: No module named 'PySide'
>>> import PySide
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'PySide'
>>> sys.path
['/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Material', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Import', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Part', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Tux', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/PartDesign', '/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Measure', '/Users/chris/homebrew/Cellar/freecad@0.21. [... cut ...]
  File "/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Draft/Draft.py", line 53, in <module>
    from draftutils.utils import ARROW_TYPES as arrowtypes
  File "/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/Mod/Draft/draftutils/utils.py", line 39, in <module>
    import PySide.QtCore as QtCore
ModuleNotFoundError: No module named 'PySide'
>>> import Draft
KeyboardInterrupt
>>> sys.path.append('/Users/chris/homebrew/Cellar/freecad@0.21.2_py310/0.21.2/MacOS')
>>> import PySide
>>> import Draft
>>>
```

ipatch, 2024-08-13:
building freecad from git source with cmake, and attempting to import Draft module i get the below error with my py312 + qt6 build.

```
╰─λ ~/.local/fc-git/MacOS/FreeCADCmd                                                                               130 (01:19.594)
FreeCAD 0.22.0, Libs: 0.22.0devR38385 (Git)
(c) Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

[FreeCAD Console mode <Use Ctrl-D (i.e. EOF) to exit.>]
>>> sys.path
['/Users/chris/.local/fc-git/Mod/Help', '/Users/chris/.local/fc-git/Mod/Material', '/Users/chris/.local/fc-git/Mod/Import', '/Users/chris/.local/fc-git/Mod/Part', '/Users/chris/.local/fc-git/Mod/Tux', '/Users/chris/.local/fc-git/Mod/PartDesign', '/Users/chris/.local/fc-git/Mod/BIM', '/Users/chris/.local/fc-git/Mod/Measure', '/Users/chris/.local/fc-git/Mod/Points', '/Users/chris/.local/fc-git/Mod/AddonManager', '/Users/chris/.local/fc-git/Mod/CAM', '/Users/chris/.local/fc-git/Mod/Idf', '/Users/chris/.local/fc-git/Mod/Inspection', '/Users/chris/.local/fc-git/Mod/Robot', '/Users/chris/.local/fc-git/Mod/Surface', '/Users/chris/.local/fc-git/Mod/OpenSCAD', '/Users/chris/.local/fc-git/Mod/Start', '/Users/chris/.local/fc-git/Mod/Sketcher', '/Users/chris/.local/fc-git/Mod/Show', '/Users/chris/.local/fc-git/Mod/MeshPart', '/Users/chris/.local/fc-git/Mod/Assembly', '/Users/chris/.local/fc-git/Mod/TechDraw', '/Users/chris/.local/fc-git/Mod/Web', '/Users/chris/.local/fc-git/Mod/Spreadsheet', '/Users/chris/.local/fc-git/Mod/Fem', '/Users/chris/.local/fc-git/Mod/Test', '/Users/chris/.local/fc-git/Mod/Plot', '/Users/chris/.local/fc-git/Mod/ReverseEngineering', '/Users/chris/.local/fc-git/Mod/Draft', '/Users/chris/.local/fc-git/Mod/Mesh', '/Users/chris/.local/fc-git/Mod', '/Users/chris/.local/fc-git/lib', '/Users/chris/.local/fc-git/Ext', '/Users/chris/homebrew/Cellar/python@3.12/3.12.4/Frameworks/Python.framework/Versions/3.12/lib/python312.zip', '/Users/chris/homebrew/Cellar/python@3.12/3.12.4/Frameworks/Python.framework/Versions/3.12/lib/python3.12', '/Users/chris/homebrew/Cellar/python@3.12/3.12.4/Frameworks/Python.framework/Versions/3.12/lib/python3.12/lib-dynload', '/Users/chris/homebrew/Cellar/python@3.12/3.12.4/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages', '/Users/chris/homebrew/Cellar/coin3d_py312/4.0.2/lib/python3.12/site-packages', '/Users/chris/homebrew/Cellar/openvino/2024.2.0_1/libexec/lib/python3.12/site-packages', '/Users/chris/homebrew/Cellar/pybind11/2.13.1/libexec/lib/python3.12/site-packages', '/Users/chris/Library/Application Support/FreeCAD/Macro/', '/Users/chris/Library/Application Support/FreeCAD/Macro', '/Users/chris/.local/fc-git/Macro', '/Users/chris/Library/Application Support/FreeCAD/Mod']
>>> import PySide
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'PySide'
>>> sys.path.append('/Users/chris/.local/fc-git/MacOS')
>>> import PySide
>>> import Draft
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/chris/.local/fc-git/Mod/Draft/Draft.py", line 53, in <module>
    from draftutils.utils import ARROW_TYPES as arrowtypes
  File "/Users/chris/.local/fc-git/Mod/Draft/draftutils/utils.py", line 42, in <module>
    from draftutils import params
  File "/Users/chris/.local/fc-git/Mod/Draft/draftutils/params.py", line 584, in <module>
    PARAM_DICT = _get_param_dictionary()
                 ^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/chris/.local/fc-git/Mod/Draft/draftutils/params.py", line 526, in _get_param_dictionary
    if fd.open(QtCore.QIODevice.ReadOnly | QtCore.QFile.Text):
               ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for |: 'OpenModeFlag' and 'OpenModeFlag'
>>>
```

ipatch, 2024-08-13:
```
➜  site-packages git:(master) ~/.local/fc-git/MacOS/FreeCADCmd
FreeCAD 0.22.0, Libs: 0.22.0devR38385 (Git)
(c) Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

[FreeCAD Console mode <Use Ctrl-D (i.e. EOF) to exit.>]
>>> sys.path
['/Users/chris/.local/fc-git/Mod/Help', '/Users/chris/.local/fc-git/Mod/Material', '/Users/chris/.local/fc-git/Mod/Import', '/Users/chris/.local/fc-git/Mod/Part', '/Users/chris/.local/fc-git/Mod/Tux', '/Users/chris/.local/fc-git/Mod/PartDesign', '/Users/chris/.local/fc-git/Mod/BIM', '/Users/chris/.local/fc-git/Mod/Measure', '/Users/chris/.local/fc-git/Mod/Points', '/Users/chris/.local/fc-git/Mod/AddonManager', '/Users/chris/.local/fc-git/Mod/CAM', '/Users/chris/.local/fc-git/Mod/Idf', '/Users/chris/.local/fc-git/Mod/Inspection', '/Users/chris/.local/fc-git/Mod/Robot', '/Users/chris/.local/fc-git/Mod/Surface', '/Users/chris/.local/fc-git/Mod/OpenSCAD', '/Users/chris/.local/fc-git/Mod/Start', '/Users/chris/.local/fc-git/Mod/Sketcher', '/Users/chris/.local/fc-git/Mod/Show', '/Users/chris/.local/fc-git/Mod/MeshPart', '/Users/chris/.local/fc-git/Mod/Assembly', '/Users/chris/.local/fc-git/Mod/TechDraw', '/Users/chris/.local/fc-git/Mod/Web', '/Users/chris/.local/fc-git/Mod/Spreadsheet', '/Users/chris/.local/fc-git/Mod/Fem', '/Users/chris/.local/fc-git/Mod/Test', '/Users/chris/.local/fc-git/Mod/Plot', '/Users/chris/.local/fc-git/Mod/ReverseEngineering', '/Users/chris/.local/fc-git/Mod/Draft', '/Users/chris/.local/fc-git/Mod/Mesh', '/Users/chris/.local/fc-git/Mod', '/Users/chris/.local/fc-git/lib', '/Users/chris/.local/fc-git/Ext', '/Users/chris/homebrew/Cellar/python@3.12/3.12.4/Frameworks/Python.framework/Versions/3.12/lib/python312.zip', '/Users/chris/homebrew/Cellar/python@3.12/3.12.4/Frameworks/Python.framework/Versions/3.12/lib/python3.12', '/Users/chris/homebrew/Cellar/python@3.12/3.12.4/Frameworks/Python.framework/Versions/3.12/lib/python3.12/lib-dynload', '/Users/chris/homebrew/Cellar/python@3.12/3.12.4/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages', '/Users/chris/homebrew/Cellar/coin3d_py312/4.0.2/lib/python3.12/site-packages', '/Users/chris/homebrew/Cellar/openvino/2024.2.0_1/libexec/lib/python3.12/site-packages', '/Users/chris/homebrew/Cellar/pybind11/2.13.1/libexec/lib/python3.12/site-packages', '/Users/chris/Library/Application Support/FreeCAD/Macro/', '/Users/chris/Library/Application Support/FreeCAD/Macro', '/Users/chris/.local/fc-git/Macro', '/Users/chris/Library/Application Support/FreeCAD/Mod']
>>> import Tux
>>> import BIM
>>> import AddonManager
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/chris/.local/fc-git/Mod/AddonManager/AddonManager.py", line 35, in <module>
    from PySide import QtGui, QtCore, QtWidgets
ModuleNotFoundError: No module named 'PySide'
>>> sys.path.append('/Users/chris/.local/fc-git/MacOS')
>>> import AddonManager
>>>
```

luzpaz, 2024-10-20:
Is this still relevant for 1.0 ? If not, then probably worth closing ?

ipatch, 2024-10-20:
> Is this still relevant for 1.0 ? If not, then probably worth closing ?

i still haven't published a formula that build and links against qt version 6, yet, so i plan keeping it open until i get a new formula in the tap.

luzpaz, 2024-10-20:
Thanks for all your work, @ipatch !

ipatch, 2024-12-30:
this has since been resolved with recently published formula from PR #662

https://github.com/FreeCAD/homebrew-freecad/blob/master/Formula/freecad%401.0.0_py312_qt6.rb

closed 2024-12-30 by ipatch