FreeCAD/homebrew-freecad #555: freecad@0.21.2_py310: build error

ipatch, 2024-08-26:
attempting to rebuild formula `freecad@0.21.2_py310.rb` this afternoon (aug 26 2024) it appears the upstream boost project has updated some of their api's (possibly). and with the updates, the current formula is failing to build, so it looks like a new patch file will be required.

build error,

```
[ 56%] Building CXX object src/Gui/CMakeFiles/FreeCADGui.dir/PythonDebugger.cpp.o
cd /Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/build/src/Gui && /Users/chris/homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_PP_VARIADICS=1 -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DBUILD_ADDONMGR -DCMAKE_BUILD_TYPE=\"RelWithDebInfo\" -DFreeCADGui_EXPORTS -DGL_SILENCE_DEPRECATION -DHAVE_CONFIG_H -DHAVE_PYSIDE2 -DHAVE_SHIBOKEN2 -DNDEBUG -DPYCXX_6_2_COMPATIBILITY -DPYSIDE_QML_SUPPORT=1 -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_NO_KEYWORDS -DQT_OPENGL_LIB -DQT_PRINTSUPPORT_LIB -DQT_SVG_LIB -DQT_UITOOLS_LIB -DQT_WIDGETS_LIB -DQT_XML_LIB -D_OCC64 -I/Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/build/src/Gui/FreeCADGui_autogen/include -I/Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/build -I/Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/build/src -I/Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/FreeCAD-0.21.2/src -I/Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/FreeCAD-0.21.2/src/Gui -I/Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/FreeCAD-0.21.2/src/Gui/Quarter -I/Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/build/src/Gui -I/Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/FreeCAD-0.21.2/src/Gui/.. -I/Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/build/src/Gui/.. -I/Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/build/src/Gui/Language -I/Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/build/src/Gui/propertyeditor -I/Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/build/src/Gui/TaskView -I/Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/build/src/Gui/Quarter -I/Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/build/src/Gui/DAGView -I/Users/chris/homebrew/opt/eigen/include/eigen3 -I/Users/chris/homebrew/opt/xerces-c/include -I/Users/chris/homebrew/opt/pyside2@5.15.11_py310/include/PySide2/QtCore -I/Users/chris/homebrew/opt/pyside2@5.15.11_py310/include/PySide2/QtGui -I/Users/chris/homebrew/opt/pyside2@5.15.11_py310/include/PySide2/QtWidgets -I/Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/build/_deps/fmt-src/include -F/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks -isystem /Users/chris/homebrew/Cellar/boost/1.86.0/include -isystem /Users/chris/homebrew/opt/python@3.10/Frameworks/Python.framework/Versions/3.10/include/python3.10 -iframework /Users/chris/homebrew/opt/qt@5/lib -isystem /Users/chris/homebrew/opt/qt@5/lib/QtCore.framework/Headers -isystem /Users/chris/homebrew/opt/qt@5/./mkspecs/macx-clang -isystem /Users/chris/homebrew/opt/qt@5/lib/QtWidgets.framework/Headers -isystem /Users/chris/homebrew/opt/qt@5/lib/QtGui.framework/Headers -isystem /Users/chris/homebrew/opt/qt@5/lib/QtOpenGL.framework/Headers -isystem /Users/chris/homebrew/opt/qt@5/lib/QtPrintSupport.framework/Headers -isystem /Users/chris/homebrew/opt/qt@5/lib/QtSvg.framework/Headers -isystem /Users/chris/homebrew/opt/qt@5/lib/QtNetwork.framework/Headers -isystem /Users/chris/homebrew/opt/qt@5/include/QtUiTools -isystem /Users/chris/homebrew/opt/shiboken2@5.15.11_py310/include/shiboken2 -isystem /Users/chris/homebrew/opt/pyside2@5.15.11_py310/include/PySide2 -isystem /Users/chris/homebrew/opt/qt@5/lib/QtXml.framework/Headers -Wall -Wextra -Wpedantic -Wno-write-strings  -Wno-undefined-var-template -O2 -g -DNDEBUG -std=gnu++17 -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.2.sdk -mmacosx-version-min=14.6.1 -fPIC -fPIC -MD -MT src/Gui/CMakeFiles/FreeCADGui.dir/PythonDebugger.cpp.o -MF CMakeFiles/FreeCADGui.dir/PythonDebugger.cpp.o.d -o CMakeFiles/FreeCADGui.dir/PythonDebugger.cpp.o -c /Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/FreeCAD-0.21.2/src/Gui/PythonDebugger.cpp
/Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/FreeCAD-0.21.2/src/Mod/Part/App/Geometry.cpp:388:12: error: no type named 'mt19937' in namespace 'boost'; did you mean 'std::mt19937'?
    static boost::mt19937 ran;
           ^~~~~~~~~~~~~~
           std::mt19937
/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/usr/include/c++/v1/__random/mersenne_twister_engine.h:522:73: note: 'std::mt19937' declared here
                                18, 1812433253>                         mt19937;
                                                                        ^
/Users/chris/tmp/homebrew/freecadA0.21.2_py310-20240826-10804-z3nbun/FreeCAD-0.21.2/src/Mod/Part/App/Geometry.cpp:395:56: error: no member named 'mt19937' in namespace 'boost'
    static boost::uuids::basic_random_generator<boost::mt19937> gen(&ran);
                                                ~~~~~~~^
[ 56%] Building CXX object src/Mod/Part/App/CMakeFiles/Part.dir/Geometry2d.cpp.o
```

ipatch, 2024-08-26:
```
╰─λ brew info boost                                                                                                  1 (06:37.289)
==> boost: stable 1.86.0 (bottled), HEAD
Collection of portable C++ source libraries
https://www.boost.org/
Installed
/Users/chris/homebrew/Cellar/boost/1.86.0 (16,215 files, 518.0MB) *
  Poured from bottle on 2024-08-25 at 20:26:48
From: https://github.com/ipatch/homebrew-core/blob/HEAD/Formula/b/boost.rb
License: BSL-1.0
==> Dependencies
Required: icu4c ✔, xz ✔, zstd ✔
==> Options
--HEAD
        Install HEAD version
```

ipatch, 2024-08-26:
may be related.

https://github.com/FreeCAD/FreeCAD/commit/c9a586f34a3f7892ce934e4114417e0916de295d

closed 2024-08-27 by ipatch

MinecraftEarthVillage, 2024-11-09:
不要下载我前段时间发的文件，那不是我发的，是另一个人控制我，如果你使用了这个病毒文件，你也会被控制，并且疯狂转发这个病毒
Don't download the file I sent some time ago, it wasn't me, but another person controlled me. If you use this virus file, you will also be controlled and forward this virus crazily.