FreeCAD/homebrew-freecad #560: FreeCAD 1.0-rc1

Andor, 2024-09-07:
Hello! Could you please add `1.0-rc1` (and consequent rc versions)?
Also, maybe it makes sense add weekly builds also?

ipatch, 2024-09-07:
didn't realize `1.0-rc1` had been released. give me a couple of days and i'll see if i can cook up a formula file 🤞.

ipatch, 2024-09-07:
https://github.com/FreeCAD/FreeCAD/releases/tag/1.0rc1

Andor, 2024-09-07:
The build is here: https://github.com/FreeCAD/FreeCAD/actions/runs/10715200858/job/29710213889
I guess I have no access to artefacts, tho.

Andor, 2024-09-10:
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc1

ipatch, 2024-10-02:
hey for what it's worth, i finally got my local homebrew setup updated, ie. running the latest version of brew and all formula updated ie. no outdated formula on my local arch linux install using "linuxbrew" then reran, `brew install freecad -v  --HEAD` and am able to build the latest git --HEAD of freecad, i got the below splash screen (looks like the splash image rotates to a new image everytime i launch freecad) with the below results running, the below test command,

[image]

<details>

<summary> test output from above command <code>$(brew --prefix)/bin/freecad -t 0</code> </summary>

```
╭─capin at rogue in /home/capin
╰─λ freecad -t 0
FreeCAD 1.1.0, Libs: 1.1.0devR38871 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

import TestFemImport
import TestObjectExistance
import TestFemCommon
import TestObjectCreate
import TestObjectType
import TestObjectOpen
import TestMaterialUnits
import TestObjectOpen
<Sketch> SketchObject.cpp(345): Edge too small: Edge2
Updating geometry: Error build geometry(0): Both points are equal
Invalid solution from DogLeg solver.
Updating geometry: Error build geometry(0): Both points are equal
Invalid solution from LevenbergMarquardt solver.
Updating geometry: Error build geometry(0): Both points are equal
Invalid solution from BFGS solver.
<PropertyLinks> PropertyLinks.cpp(449): SketchValidateCoincidentsTest#Sketch001.ExternalGeometry missing element reference SketchValidateCoincidentsTest#Sketch ;g3;SKT.Edge3
<Sketch> SketchObject.cpp(9052): Failed to project external geometry in SketchValidateCoincidentsTest#Sketch001: Sketch.;g3;SKT
Invalid shape name ?Edge3
<Sketch> SketchObject.cpp(9134): External geometry SketchValidateCoincidentsTest#Sketch001.e1 missing reference: Sketch.;g3;SKT
<Sketch> SketchObject.cpp(9134): External geometry SketchValidateCoincidentsTest#Sketch001.e1 missing reference: Sketch.;g3;SKT
Sketch001: Sketch with redundant constraints
Please remove the following redundant constraint:
2

SketchObject::onChanged(): Unmanaged change of Constraint Property results in invalid constraint indices
4.1e-08 <App> Document.cpp(2540): The graph must be a DAG.
0.995636 Tree.cpp(296): [ComboView] duplicate child item SketchSolverTest#Body.Hole
1.00103 Tree.cpp(296): [ComboView] duplicate child item SketchSolverTest#Body.Hole
2.81073 Tree.cpp(296): [ComboView] duplicate child item SketchSolverTest#Body.Hole
2.83105 <App> Document.cpp(2900): SketchSolverTest#Hole still touched after recompute
2.83564 <App> Document.cpp(2540): The graph must be a DAG.
<PropertyLinks> PropertyLinks.cpp(429): SketchSolverTest#Sketch2.ExternalGeometry auto change element reference SketchSolverTest#Hole ;#bf:1;:M;CUT;:Hdf2:7,E.Edge29 -> ;#bf:1;:M#158;CUT;:H-df2:b,E.Edge29
2.98056 Tree.cpp(296): [ComboView] duplicate child item SketchSolverTest#Body.Hole
<PropertyLinks> PropertyLinks.cpp(429): SketchSolverTest#Sketch2.ExternalGeometry auto change element reference SketchSolverTest#Hole ;#bf:1;:M#158;CUT;:H-df2:b,E.Edge29 -> ;#bf:1;:M#15e;CUT;:H-df2:b,E.Edge29
3.13875 <App> Document.cpp(2900): SketchSolverTest#Hole still touched after recompute
TDHatch path: /home/capin/homebrew/Cellar/freecad@0.21.2_py310/HEAD-813ee36/Mod/TechDraw/TDTest
making hatch
finished hatch
document created
box created
sphere created
Fusion created
making a page
Page created
making a projection group
Group created
adding views
added Front
Anchor values set
Front/Anchor recomputed
added Left
added Top
added Right
added Rear
added Bottom
[... 1273 lines cut ...]
System exit
```

</details>

```
OS: Arch Linux (i3/i3)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.10.15, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.1, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * A2plus 0.4.65
  * Curves 0.6.42
```

---

[image]

[image]

ipatch, 2024-10-17:
i'm going to go ahead and close this issue, as i just published a new formula based on the rc2 release. you can see PR #574

closed 2024-10-17 by ipatch