FreeCAD/homebrew-freecad #561: netgen dependency error

sslupsky, 2024-09-24:
Trying to install the netgen dependency and I am getting an error:

```
==> Fetching freecad/freecad/nglib@6.2.2105
==> Downloading https://github.com/FreeCAD/homebrew-freecad/releases/download/nglib@6.2.2105-6.2.2105_1/nglib%406.2.2105-6.2.2105_1.big_sur.bottle.tar.gz
curl: (56) The requested URL returned error: 404                                                                                                                                                                                  

Error: nglib@6.2.2105: Failed to download resource "nglib@6.2.2105"
Download failed: https://github.com/FreeCAD/homebrew-freecad/releases/download/nglib@6.2.2105-6.2.2105_1/nglib%406.2.2105-6.2.2105_1.big_sur.bottle.tar.gz
```

I am installing on macOS 14.6.1.  Not sure why it is attempting to download a big_sur bottle?

ipatch, 2024-09-24:
you could try building netgen from source, and not sure why it's trying to install the big sur bottle.

try something like,

```
brew search netgen
brew install netgen
```

i haven't updated the formula in a while though, and there may be a newer version of netgen available.

sslupsky, 2024-09-24:
Not showing when I do a brew search.  Netgen only appears in the freecad tap.

sslupsky, 2024-09-24:
I found a cmake switch to turn off the Netgen build for the time being and was able to compile it.

luzpaz, 2024-10-17:
Is there progress here ? What's the workflow ?

ipatch, 2024-10-17:
> Is there progress here ? What's the workflow ?

could you please elaborate on your question? ie. are you attempting to install freecad using `brew install` ?